jneilliii/OctoPrint-BedLevelVisualizer #263: Please make a basic setup video/ detailed instructions with example

breadman6, 2020-06-14:
I'm not having an easy time getting this set up. 
I had this working then had to reload octopi. I cannot get this working again. 
I have bed leveling expert plugin on my system which adds "Bed Leveling" button to the control tab on octopi, as well as your plugin, bed leveling visualizer, this used to be the button that generated the 3d topography map.

Long story short, I have the recommended script for bilinear leveling saved. I have tried to add @BEDLEVELVISUALIZER in terminal before the leveling results are printed. I have messed with the timeout settings. I have rebooted the system multiple times. I'd just like to see a video of someone installing and getting this working from the start. The readme of this isn't detailed enough. Need like a step by step instructions. 

Would be great if your plugin generated a button on the control screen that would run your plugin's script. Sorry, not the most tech oriented person, so this is giving me a headache. Please make it more accessible for others like me.

jneilliii, 2020-06-14:
The biggest problem with your issue is you haven't really provided anything that could actually help you. The largest problem that I've seen is not knowing what your commands are and on initial install it doesn't always work when you put the commands in the wizard only (Next version removes the wizard all together to get rid of this issue). Do this for me. Go into the plugin's settings, enable debug logging, save, and then restart OctoPrint. Once the page reloads (if it doesn't prompt to reload press shift+F5 to force a reload) go to the tab and press the update mesh button. If it doesn't work, provide your plugin_bedlevelvisualizer_debug.log file and a screenshot of your settings on the General tab.

breadman6, 2020-06-14:
Ok, so, my problem was I couldn't find the tab for the bed visualizer LOL. Mesh was being updated, just couldn't see it. Had to click the dang 4 line button on the upper right of the octopi interface. That's what happens when you step away from this for a few months and jump back into it expecting it to be like riding a bike. My bad. Works great, great work. But a few advertising your work would only help advertise. Throw up a patreon. All I see is these videos of "Top 10 Octopi Plugins". Get someone to just do a review and demo of your plugin with the next version. Bed leveling is the Achilles heel of all 3d printers.
Thanks again.

jneilliii, 2020-06-14:
Click sponsor above and you'll find links for my Patreon. It's been added to the readme but that branch hasn't been pushed to the Master branch yet as I'm trying to make some processing improvements and removing the wizard, etc.

DoughyInTheMiddle, 2020-07-11:
I look forward to some docs too. I'd have figured those in the know on YT would have done it (Chep, Teaching Tech, etc.), but the only finds for octoprint and visualizer are just to recommend it.

Among the things are how to necessarily read and make the adjustments. Sure, the graph shows the tilt, and the the chart shows the numeric values to adjust by, and it seems intuitive that the +/- means it's too high (+) and too low (-) and to adjust accordingly, but I'm not 100% sure how to make those adjustments.

Additionally, when you hit an error from the BLTouch read (it hits the bed and then blinks), that produces craters on the visualizer. Is this bed? Is this BLTouch? Is there just a leveling adjustment that can compensate?

I love the app, and the informational feed back is great, but a guide on how to use the information to make the adjustments without just inferring would be awesome. Thank you for all your work on this great program!

jneilliii, 2020-07-26:
I've reached out via twitter to some known YouTubers and [crosslink](https://www.youtube.com/c/CrosslinkChannel) has accepted the challenge. Daniel stated it would take about 4 weeks before he can get to it, but hopefully we'll have something soon.

kreuzhofer, 2020-09-10:
I am still busy, didn't forget about this but everything is delayed at the moment.