jneilliii/OctoPrint-BedLevelVisualizer #264: Can't get mesh data from Klipper

NameOfTheDragon, 2020-06-16:
I've seen Issue #92, which suggests the following should be used to get mesh data from Klipper:

```
@BEDLEVELVISUALIZER
BED_MESH_OUTPUT
```
And I have put that in my config like so:
[image]

However this is not working. The update process hangs for a long time and then times out. In the terminal window I see the following output:

```
Send: BED_MESH_OUTPUT
Recv: // Mesh Leveling Probed Z positions:
Recv: // -0.157500 -0.096250 -0.050000 -0.048750 -0.072500
Recv: // -0.088750 -0.065000 -0.040000 -0.046250 -0.045000
Recv: // -0.088750 -0.040000 -0.016250 -0.013750 -0.035000
Recv: // -0.118750 -0.055000 -0.008750 -0.007500 -0.007500
Recv: // -0.125000 -0.062500 -0.027500 -0.018750 -0.017500
Recv: Mesh X,Y: 37,37
Recv: Search Height: 5
Recv: Mesh Average: -0.05
Recv: Mesh Range: min=-0.1575 max=-0.0070
Recv: Interpolation Algorithm: bicubic
Recv: Measured points:
Recv:   -0.125000  -0.121973  -0.116181  -0.108389  -0.099362  -0.089866  -0.080667  -0.072529  -0.066218  -0.062500  -0.059684  -0.055798  -0.051185  -0.046190  -0.041156  -0.036426  -0.032344  -0.029254  -0.027500  -0.026454  -0.025296  -0.024083  -0.022870  -0.021713  -0.020667  -0.019787  -0.019130  -0.018750  -0.018534  -0.018333  -0.018148  -0.017982  -0.017836  -0.017713  -0.017615  -0.017543  -0.017500
Recv:   -0.124755  -0.121725  -0.115927  -0.108127  -0.099091  -0.089584  -0.080371  -0.072219  -0.065894  -0.062160  -0.059312  -0.055364  -0.050671  -0.045583  -0.040453  -0.035633  -0.031474  -0.028330  -0.026552  -0.025514  -0.024387  -0.023220  -0.022064  -0.020968  -0.019982  -0.019155  -0.018537  -0.018178  -0.017969  -0.017773  -0.017591  -0.017426  -0.017281  -0.017157  -0.017058  -0.016986  -0.016943
Recv:   -0.124322  -0.121284  -0.115473  -0.107656  -0.098598  -0.089066  -0.079826  -0.071645  -0.065289  -0.061525  -0.058611  -0.054543  -0.049689  -0.044418  -0.039099  -0.034101  -0.029791  -0.026540  -0.024716  -0.023693  -0.022624  -0.021547  -0.020501  -0.019524  -0.018654  -0.017929  -0.017387  -0.017066  -0.016871  -0.016681  -0.016502  -0.016336  -0.016187  -0.016059  -0.015955  -0.015879  -0.015835
Recv:   -0.123731  -0.120684  -0.114857  -0.107016  -0.097930  -0.088365  -0.079089  -0.070869  -0.064473  -0.060667  -0.057666  -0.053436  -0.048368  -0.042852  -0.037280  -0.032043  -0.027531  -0.024136  -0.022250  -0.021247  -0.020256  -0.019300  -0.018402  -0.017585  -0.016871  -0.016283  -0.015843  -0.015574  -0.015396  -0.015217  -0.015041  -0.014875  -0.014722  -0.014588  -0.014478  -0.014398  -0.014352
Recv:   -0.123018  -0.119960  -0.114115  -0.106249  -0.097133  -0.087533  -0.078218  -0.069955  -0.063513  -0.059660  -0.056562  -0.052148  -0.046835  -0.041040  -0.035180  -0.029670  -0.024928  -0.021371  -0.019414  -0.018434  -0.017533  -0.016717  -0.015989  -0.015355  -0.014821  -0.014390  -0.014068  -0.013860  -0.013703  -0.013538  -0.013370  -0.013205  -0.013052  -0.012915  -0.012802  -0.012718  -0.012671
Recv:   -0.122214  -0.119146  -0.113285  -0.105397  -0.096253  -0.086621  -0.077268  -0.068965  -0.062480  -0.058580  -0.055381  -0.050779  -0.045217  -0.039138  -0.032983  -0.027196  -0.022219  -0.018495  -0.016466  -0.015511  -0.014705  -0.014033  -0.013483  -0.013040  -0.012692  -0.012425  -0.012226  -0.012082  -0.011949  -0.011802  -0.011647  -0.011492  -0.011344  -0.011210  -0.011098  -0.011016  -0.010971
Recv:   -0.121352  -0.118277  -0.112404  -0.104501  -0.095337  -0.085680  -0.076299  -0.067963  -0.061440  -0.057500  -0.054209  -0.049435  -0.043643  -0.037301  -0.030874  -0.024831  -0.019638  -0.015761  -0.013667  -0.012737  -0.012020  -0.011486  -0.011105  -0.010844  -0.010673  -0.010563  -0.010481  -0.010398  -0.010292  -0.010169  -0.010035  -0.009898  -0.009765  -0.009644  -0.009542  -0.009467  -0.009426
Recv:   -0.120465  -0.117388  -0.111511  -0.103602  -0.094430  -0.084762  -0.075367  -0.067011  -0.060464  -0.056494  -0.053130  -0.048217  -0.042239  -0.035684  -0.029039  -0.022789  -0.017420  -0.013420  -0.011275  -0.010368  -0.009729  -0.009314  -0.009076  -0.008971  -0.008952  -0.008976  -0.008996  -0.008968  -0.008890  -0.008796  -0.008693  -0.008586  -0.008482  -0.008387  -0.008306  -0.008247  -0.008215
Recv:   -0.119587  -0.116513  -0.110642  -0.102742  -0.093579  -0.083919  -0.074529  -0.066174  -0.059621  -0.055636  -0.052228  -0.047228  -0.041134  -0.034445  -0.027661  -0.021280  -0.015802  -0.011726  -0.009549  -0.008662  -0.008081  -0.007752  -0.007618  -0.007626  -0.007718  -0.007839  -0.007935  -0.007949  -0.007899  -0.007843  -0.007782  -0.007721  -0.007662  -0.007609  -0.007565  -0.007532  -0.007514
Recv:   -0.118750  -0.115686  -0.109836  -0.101963  -0.092831  -0.083203  -0.073843  -0.065513  -0.058978  -0.055000  -0.051587  -0.046571  -0.040454  -0.033738  -0.026926  -0.020519  -0.015019  -0.010929  -0.008750  -0.007876  -0.007325  -0.007037  -0.006953  -0.007013  -0.007157  -0.007326  -0.007461  -0.007500  -0.007478  -0.007466  -0.007463  -0.007466  -0.007473  -0.007481  -0.007490  -0.007497  -0.007500
Recv:   -0.117151  -0.114118  -0.108326  -0.100533  -0.091493  -0.081964  -0.072701  -0.064460  -0.057997  -0.054069  -0.050713  -0.045791  -0.039794  -0.033214  -0.026540  -0.020265  -0.014880  -0.010875  -0.008741  -0.007883  -0.007339  -0.007053  -0.006966  -0.007023  -0.007165  -0.007337  -0.007482  -0.007541  -0.007576  -0.007673  -0.007815  -0.007985  -0.008166  -0.008342  -0.008495  -0.008609  -0.008668
Recv:   -0.114220  -0.111246  -0.105567  -0.097925  -0.089062  -0.079720  -0.070644  -0.062574  -0.056254  -0.052425  -0.049194  -0.044482  -0.038754  -0.032476  -0.026115  -0.020136  -0.015006  -0.011190  -0.009154  -0.008323  -0.007777  -0.007467  -0.007346  -0.007365  -0.007473  -0.007624  -0.007768  -0.007857  -0.007998  -0.008301  -0.008722  -0.009214  -0.009732  -0.010230  -0.010664  -0.010986  -0.011153
Recv:   -0.110343  -0.107450  -0.101922  -0.094484  -0.085860  -0.076774  -0.067950  -0.060113  -0.053987  -0.050296  -0.047239  -0.042819  -0.037467  -0.031614  -0.025689  -0.020125  -0.015351  -0.011799  -0.009898  -0.009101  -0.008546  -0.008198  -0.008020  -0.007975  -0.008027  -0.008139  -0.008275  -0.008398  -0.008679  -0.009259  -0.010051  -0.010974  -0.011943  -0.012873  -0.013680  -0.014282  -0.014593
Recv:   -0.105904  -0.103105  -0.097755  -0.090557  -0.082213  -0.073427  -0.064900  -0.057336  -0.051438  -0.047908  -0.045055  -0.040979  -0.036068  -0.030713  -0.025302  -0.020224  -0.015869  -0.012625  -0.010883  -0.010124  -0.009558  -0.009162  -0.008913  -0.008791  -0.008772  -0.008835  -0.008958  -0.009119  -0.009560  -0.010453  -0.011672  -0.013087  -0.014570  -0.015993  -0.017229  -0.018149  -0.018625
Recv:   -0.101288  -0.098590  -0.093431  -0.086490  -0.078447  -0.069981  -0.061772  -0.054501  -0.048846  -0.045487  -0.042850  -0.039136  -0.034691  -0.029861  -0.024990  -0.020425  -0.016511  -0.013594  -0.012018  -0.011301  -0.010721  -0.010274  -0.009951  -0.009747  -0.009653  -0.009664  -0.009772  -0.009971  -0.010576  -0.011794  -0.013450  -0.015372  -0.017385  -0.019317  -0.020994  -0.022242  -0.022888
Recv:   -0.096880  -0.094283  -0.089314  -0.082629  -0.074885  -0.066738  -0.058845  -0.051864  -0.046449  -0.043259  -0.040831  -0.037466  -0.033470  -0.029144  -0.024794  -0.020722  -0.017232  -0.014628  -0.013213  -0.012536  -0.011946  -0.011452  -0.011061  -0.010780  -0.010617  -0.010579  -0.010673  -0.010907  -0.011667  -0.013188  -0.015255  -0.017651  -0.020161  -0.022569  -0.024658  -0.026213  -0.027019
Recv:   -0.093064  -0.090561  -0.085767  -0.079319  -0.071852  -0.064000  -0.056399  -0.049684  -0.044490  -0.041451  -0.039206  -0.036146  -0.032539  -0.028651  -0.024751  -0.021107  -0.017984  -0.015652  -0.014377  -0.013736  -0.013142  -0.012612  -0.012168  -0.011826  -0.011608  -0.011531  -0.011616  -0.011881  -0.012769  -0.014544  -0.016953  -0.019745  -0.022669  -0.025473  -0.027907  -0.029718  -0.030656
Recv:   -0.090226  -0.087801  -0.083156  -0.076908  -0.069674  -0.062070  -0.054713  -0.048219  -0.043206  -0.040289  -0.038183  -0.035350  -0.032032  -0.028468  -0.024902  -0.021572  -0.018721  -0.016590  -0.015419  -0.014810  -0.014218  -0.013672  -0.013197  -0.012821  -0.012571  -0.012474  -0.012556  -0.012844  -0.013822  -0.015769  -0.018412  -0.021474  -0.024680  -0.027755  -0.030423  -0.032409  -0.033437
[... cut ...] .048392  -0.047935  -0.047682  -0.047611  -0.047704  -0.047938  -0.048294  -0.048750  -0.049907  -0.052170  -0.055222  -0.058750  -0.062438  -0.065972  -0.069037  -0.071318  -0.072500
Recv: 
Recv: ok
```
Perhaps I'm doing it wrong but the only information I could find on how to set things up was Issue #92 . It obviously did work for some people then but its not working for me now.

I can't see how the bed visualizer plugin would be able to determine the probing X,Y locations from the data above.

jneilliii, 2020-06-16:
You are correct. That should be working for you. Do me a favor and enable debug logging in the plugin's settings and restart OctoPrint and try again. If it doesn't work, upload your plugin_bedlevelvisualizer_debug,log file from the logging section of OctoPrint's settings.

NameOfTheDragon, 2020-06-17:
Heh - you know what? It's working today. No idea what changed!! Sorry for the false alarm.

closed 2020-06-17 by NameOfTheDragon