jneilliii/OctoPrint-BedLevelVisualizer #265: Hitting "home" button resets the mesh

BNWEIN, 2020-06-22:
Hi.

I run "Update Mesh" command, and it works, and shows the current mesh, however, when i hit the "home" button, it resets the mash back to an older or non existent mesh?

BNWEIN, 2020-06-22:
Sorry i should have added more info

Octoprint 1.4.0 running on Octopi 0.17.0
Bed Visualizer 0.1.13 

I have uninstalled Bed Visualizer, rebooted, re-installted, same issue.

jneilliii, 2020-06-22:
You are correct, that is firmware related and not an issue with the plugin itself. After running the mesh are you sending any gcode commands to save the mesh to  EEPROM? The option to store mesh does not do that for you, it just stores the offsets into octoprint's settings, not EEPROM.

BNWEIN, 2020-06-22:
So in my settings i just have 

G28
G29 T

nothing else.  This used to work as far as i know (i had to reinstall octoprint fresh). 

What can i add to the above? 

As an FYI, if i reboot it shows the last mesh that is correct. Then a run a command to get a new mesh, this works, and shows the correct mesh, then i hit the little home button within bed visualiser, and it switches back to an old version? I can demo this on a video if that is easier?  I just want to make sure we are talking about the same issue?

jneilliii, 2020-06-22:
Typically `M500` at the end would store your mesh.

BNWEIN, 2020-06-22:
Sorry still not sure you are understanding the issue. Either way, after adding M500 the issue continues.

FYI again to be clear, the mesh is working, Bed Levelling is working, its just the image shown on the visualiser page that seems to reset when hitting the little "home" button on the visualiser page which is suppose to reset the image so the front is facing the right way.

Would you like a video?

BNWEIN, 2020-06-22:
Please review video here.

Please note what the mesh looks like before i press the "home" button

[https://youtu.be/0_IR_b-kbk0](url)

jneilliii, 2020-06-22:
Oh, thanks for the video. I was thinking you were running home `G28` and it was clearing the mesh. Now it makes sense. That home button stuff was added by someone else, so I'll have to dig into how they implemented that and figure out what's going on.

BNWEIN, 2020-06-22:
Thanks. I did think maybe i wasnt being very clear so i am glad the video helped!  Let me know anything i can do to help further investigate!

tablatronix, 2020-07-03:
Same here, I update meh sit shows new mesh, I click home and it shows some old mesh, cache maybe ?

BNWEIN, 2020-07-03:
Is there any further update here how a resolution. Its a really frustrating issue.

jneilliii, 2020-07-03:
Sorry, I have a full time job and haven't had the time to work on this. If you want faster resolution, dig in and fix it yourself, I'll be happy to accept a pull request.

BNWEIN, 2020-07-03:
Sorry i totally didn't mean to sound ungrateful but can see i did! Apologies! If i had the skill you do to be able to work on it, then i would but unfortunately i don't.

Sorry if i came across badly!

tablatronix, 2020-07-03:
I can try to take a look, but if the code is not commented I probably won't dig too deep.
Ill also look for the commit that added that

jneilliii, 2020-07-03:
lol, code is definitely not commented @tablatronix but the line where you want to look is [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/2e6b0d41059311722766530f8371eb10a29bcd0c/octoprint_bedlevelvisualizer/static/js/bedlevelvisualizer.js#L241.

No problem @BNWEIN, just trying to set expectations.

BNWEIN, 2020-07-27:
I believe this issue is now fixed in the latest update! 

Thank you!

jneilliii, 2020-07-27:
Thanks @BNWEIN, I don't remember changing anything in the code to fix it. Feel free to re-open this ticket if it comes back.

BNWEIN, 2020-07-28:
Sorry, can you re-open this. I was wrong. The issue is still there.

jneilliii, 2020-08-01:
I think is should be fixed by the above commit. If you don't mind giving it a test for me that would be great. Use the URL below in Plugin Manager > Get More > ...from URL and click Install button.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.15.zip

jneilliii, 2020-08-01:
I just pushed a small change that utilizes the in-built `resetCameraLastSave3d` button in the modebar. This should be a better approach. You can use the same URL above to install if you haven't already. Otherwise just reinstall with that same one.

BNWEIN, 2020-08-03:
Good morning.

One strange thing here. The "Home" button has disappeared and is no longer available?  So is the fix just to remove this button, so the mesh cannot be reset when you click it?  That button is needed really? Its now very difficult for me to identify which is the front corner.

jneilliii, 2020-08-03:
No, I'm just using the default button for resetting the view. The icon looks like a film camera is named resetCameraLastSave3d when you hover over it.

TBJ, 2020-09-03:
Ran across this thread today, I tested out the 0.1.15 and can confirm that the new 'film camera' icon resets the view without loading older data.

jneilliii, 2020-09-03:
Thanks for confirmation @TBJ

jneilliii, 2020-09-03:
I have a couple of more things I'm trying to figure out before official release.

commit 4abb610e (2020-08-01): change home button press to a different strategy, resolves #265
commit be84a20d (2020-08-01): update plotly to version 1.54.7
add option to download snapshots on graph rendering completion, #239
remove custom home button in preference of resetCameraLastSave3d, resolves #265
commit 3fb614e6 (2020-10-03): 0.1.15 (#304)

fix axis label colors, resolves #278
make colorscale limits allow option auto for the old method of coloring, resolves #275
add responsive option to plotly config, resolves #274
first stab at SD card support, resolves #223
fix little issue with reactive option where graph was getting overlaid on top of correction table
change home button press to a different strategy, resolves #265
update plotly to version 1.54.7
add option to download snapshots on graph rendering completion, #239
remove custom home button in preference of resetCameraLastSave3d, resolves #265
numpy versioning fo [... cut]
commit 7565c066 (2020-10-03): 0.1.15 (#304)

fix axis label colors, resolves #278
make colorscale limits allow option auto for the old method of coloring, resolves #275
add responsive option to plotly config, resolves #274
first stab at SD card support, resolves #223
fix little issue with reactive option where graph was getting overlaid on top of correction table
change home button press to a different strategy, resolves #265
update plotly to version 1.54.7
add option to download snapshots on graph rendering completion, #239
remove custom home button in preference of resetCameraLastSave3d, resolves #265
numpy versioning fo [... cut]
closed 2020-10-03 by jneilliii