jneilliii/OctoPrint-BedLevelVisualizer #266: Problem installing plugin on OctoPrint-python3

sylwesterr, 2020-06-26:
When upgrading OctoPrint to use python3 (fresh installation actually) the plugin installation reports: "Plugin management log

    Install Unknown: "
and after restart the plugin does not appear on home screen as well as not in OctoPrint settings as installed.
Here is the output from installation "terminal":
Installing plugin "Bed Level Visualizer" from https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip.../home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip --no-cache-dirLooking in indexes: https://pypi.org/simple, https://www.piwheels.org/simpleCollecting https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zipDownloading https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zipRequirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from Bed-Visualizer==0.1.13) (1.4.0)Collecting numpyDownloading https://www.piwheels.org/simple/numpy/numpy-1.19.0-cp37-cp37m-linux_armv7l.whl (10.5 MB)Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (1.6.5)Requirement already satisfied: netifaces<0.11,>=0.10.9 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (0.10.9)Requirement already satisfied: PyYAML<6,>=5.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (5.3.1)Requirement already satisfied: filetype<2,>=1.0.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (1.0.7)Requirement already satisfied: pkginfo<1.6,>=1.5.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (1.5.0.1)Requirement already satisfied: Jinja2<2.9,>=2.8.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (2.8.1)Requirement already satisfied: wrapt<1.12,>=1.11.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (1.11.2)Requirement already satisfied: Click<8,>=7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (7.1.2)Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (2020.6.8)Requirement already satisfied: pylru<1.3,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (1.2.0)Requirement already satisfied: watchdog<0.10,>=0.9.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (0.9.0)Requirement already satisfied: pyserial<3.5,>=3.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (3.4)Requirement already satisfied: frozendict<1.3,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (1.2)Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (3.1.1)Requirement already satisfied: tornado==4.5.3 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (4.5.3)Requirement already satisfied: semantic-version<2.9,>=2.8 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (2.8.5)Requirement already satisfied: future<0.19,>=0.18.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (0.18.2)Requirement already satisfied: websocket-client<0.57,>=0.56 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (0.56.0)Requirement already satisfied: Flask-Login<0.5,>=0.4.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (0.4.1)Requirement already satisfied: requests<3,>=2.22.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (2.24.0)Requirement already satisfied: flask<0.13,>=0.12 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (0.12.5)Requirement already satisfied: werkzeug<0.17,>=0.16 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (0.16.1)Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (0.1.5.post0)Requirement already satisfied: cachelib<0.2,>=0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (0.1.1)Requirement already satisfied: netaddr<0.8,>=0.7.19 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (0.7.20)Requirement already satisfied: rsa<5,>=4.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (4.6)Requirement already satisfied: blinker<1.5,>=1.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (1.4)Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (5.2.1)Requirement already satisfied: sentry-sdk==0.13.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (0.13.2)Requirement already satisfied: psutil<5.7,>=5.6.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (5.6.7)Requirement already satisfied: Flask-Assets<0.13,>=0.12 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (0.12)Requirement already satisfied: Flask-Babel<0.13,>=0.12 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (0.12.2)Requirement already satisfied: emoji<0.6,>=0.5.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.13) (0.5.4)Requirement already satisfied: Unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python3.7/site-packages (from awesome-slugify<1.7,>=1.6.5->OctoPrint->Bed-Visualizer==0.1.13) (0.4.21)Requirement already satisfied: MarkupSafe in /home/pi/oprint/lib/python3.7/site-packages (from Jinja2<2.9,>=2.8.1->OctoPrint->Bed-Visualizer==0.1.13) (1.1.1)Requirement already satisfied: pathtools>=0.1.1 in /home/pi/oprint/lib/python3.7/site-packages (from watchdog<0.10,>=0.9.0->OctoPrint->Bed-Visualizer==0.1.13) (0.1.2)Requirement already satisfied: argh>=0.24.1 in /home/pi/oprint/lib/python3.7/site-packages (from watchdog<0.10,>=0.9.0->OctoPrint->Bed-Visualizer==0.1.13) (0.26.2)Requirement already satisfied: setuptools>=36 in /home/pi/oprint/lib/python3.7/site-packages (from markdown<3.2,>=3.1->OctoPrint->Bed-Visualizer==0.1.13) (47.3.1)Requirement already satisfied: six in /home/pi/oprint/lib/python3.7/site-packages (from websocket-client<0.57,>=0.56->OctoPrint->Bed-Visualizer==0.1.13) (1.15.0)Requirement already satisfied: chardet<4,>=3.0.2 in /home/pi/oprint/lib/python3.7/site-packages (from requests<3,>=2.22.0->OctoPrint->Bed-Visualizer==0.1.13) (3.0.4)Requirement already satisfied: idna<3,>=2.5 in /home/pi/oprint/lib/python3.7/site-packages (from requests<3,>=2.22.0->OctoPrint->Bed-Visualizer==0.1.13) (2.9)Requirement already satisfied: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in /home/pi/oprint/lib/python3.7/site-packages (from requests<3,>=2.22.0->OctoPrint->Bed-Visualizer==0.1.13) (1.25.9)Requirement already satisfied: certifi>=2017.4.17 in /home/pi/oprint/lib/python3.7/site-packages (from requests<3,>=2.22.0->OctoPrint->Bed-Visualizer==0.1.13) (2020.6.20)Requirement already satisfied: itsdangerous>=0.21 in /home/pi/oprint/lib/python3.7/site-packages (from flask<0.13,>=0.12->OctoPrint->Bed-Visualizer==0.1.13) (1.1.0)Requirement already satisfied: pyasn1>=0.1.3 in /home/pi/oprint/lib/python3.7/site-packages (from rsa<5,>=4.0->OctoPrint->Bed-Visualizer==0.1.13) (0.4.8)Requirement already satisfied: webassets>=0.11.1 in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Assets<0.13,>=0.12->OctoPrint->Bed-Visualizer==0.1.13) (2.0)Requirement already satisfied: Babel>=2.3 in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Babel<0.13,>=0.12->OctoPrint->Bed-Visualizer==0.1.13) (2.8.0)Requirement already satisfied: pytz>=2015.7 in /home/pi/oprint/lib/python3.7/site-packages (from Babel>=2.3->Flask-Babel<0.13,>=0.12->OctoPrint->Bed-Visualizer==0.1.13) (2020.1)Building wheels for collected packages: Bed-VisualizerBuilding wheel for Bed-Visua [... cut ...] wheel for Bed-Visualizer (setup.py): finished with status 'done'Created wheel for Bed-Visualizer: filename=Bed_Visualizer-0.1.13-py3-none-any.whl size=2518827 sha256=0915b29022ff28d1e60bc4eda5d1a7877c0939d45b6858627d5fe98a5b9e515fStored in directory: /tmp/pip-ephem-wheel-cache-ca9ju3gy/wheels/75/44/35/74bcf9f68ddaab8d67a995ba5a7e3046f91b4d7169880d062eSuccessfully built Bed-VisualizerInstalling collected packages: numpy, Bed-VisualizerSuccessfully installed Bed-Visualizer-0.1.13 numpy-1.19.0Done!

jneilliii, 2020-06-26:
Check this post. 

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/224#issuecomment-614968499

sylwesterr, 2020-06-26:
Ok, that worked, thanks!!! I really do appreciate this cool plugin :) ... the issue hover is still not fully resolved, because the installation script, if numpy is not installed, downloads numpy, compiles and installs itself (v 1.19 currently, while APT installs 1.16.2) and does not install atlas dev libraries - not sure which one is more 'problematic' here, but there is no info about those two prerequisites... or I am blind and it is written on homepage or plugin info...

jneilliii, 2020-06-26:
That's the hard part, it's a dependency for numpy,  not my plugin, and apparently the build doesn't declare the proper requirements for compiling I guess. I'd rather get rid of the numpy dependency altogether, but haven't really been able to work on that yet.

sylwesterr, 2020-06-28:
Fair enough... out of curiosity - this problem I experienced is more related to numpy version or rather lack of atlas dev libraries?

jneilliii, 2020-06-28:
I think it's the atlas dev libraries not being there for python 3.

jneilliii, 2020-07-04:
Closing as duplicate to #213

closed 2020-07-04 by jneilliii