jneilliii/OctoPrint-BedLevelVisualizer #267: Plugin hangs after running

Sarithz, 2020-06-26:
I have an issue with the plugin doing its 9 point test and then the system just hangs as if it is not updating the mesh and I have to hit the cancel button.  This is running on a RPi 4 connected to a Geeetech A30M printer.  I have my GCODE for the process listed below:

G28	
M155 S30	
@BEDLEVELVISUALIZER	
G29 T	
M155 S3
M500

jneilliii, 2020-06-26:
Please enable debug logging in the plugin's settings, restart octoprint and try again. If it doesn't work upload the plugin_bedlevelvisualizer_debug.log file found in the octoprint's settings in logging section and a screenshot of all the plugin's settings on the first tab.

Sarithz, 2020-06-26:
Thank you for the fast reply.  Attached are the requested items.

[image]

[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4839371/plugin_bedlevelvisualizer_debug.log)

jneilliii, 2020-06-26:
Thanks. You are plagued with some random firmware that is reporting mesh data in translated text. It happened to someone else over [here](https://community.octoprint.org/t/bed-level-visualizer/1383/62). But I have yet to implement a fix.

Sarithz, 2020-06-26:
Yeah, I just added a new printer, a Geeetech A30M, and I think it uses its own forked version of Marlin.  I’ve gotten little to no support from Geeetech on this unit and am tempted to replace the motherboard with a BTT 1.4 turbo, but am concerned if it will work with the built in touch screen if the printer. 

Thank you,

James Barrett

jneilliii, 2020-06-26:
No worries. I just pushed an update to my latest dev branch that resolves this for you. I'm still working on it and some things are a little off with the mesh table and angle corrections, but it would still graph and function from that perspective if you want to try it out. You can install using the URL below in plugin manager if you want to give it a go and let me know.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.14.zip

Sarithz, 2020-06-27:
Thanks.  If I can help test, let me know.  I uninstalled the 1.13 and installed this one and get these errors:
[image]

jneilliii, 2020-06-27:
Huh, I wasn't getting those errors with my testing of your data. When you uninstalled did you use the option to clear data? Sometimes that corrupts the config.yaml file (should be fixed next octoprint release).

Sarithz, 2020-06-27:
Yes, I selected the Octopi option to clean up the data during the uninstall.

 

Thank you,

 

James Barrett

jneilliii, 2020-06-27:
Ok, if you get a chance and SSH to your pi and run this command while not printing it might be able to fix it. 
```
nano ~/.octoprint/config.yaml
```
look for the line `bedlevelivisualizer: null` in the plugins section. If you see it delete the line and then press `ctrl+x` to exit, `y` to save, enter to keep the same name. Then run this command to reload the updated settings.
```
sudo service octoprint restart
```

closed 2020-07-26 by jneilliii