jneilliii/OctoPrint-BedLevelVisualizer #268: Define gcode to monitor

SixFive7, 2020-07-05:
Hi,

As I understand it this plugin hooks into the octoprint "send gcodes event" to scan for an instance of the @BEDLEVELVISUALIZER gcode. Correct me if I;m wrong!
This means the slicer settings that generate the gcode sequence have to be changed from this:

```
G80
G81
```

to this:

```
G80
@BEDLEVELVISUALIZER
G81
```

for the plugin to work.
Would it be possible to allow BedLevelVisualizer to directly detect a G81 gcode? This would allow for updating the visualizer (and storing the most recent values) whenever somebody prints, without them having to adjust any slicer settings.

Kind regards

jneilliii, 2020-07-05:
You are correct, your slicer's gcode would need to be changed to
```
G80
@BEDLEVELVISUALIZER
G81
```
for start gcode, or alternatively add it to OctoPrint's GCODE Scripts print start section and remove from your slicer settings. Unfortunately, I cannot just collect data after receiving the G81 command because not everyone uses PrusaFirmware like you do, and my implementation allows for just about all firmware to be able to use it. This versatility is not something I'm willing to give up.

closed 2020-07-05 by jneilliii

SixFive7, 2020-07-05:
Hi, I completely understand the argument about versatility and a such that was not my intended recommendation. Instead, I would argue for a setting where one can define what to monitor. Be it a G80, G29, G29 T, M420 S1 V or BED_MESH_OUTPUT. Much like there is already a setting for what gcode the execute on a manual bed level action.
If you are open to this I would consider creating a pull request for the necessary changes.

jneilliii, 2020-07-05:
I'm concerned of clogging up the gcode communication layer between the printer and octoprint in this specific situation. Monitoring every gcode command versus monitoring for rarely used at commands was a performance decision. I don't want my plugin to be the cause of blobbing due to communication blocking. If you choose to submit a PR, I will review it as a possible merge, but please use the latest development branch 0.1.14 as a starting point, because there are already loads of enhancements and performance tweaks incorporated into that version. Please consider the upgrade path as well for end users and how this change will effect them.