jneilliii/OctoPrint-BedLevelVisualizer #269: Makergear M3_ID with Marlin 1.1.6.1.1?

atestani, 2020-07-06:
Does this plug-in support Makergear M3-ID (dual extruder) with Marlin 1.1.6.1.1?  The M3-ID printer comes equipped with the BLTouch probe with firmware support included.

I have the plug-installed and it will do the mesh probing but no data or graph is displayed.  I have tried various gcode but none seem to work.  Timeout is set for the default 1800 seconds.  If the plug-in should work with this firmware, what gcode should i be using?

Off topic here but may help with a solution: Another plug-in: OctoPrint-BLTouch is also installed so BLTouch controls appear on the Control tab.  Clicking any button just causes a buzzing sound from the printer...  nothing else.  The buzzing continues until I restart OctoPrint.

jneilliii, 2020-07-06:
The BLTouch plugin is mine as well. The only thing it does is send gcode commands configured in the plugin's settings. If you know the proper commands to send to control the BLTouch, I recommend you adjusting those settings. 

As for the graphing issue, that's completely dependent on the output of the leveling command. 1.1.6.1.1 is an extremely old Marlin variant, and I do not know if that sends the report data or not. If the firmware is stock, the reporting mechanism could have been stripped out to save space. So the short answer is if it doesn't report the mesh data, this plugin will not work for you.

atestani, 2020-07-06:
OK... thanks.

For the BLTouch plug-in, I;ll do what you say in terms of adjusting the
gcode to get it going.  If at the end i click the save button, can I use
that data on subsequent prints to skip the propping operation?

The Marlin version is adapted by MakerGar to run on the RamBo in the
printer.  It just got the printer so it must have the latest version of
MakerGear's variant.  I'm going to contact MakerGear about this.

Thanks again for your help and the great plug-ins you have created!

jneilliii, 2020-07-06:
> If at the end i click the save button, can I use
that data on subsequent prints to skip the propping operation?

That's another firmware dependent question. I can't say for sure. 

> The Marlin version is adapted by MakerGar to run on the RamBo in the
printer.

I bet you can find current versions of Marlin that work with this. I myself have used RamBo boards for many years and the default Marlin is able to handle it without problems. The trick is knowing what's connected to which pin on the board, which the manufacturer may have tweaked in some way, so again hard to say.

atestani, 2020-07-07:
I received an excellent response from MakerGear to my  questions about both of the plug-ins (sorry to mix the two into one issue).  They are working on Marlin and Octoprint updates.  I did learn some things I will pass along. 

For the BLTouch plug-in,  all that is required is to change "P0" to "P1" and the basic buttons all work (except alarm).  For the "Probe Bed" button, a G28 must precede the G29 or the G29 is ignored.

For the visualizer, the issue is the format of the mesh data.  Marlin 1.1.6.1.1 is outputting it this way:
_Recv: Bilinear Leveling Grid:
Recv:  = [[ 10.000, 100.000, 190.000, 10.000, 100.000, 190.000, 10.000, 100.000, 190.000]  ,[ 0.000, 0.000, 0.000, 115.000, 115.000, 115.000, 230.000, 230.000, 230.000]  ,[ -0.040, -0.005, -0.146, -0.058, -0.001, -0.123, -0.014, 0.013, -0.139]];_
 which, apparently, the plug-in is not happy with.

jneilliii, 2020-07-07:
Ok, just to make sure I understand, the output of the command in terminal when you run G28 followed by G29 is like this exactly?

```
Recv: Bilinear Leveling Grid:
Recv: = [[ 10.000, 100.000, 190.000, 10.000, 100.000, 190.000, 10.000, 100.000, 190.000] ,[ 0.000, 0.000, 0.000, 115.000, 115.000, 115.000, 230.000, 230.000, 230.000] ,[ -0.040, -0.005, -0.146, -0.058, -0.001, -0.123, -0.014, 0.013, -0.139]];
```

If so, I can adjust the plugin to be able to handle that I believe.

atestani, 2020-07-07:
Actually, I am a bit confused here as a G28 does the homing and then starts
the ABL process!  Here is what I get:

Send: G28Recv: echo:Active Extruder: 0Recv: current_position[Z_AXIS]
0.00Recv: Raise Z (before homing) to 10.00[...]Recv: echo:Active
Extruder: 0[...]Recv: echo:enqueueing "T0"Recv: echo:enqueueing "G29
P3"Recv: echo:Active Extruder: 0[...]Recv: echo:Active Extruder:
1Recv: echo:Active Extruder: 0Recv: X:-20.77 Y:-15.50 Z:11.08 E:0.00
Count X:-1664 Y:-1242 Z:11165Recv: okRecv: echo:Active Extruder:
0[...]Recv: Bilinear Leveling Grid:Recv:  = [[ 10.000, 100.000,
190.000, 10.000, 100.000, 190.000, 10.000, 100.000, 190.000]  ,[
0.000, 0.000, 0.000, 115.000, 115.000, 115.000, 230.000, 230.000,
230.000]  ,[ -0.050, 0.004, -0.150, -0.058, 0.006, -0.118, -0.023,
0.008, -0.144]];Recv: X:169.00 Y:230.00 Z:11.19 E:0.00 Count X:13537
Y:18423 Z:11165

For a G29 T alone I get this:

Send: G29 T[...]Recv: Bilinear Leveling Grid:Recv:  = [[ 10.000,
100.000, 190.000, 10.000, 100.000, 190.000, 10.000, 100.000, 190.000]
,[ 0.000, 0.000, 0.000, 115.000, 115.000, 115.000, 230.000, 230.000,
230.000]  ,[ -0.046, 0.008, -0.146, -0.051, 0.005, -0.112, -0.018,
0.014, -0.135]];Recv: X:169.00 Y:230.00 Z:11.19 E:0.00 Count X:13537
Y:18423 Z:11165

Recv: ok

I emailed MakerGear about this

jneilliii, 2020-07-07:
```
Send: G28
Recv: echo:Active Extruder: 0
Recv: current_position[Z_AXIS] 0.00
Recv: Raise Z (before homing) to 10.00[...]
Recv: echo:Active Extruder: 0[...]
Recv: echo:enqueueing "T0"
Recv: echo:enqueueing "G29 P3"
Recv: echo:Active Extruder: 0[...]
Recv: echo:Active Extruder: 1
Recv: echo:Active Extruder: 0
Recv: X:-20.77 Y:-15.50 Z:11.08 E:0.00 Count X:-1664 Y:-1242 Z:11165
Recv: ok
Recv: echo:Active Extruder: 0[...]
Recv: Bilinear Leveling Grid:Recv:  = [[ 10.000, 100.000,190.000, 10.000, 100.000, 190.000, 10.000, 100.000, 190.000]  ,[0.000, 0.000, 0.000, 115.000, 115.000, 115.000, 230.000, 230.000,230.000]  ,[ -0.050, 0.004, -0.150, -0.058, 0.006, -0.118, -0.023,0.008, -0.144]];
Recv: X:169.00 Y:230.00 Z:11.19 E:0.00 Count X:13537 Y:18423 Z:11165
```

jneilliii, 2020-07-07:
So yeah, your leveling report is very close to the scad format already supported, which is returned in multiple lines rather than just one big line like below, so I think the changes would be minimal to support it. 

```
Recv: Mesh Bed Level data:
Recv: 6 x 6 mesh. Z offset: 0.00000
Recv: Measured points:
Recv: measured_z = [
Recv:  [ +0.200, +0.093, +0.039, -0.005, -0.048, -0.072 ],
Recv:  [ +0.176, +0.063, -0.018, -0.039, -0.048, -0.040 ],
Recv:  [ +0.178, +0.059, -0.024, -0.040, -0.054, -0.047 ],
Recv:  [ +0.206, +0.078, -0.008, -0.036, -0.064, -0.082 ],
Recv:  [ +0.243, +0.098, -0.008, -0.064, -0.127, -0.164 ],
Recv:  [ +0.249, +0.124, +0.001, -0.072, -0.159, -0.247 ]
Recv: ];
Recv: ok
```

atestani, 2020-07-07:
Great, thanks...

What I am still puzzled about is why a simple G28 is "calling" a G29.  I've
emailed Makergear about that.

atestani, 2020-07-17:
Makergear  verified that G28 does spawn a G29 T call.  

In any case, are you updating your plug-in to accept the Marlin 1.1.6.1.1 output format and if so when do you expect to release it?  

Thanks!

jneilliii, 2020-07-17:
I have flagged this as a enhancement and will incorporate that report format. ETA as to when that happen is unknown at this time. I do have a full time job and do this on the side, so as time permits. As mentioned, I don't think it will take long to add this and may be able to get the changes squeezed in for the next release.

jneilliii, 2020-07-17:
Hey, quick question. Do you get a response from the command `M420 V` that includes the mesh data as well? Is it in the same format?

atestani, 2020-07-17:
Same thing with M420 V.  This was after I finished a print so the mesh data wa already available. I restarted the printer and ran G28 M420V and got the same thing.  Recall on this version of MG's Marlin calling G29 returns an error message to run G28 first.  H28 calls G29 T
```
Send: M420 V
Recv: Bilinear Leveling Grid:
Recv:  = [[ 10.000, 100.000, 190.000, 10.000, 100.000, 190.000, 10.000, 100.000, 190.000]  ,[ 0.000, 0.000, 0.000, 115.000, 115.000, 115.000, 230.000, 230.000, 230.000]  ,[ 0.294, 0.019, -0.137, -0.027, 0.018, -0.122, 0.024, 0.034, -0.132]];
Recv: echo:Bed Leveling Off
Recv: echo:Fade Height Off
Recv: ok
```

jneilliii, 2020-07-18:
I think I have a working solution in my current dev branch. If you want to test it out for me and report back that would be great. Use the URL below in plugin manager > get more > ... from URL and click Install.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.14.zip

atestani, 2020-07-18:
Quick test it isn't working  (It's late why are we working on this now?
<g>).  This is the script I am using:

M155 S30 ; reduce temperature reporting rate to reduce output pollution
@BEDLEVELVISUALIZER ; tell the plugin to watch for reported mesh
G28   ; run bilinear probing... this homes and calls G29 T
M155 S3  ; reset temperature reporting

The mesh is being created but not seen by the plug-in.  Here is the
terminal section:

Recv: Bilinear Leveling Grid:
Recv:  = [[ 10.000, 100.000, 190.000, 10.000, 100.000, 190.000, 10.000,
100.000, 190.000]  ,[ 0.000, 0.000, 0.000, 115.000, 115.000, 115.000,
230.000, 230.000, 230.000]  ,[ -0.024, -0.001, -0.155, -0.045, 0.002,
-0.137, 0.004, 0.017, -0.145]];

Anything specific you want me to try?  Would it help for you to run my
machine with a remote debugger, etc. Let me know how I can help.

jneilliii, 2020-07-18:
Is there an `ok` returned just prior to the report?

atestani, 2020-07-18:
No and Yes

With this script:

;M155 S30 ; reduce temperature reporting rate to reduce output pollution
@BEDLEVELVISUALIZER ; tell the plugin to watch for reported mesh
G28   ; run bilinear probing - homes then runs G29 P3
;M155 S3  ; reset temperature reporting

The line prior to the report is:

Recv: echo:Active Extruder: 0

However, I knew that if I issued a G29T without having homed the G29T
command was rejected.  I also know that a simple G28 spawns  a G29 P3 which
does not return ok before the report. I changed the script to have a G29T
and manually issued  G28 X Y Z  which did only the home operation (without
the G29 T being called).  The terminal output is as follows but the plug-in
is just spinning waiting for the mesh data.

end: G28 X Y Z
Recv: echo:Active Extruder: 0
Recv: current_position[Z_AXIS] 11.08
[...]
Recv: echo:Active Extruder: 0
[...]
Recv: echo:Active Extruder: 0
[...]
Recv: echo:Active Extruder: 1
Recv: echo:Active Extruder: 0
Recv: X:-20.77 Y:-15.50 Z:11.08 E:0.00 Count X:-1664 Y:-1242 Z:11165
Recv: ok
[...]
Send: @BEDLEVELVISUALIZER
Send: @BEDLEVELVISUALIZER
Send: G29 T
Recv: echo:Unknown command: "@BEDLEVELVISUALIZER"
Recv: ok
Recv: echo:Unknown command: "@BEDLEVELVISUALIZER"
Recv: ok
[...]
Recv: Bilinear Leveling Grid:
Recv:  = [[ 10.000, 100.000, 190.000, 10.000, 100.000, 190.000, 10.000,
100.000, 190.000]  ,[ 0.000, 0.000, 0.000, 115.000, 115.000, 115.000,
230.000, 230.000, 230.000]  ,[ -0.058, -0.031, -0.191, -0.046, 0.003,
-0.143, -0.023, -0.004, -0.176]];
Recv: X:169.00 Y:230.00 Z:11.22 E:0.00 Count X:13537 Y:18423 Z:11165
Recv: ok

jneilliii, 2020-07-18:
Would you please try the following in your gcode settings for me?

```
G28
@BEDLEVELVISUALIZER
M420 V
```

atestani, 2020-07-18:
Still not working.  Here is the terminal showing the mesh data was sent
twice:

Send: G28Recv: echo:Active Extruder: 0Recv: current_position[Z_AXIS]
0.00Recv: Raise Z (before homing) to 10.00[...]Recv: echo:Active
Extruder: 0[...]Recv: echo:enqueueing "T0"Recv: echo:enqueueing "G29
P3"Recv: echo:Active Extruder: 0[...]Recv: echo:Active Extruder:
1Recv: echo:Active Extruder: 0Recv: X:-20.77 Y:-15.50 Z:11.08 E:0.00
Count X:-1664 Y:-1242 Z:11165Recv: okSend: @BEDLEVELVISUALIZERSend:
M420 VRecv: echo:Active Extruder: 0[...]Recv: Bilinear Leveling
Grid:Recv:  = [[ 10.000, 100.000, 190.000, 10.000, 100.000, 190.000,
10.000, 100.000, 190.000]  ,[ 0.000, 0.000, 0.000, 115.000, 115.000,
115.000, 230.000, 230.000, 230.000]  ,[ -0.054, -0.028, -0.188,
-0.047, 0.005, -0.138, -0.023, -0.001, -0.175]];Recv: X:169.00
Y:230.00 Z:11.21 E:0.00 Count X:13537 Y:18423 Z:11165Recv:
echo:Unknown command: "@BEDLEVELVISUALIZER"Recv: ok[...]Recv: Bilinear
Leveling Grid:Recv:  = [[ 10.000, 100.000, 190.000, 10.000, 100.000,
190.000, 10.000, 100.000, 190.000]  ,[ 0.000, 0.000, 0.000, 115.000,
115.000, 115.000, 230.000, 230.000, 230.000]  ,[ -0.054, -0.028,
-0.188, -0.047, 0.005, -0.138, -0.023, -0.001, -0.175]];Recv: echo:Bed
Leveling OnRecv: echo:Fade Height OffRecv: ok[...]

jneilliii, 2020-07-18:
```
Send: G28Recv: echo:Active Extruder: 0
Recv: current_position[Z_AXIS]0.00
Recv: Raise Z (before homing) to 10.00[...]
Recv: echo:Active Extruder: 0[...]
Recv: echo:enqueueing "T0"
Recv: echo:enqueueing "G29 P3"
Recv: echo:Active Extruder: 0[...]
Recv: echo:Active Extruder:1
Recv: echo:Active Extruder: 0
Recv: X:-20.77 Y:-15.50 Z:11.08 E:0.00 Count X:-1664 Y:-1242 Z:11165
Recv: ok
Send: @BEDLEVELVISUALIZER
Send: M420 V
Recv: echo:Active Extruder: 0[...]
Recv: Bilinear Leveling Grid:
Recv:  = [[ 10.000, 100.000, 190.000, 10.000, 100.000, 190.000,10.000, 100.000, 190.000]  ,[ 0.000, 0.000, 0.000, 115.000, 115.000,115.000, 230.000, 230.000, 230.000]  ,[ -0.054, -0.028, -0.188,-0.047, 0.005, -0.138, -0.023, -0.001, -0.175]];
Recv: X:169.00 Y:230.00 Z:11.21 E:0.00 Count X:13537 Y:18423 Z:11165
Recv: echo:Unknown command: "@BEDLEVELVISUALIZER"
Recv: ok[...]
Recv: Bilinear Leveling Grid:
Recv:  = [[ 10.000, 100.000, 190.000, 10.000, 100.000,190.000, 10.000, 100.000, 190.000]  ,[ 0.000, 0.000, 0.000, 115.000,115.000, 115.000, 230.000, 230.000, 230.000]  ,[ -0.054, -0.028,-0.188, -0.047, 0.005, -0.138, -0.023, -0.001, -0.175]];
Recv: echo:Bed Leveling On
Recv: echo:Fade Height Off
Recv: ok[...]
```

Ok, I think I'm going to have to get your full debug log. Please enable debug logging in the plugin's settings, restart octoprint and try a mesh update. Then grab the file `plugin_bedlevelvisualizer_debug.log` from OctoPrint's logging section in settings and drag it here into a comment. From that I should be able to accurately re-create the pattern that your printer is reporting, because I think something might be getting jumbled during email replies.

atestani, 2020-07-18:
OK... I set the date/time on the printer RPi so of times look weird this is why.  Debugging in the plugin has always been enabled but the active log looks very strange as it only shows me cancelling the mesh acquisition.  I am attaching it as well as a past log that may help as it does show some mesh data
[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4942025/plugin_bedlevelvisualizer_debug.log)
[plugin_bedlevelvisualizer_debug2020-07-07.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4942026/plugin_bedlevelvisualizer_debug2020-07-07.log)

jneilliii, 2020-07-18:
Weird. the way I'm debugging this on my side seems to be working so I'm a little confused unless it's the unknown command followed by ok in that communication. I'm going to try to add that to my virtual report and see if it make a difference here. The below was from what you had sent previously.

[image]

atestani, 2020-07-19:
Anything I can do to help?  Is your virtual report generator something we MakerGear users can can use until MG updates their old Marlin firmware?  If you've got the parsing for the way the report data is formatted, that would be a perfectly good solution and spare you from spending any more time to make this work!

jneilliii, 2020-07-19:
Unfortunately, it only works with the virtual printer in OctoPrint for development. I did just push another update to the 0.1.4 branch linked below that tweaked a couple of things and might fix the problem.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.14.zip

atestani, 2020-07-19:
Still, no go.  I noticed that with some scripts there wasn't an "ok" preceding the mesh report so I tried a bunch of different scripts and got some with the "ok" present.  I  found some other things in this Marlin version like:
 - G29 without the T still outputs a mesh report
 - G29 doesn't respect the T parameter, e.g. G29 T1
 - M420 does not respect the T parameter
 - G28 X Y Z will home as expected but G28 alone calls G29 P3 and outputs a mesh report (known previously)

I am attaching 8 terminal logs in case you are interested.  You can see the various commands from the script I used.

However, while I sincerely appreciate your efforts to make this work, IMHO, I don't think it is worth the time you are spending on it given the age of this Marlin version.   At this point, you may be doing it because it is simply bugging you ( I do lots of code myself and know about this "problem") so I will continue to run any tests or provide data as long as you want.

I understand about the virtual report/printer, etc. but here is another (possibly bad) idea.  Do you have some Python and/or javascript/Plotly code that can be put together to allow the manual entry of the mesh data to get the visualization?  That would be great for the current issue and anyone else having similar problems. 

Thanks again.

[run 1.txt](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4943815/run.1.txt)
[run 2.txt](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4943816/run.2.txt)
[run 3.txt](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4943817/run.3.txt)
[run 4.txt](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4943818/run.4.txt)
[run 5.txt](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4943812/run.5.txt)
[run 6.txt](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4943813/run.6.txt)
[run 7.txt](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4943814/run.7.txt)
[run 8.txt](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4943837/run.8.txt)

jneilliii, 2020-07-19:
> Do you have some Python and/or javascript/Plotly code that can be put together to allow the manual entry of the mesh data to get the visualization? That would be great for the current issue and anyone else having similar problems.

I don't. What I typically do is re-create the communication with the virtual printer, for example your communication is [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/0.1.14/virtual_level_report_old_marlin_makergear.gcode). I was using the to mimic your printer, but apparently for some reason it's not directly comparing to real world.

jneilliii, 2020-07-19:
> M420 does not respect the T parameter

M420 has a V parameter typically.

atestani, 2020-07-19:
Yes, but it also has a T parameter
https://marlinfw.org/docs/gcode/M420.html

atestani, 2020-07-19:
Understood about the javascript, etc..

I looked at what your simulator is outputting and it is way different than
what is happening on my printer.  Below is what I get with the same script
you are using.  Notice there are two mesh outputs, one preceded by an "ok"
and not the other.  The G29 that G28 calls does has the ok but the M420 V
does not.  I tried it with putting @@BEDLEVELVISUALIZER but it still didn't
work. I also tried without the M420 V and then there is a line of
output after the OK.

So obviously you have the parsing of the differently formatted mesh data
working but it is getting the plugin to grab the data that's the problem.
Do you want to share exactly how you do that and maybe I can find a script
sequence to make it work?

Send: G28Send: @BEDLEVELVISUALIZERSend: M420 VRecv: echo:Active
Extruder: 0Recv: current_position[Z_AXIS] 11.08[...]Recv: echo:Active
Extruder: 0[...]Recv: echo:enqueueing "T0"Recv: echo:Active Extruder:
0[...]Recv: echo:Active Extruder: 1Recv: echo:Active Extruder: 0Recv:
X:-20.77 Y:-15.50 Z:11.08 E:0.00 Count X:-1664 Y:-1242 Z:11165Recv:
okRecv: echo:enqueueing "G29 P3"Recv: echo:Unknown command:
"@BEDLEVELVISUALIZER"Recv: okRecv: Bilinear Leveling Grid:Recv:  = [[
10.000, 100.000, 190.000, 10.000, 100.000, 190.000, 10.000, 100.000,
190.000]  ,[ 0.000, 0.000, 0.000, 115.000, 115.000, 115.000, 230.000,
230.000, 230.000]  ,[ -0.060, -0.025, -0.178, -0.050, 0.010, -0.129,
-0.027, 0.007, -0.156]];Recv: echo:Bed Leveling OffRecv: echo:Fade
Height OffRecv: okRecv: echo:Active Extruder: 0[...]Recv: Bilinear
Leveling Grid:Recv:  = [[ 10.000, 100.000, 190.000, 10.000, 100.000,
190.000, 10.000, 100.000, 190.000]  ,[ 0.000, 0.000, 0.000, 115.000,
115.000, 115.000, 230.000, 230.000, 230.000]  ,[ -0.061, -0.029,
-0.187, -0.048, 0.003, -0.138, -0.027, 0.002, -0.168]];Recv: X:169.00
Y:230.00 Z:11.21 E:0.00 Count X:13537 Y:18423 Z:11165[...]

jneilliii, 2020-07-19:
Ok, so the current workflow is that the plugin monitors the @ commands being sent to your printer. If it detects the custom command `@BEDLEVELVISUALIZER` it flags a boolean True to instruct the gcode receiving hook to start monitoring the serial communication in this [callback](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/d0540a4224ab49802d72fa601031944cf62bb30b/octoprint_bedlevelvisualizer/__init__.py#L139). Then as lines are received the gcode processing hook compares those received lines against several different regular expressions, in your case the one that I added [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/d0540a4224ab49802d72fa601031944cf62bb30b/octoprint_bedlevelvisualizer/__init__.py#L43). If that match occurs then it is recorded [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/d0540a4224ab49802d72fa601031944cf62bb30b/octoprint_bedlevelvisualizer/__init__.py#L200-L206). That data is converted into a list using json.loads, and then when the next ok comes through the final processing occurs and the data is sent back to the front end [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/d0540a4224ab49802d72fa601031944cf62bb30b/octoprint_bedlevelvisualizer/__init__.py#L220).

atestani, 2020-07-19:
If the following is a dumb idea no need to provide any explanation, just
say it is a dumb idea!

Would it help to change this line
if ("ok" in line or (self.repetier_firmware and "T:" in line)) and len(self.
mesh) > 0:
to:
if ("ok" in line or (self.repetier_firmware and "T:" in line)) or (self.
regex_makergear and "Grid:" in line)) and len(self.mesh) > 0:

(or whatever the flag name is for the makergear firmware... I just guessed
it was self.regex.makergear)

My thinking is that the line *"Recv: Bilinear Leveling Grid:"  **follows*
the *"Recv: ok"*

i.e.  from the terminal:

Recv: ok
[...]
Recv: Bilinear Leveling Grid:
Recv:  = [[ 10.000, 100.000, 190.000, 10.000, 100.000, 190.000, 10.000,
100.000, 190.000]  ,[ 0.000, 0.000, 0.000, 115.000, 115.000, 115.000,
230.000, 230.000, 230.000]  ,[ -0.050, -0.017, -0.171, -0.050, 0.003,
-0.136, -0.020, 0.003, -0.161]];

jneilliii, 2020-07-19:
The mesh wouldn't be collected yet at that point in time I don't think so the len condition would fail anyway.

jneilliii, 2021-01-13:
This issue may be resolved with my most recent release candidate of the plugin. It accounts for numbers that start with one more than one digit before the decimal point, which I think might have been the underlying issue here. If you want to try it out, you can caopy/paste the URL below into plugin manager > get more > ...from URL and clicking Install.

```
https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/1.0.1rc1.zip
```

atestani, 2021-01-13:
Thanks for the heads up.  Unfortunately it still isn't working.  I'll
provide some data below.  BTW, I have contacted MakerGear multiple times
(including management) about getting the firmware and Octoprint fork
updated and I keep getting: "it is planned but we don't know when".  So we
MakerGear users are still working with prehistoric versions of both.

Here is the gcode (which is what you had me use the last time we worked on
this):

G28
@BEDLEVELVISUALIZER
M420 V

Here is the terminal output:

Recv: Bilinear Leveling Grid:
Recv:  = [[ 10.000, 100.000, 190.000, 10.000, 100.000, 190.000,
10.000, 100.000, 190.000]  ,[ 0.000, 0.000, 0.000, 115.000, 115.000,
115.000, 230.000, 230.000, 230.000]  ,[ -0.059, -0.057, -0.191,
-0.040, 0.003, -0.126, -0.030, -0.031, -0.175]];
Recv: X:169.00 Y:230.00 Z:11.22 E:0.00 Count X:13537 Y:18423 Z:11165

I have logging enabled in the plugin settings but the log doesn't say
much except that I cancelled which I did after the "retrieving"
spinner ran for a long time.  I started by clicking the "Update Mesh"
button to get the mesh and G28 ran as normal.  However, the process
never returned so I finally clicked the Cancel button. I deleted all
logs and ran the plugin again, this time no log at all.

jneilliii, 2021-01-13:
Ok, thanks. I think I see now that my previous changes wouldn't resolve this. I need to create a custom regex parser for this output. Best I can tell from this format, it's an array containing the x points, y points, and z offsets. 

```
= [[ 10.000, 100.000, 190.000, 10.000, 100.000, 190.000,
10.000, 100.000, 190.000]  ,[ 0.000, 0.000, 0.000, 115.000, 115.000,
115.000, 230.000, 230.000, 230.000]  ,[ -0.059, -0.057, -0.191,
-0.040, 0.003, -0.126, -0.030, -0.031, -0.175]]
```

atestani, 2021-01-13:
Let me know if there is anything I can do to help!

Reminder:  You had a version of the plugin working in your simulator with
MakerGear's old Marlin version.  It wouldn't work in my actual
setup, however.

Thanks for continuing to work on this.  There have been several times where
I wanted to use it but obviously couldn't.

atestani, 2021-03-02:
Does this mean I should try it again with my prehistoric MakerGear firmware
and Octoprint fork?

jneilliii, 2021-03-02:
Oh sorry, this got closed inadvertently.

atestani, 2021-03-02:
No problem... but you did get my hopes up! <g>

jneilliii, 2021-03-19:
Actually looking back at the code, this might be fixed. Have you tried with 1.0.1 yet?

atestani, 2021-03-19:
I uninstalled and reinstalled (had 101rc5) and it still doesn't work. If I
click on Update Mesh Now, it goes through the calibration and then simply
spins at "Please wait...." and I have to cancel it.

Here is the script:
@BEDLEVELVISUALIZER
G28
@BEDLEVELVISUALIZER
M420 V

and this is the result:
Recv: Bilinear Leveling Grid:Recv: = [[ 10.000, 100.000, 190.000, 10.000,
100.000, 190.000, 10.000, 100.000, 190.000] ,[ 0.000, 0.000, 0.000,
115.000, 115.000, 115.000, 230.000, 230.000, 230.000] ,[ -0.091, -0.051,
-0.151, -0.065, 0.002, -0.093, -0.037, -0.026, -0.155]];
Recv: X:169.00 Y:230.00 Z:11.21 E:0.00 Count X:13537 Y:18423 Z:11165

 I reran with just G28 and got the same thing.

PR #398 (2021-02-28): 1.0.1
**Added**
* optional timeout override to @BEDLEVELVISUALIZER command, resolves #314
* settings button on tab for quick access, resolves #349

**Updated**
* README.md adding note that auto bed leveling must be possible on printer, resolves #348

**Fixed**
* allow for probe points that contain more than single digits, resolves #352, resolves #269
* resolve issues with circular beds