jneilliii/OctoPrint-BedLevelVisualizer #270: [Feature Request] Support for Delta Printers

susisstrolch, 2020-07-14:
With Delta printers (round bed) the outer corners are pretty meaningless 'cause they don't exist.
So you could mask the data by the round bed, only showing real existing (interpolated) Z-Points.

jneilliii, 2020-07-14:
If you set the origin center option it will do just that.

jneilliii, 2020-07-17:
That's not how it worked when I tested it. Is your octoprint settings for your printer profile set to round bed? Enable debug logging in the plugin's settings restart octoprint and update mesh. Send me the plugin_bedlevelvisualizer_debug.log file from the logging section of octoprint.

susisstrolch, 2020-07-17:
Ok, here we go...
(btw. I'm using version Bed Visualizer (0.1.13) )

```
[2020-07-17 15:31:55,472] DEBUG: mesh collection started
[2020-07-17 15:31:55,481] DEBUG: Bilinear Leveling Grid:
[2020-07-17 15:31:55,496] DEBUG: 0      1      2      3      4      5      6      7      8
[2020-07-17 15:31:55,499] DEBUG: 0 +0.077 +0.019 -0.170 -0.210 -0.315 -0.204 -0.317 -0.334 -0.261
[2020-07-17 15:31:55,502] DEBUG: [u'+0.077', u'+0.019', u'-0.170', u'-0.210', u'-0.315', u'-0.204', u'-0.317', u'-0.334', u'-0.261']
[2020-07-17 15:31:55,506] DEBUG: 1 -0.033 +0.064 -0.181 -0.273 -0.315 -0.359 -0.338 -0.143 -0.104
[2020-07-17 15:31:55,508] DEBUG: [u'-0.033', u'+0.064', u'-0.181', u'-0.273', u'-0.315', u'-0.359', u'-0.338', u'-0.143', u'-0.104']
[2020-07-17 15:31:55,512] DEBUG: 2 -0.049 -0.214 -0.127 -0.187 -0.177 -0.214 -0.291 -1.258 -0.013
[2020-07-17 15:31:55,513] DEBUG: [u'-0.049', u'-0.214', u'-0.127', u'-0.187', u'-0.177', u'-0.214', u'-0.291', u'-1.258', u'-0.013']
[2020-07-17 15:31:55,517] DEBUG: 3 +0.038 -0.184 -0.198 -0.195 -0.189 -0.197 -0.205 -0.255 +0.056
[2020-07-17 15:31:55,518] DEBUG: [u'+0.038', u'-0.184', u'-0.198', u'-0.195', u'-0.189', u'-0.197', u'-0.205', u'-0.255', u'+0.056']
[2020-07-17 15:31:55,521] DEBUG: 4 -0.169 -0.177 -0.151 -0.151 -0.117 -0.112 -0.194 -0.188 -0.164
[2020-07-17 15:31:55,523] DEBUG: [u'-0.169', u'-0.177', u'-0.151', u'-0.151', u'-0.117', u'-0.112', u'-0.194', u'-0.188', u'-0.164']
[2020-07-17 15:31:55,526] DEBUG: 5 +0.045 -0.181 -0.205 -0.178 -0.160 -0.160 -0.172 -0.189 +0.053
[2020-07-17 15:31:55,528] DEBUG: [u'+0.045', u'-0.181', u'-0.205', u'-0.178', u'-0.160', u'-0.160', u'-0.172', u'-0.189', u'+0.053']
[2020-07-17 15:31:55,530] DEBUG: 6 +0.048 -0.141 -0.149 -0.187 -0.185 -0.202 -0.218 -0.238 +0.012
[2020-07-17 15:31:55,532] DEBUG: [u'+0.048', u'-0.141', u'-0.149', u'-0.187', u'-0.185', u'-0.202', u'-0.218', u'-0.238', u'+0.012']
[2020-07-17 15:31:55,535] DEBUG: 7 +0.095 +0.082 -0.153 -0.200 -0.218 -0.240 -0.276 -0.089 -0.073
[2020-07-17 15:31:55,536] DEBUG: [u'+0.095', u'+0.082', u'-0.153', u'-0.200', u'-0.218', u'-0.240', u'-0.276', u'-0.089', u'-0.073']
[2020-07-17 15:31:55,539] DEBUG: 8 +0.117 +0.053 +0.012 +0.034 -0.150 -0.010 -0.108 -0.171 -0.183
[2020-07-17 15:31:55,541] DEBUG: [u'+0.117', u'+0.053', u'+0.012', u'+0.034', u'-0.150', u'-0.010', u'-0.108', u'-0.171', u'-0.183']
[2020-07-17 15:31:55,543] DEBUG: 
[2020-07-17 15:31:55,545] DEBUG: echo:Bed Leveling OFF
[2020-07-17 15:31:55,548] DEBUG: echo:Fade Height OFF
[2020-07-17 15:31:55,550] DEBUG: ok
[2020-07-17 15:31:55,551] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 200.0, 'x_min': 0, 'type': 'circular', 'y_max': 200.0, 'z_max': 245.0}
[2020-07-17 15:31:55,552] DEBUG: stopping mesh collection
[2020-07-17 15:31:55,554] DEBUG: using relative offsets
[2020-07-17 15:31:55,555] DEBUG: using center origin
[2020-07-17 15:31:55,557] DEBUG: [[0.194, 1.136, -0.053000000000000005, -0.09299999999999999, -0.198, -0.08699999999999998, -0.2, -0.21700000000000003, -0.14400000000000002], [0.084, 0.181, -0.06399999999999999, -0.15600000000000003, -0.198, -0.242, -0.22100000000000003, -0.02599999999999998, 0.013000000000000012], [0.068, -0.09699999999999999, -0.009999999999999995, -0.06999999999999999, -0.059999999999999984, -0.09699999999999999, -0.174, -1.141, 0.10400000000000001], [0.155, -0.06699999999999999, -0.081, -0.078, -0.072, -0.08, -0.08799999999999998, -0.138, 0.17300000000000001], [-0.052000000000000005, -0.059999999999999984, -0.03399999999999999, -0.03399999999999999, 0.0, 0.0050000000000000044, -0.077, -0.071, -0.047], [0.162, -0.06399999999999999, -0.08799999999999998, -0.060999999999999985, -0.043, -0.043, -0.05499999999999998, -0.072, 0.17], [0.165, -0.02399999999999998, -0.03199999999999999, -0.06999999999999999, -0.06799999999999999, -0.085, -0.10099999999999999, -0.12099999999999998, 0.129], [0.21200000000000002, 0.199, -0.03599999999999999, -0.083, -0.10099999999999999, -0.12299999999999998, -0.15900000000000003, 0.02800000000000001, 0.04400000000000001], [0.234, 0.17, 0.129, 0.15100000000000002, -0.03299999999999999, 0.10700000000000001, 0.009000000000000008, -0.054000000000000006, -0.06599999999999999]]
```

susisstrolch, 2020-07-17:
And here's my printer definition:
```
pi@hummingboard:~/.octoprint/printerProfiles$ cat vertex_delta.profile 
axes:
  e:
    inverted: false
    speed: 1500
  x:
    inverted: false
    speed: 3000
  y:
    inverted: false
    speed: 3000
  z:
    inverted: false
    speed: 3000
color: default
extruder:
  count: 1
  nozzleDiameter: 0.4
  offsets:
  - - 0.0
    - 0.0
  sharedNozzle: false
heatedBed: true
heatedChamber: false
id: vertex_delta
model: K8800
name: Vertex Delta
volume:
  custom_box: false
  depth: 200.0
  formFactor: circular
  height: 245.0
  origin: center
  width: 200.0
```

jneilliii, 2020-07-17:
Ok, that's a new one to me...the previous Delta reports that I have been given left out the data that was outside the bounds of the bed diameter.

jneilliii, 2020-07-17:
What firmware and version are you using?

susisstrolch, 2020-07-17:
Im running the most current Marlin bugfix-2.0x.
I'm using bilinear leveling. In contrast to UBL leveling it extrapolates (as you can see) to a square bed. 
Therefore I thought it would be helpfull to have this correction in visualiser.

jneilliii, 2020-07-17:
Perfect, I'll see what I can figure out with the plotly js library or possibly with the mesh data itself to overlay the bed diameter and remove the extraneous data. Haven't done much work with this so will probably take a while for me to get my head around the math and get it to work with plotly.

jneilliii, 2020-07-17:
Quick question, do you have extrapolation turned on in the firmware?

susisstrolch, 2020-07-17:
No, currently not - but just going to recompile with extrapolation to see whats comming around...

jneilliii, 2020-07-17:
Ok, my understanding from some delta user's on the Discord is that the extrapolation would extend past the probe points, which is why I was wondering if you had it enabled.

susisstrolch, 2020-07-17:
Just tried - can't use it - controller has insufficient RAM to use that option.

jneilliii, 2020-07-19:
I just pushed an update to my current development branch that is a first step in the right direction. May need to tweak it a little more to get some more accuracy because I was getting some unexpected results in the graphing side, even though the data points seemed to be extracted correctly. If you want to test it out that would be great. Use the URL below in plugin manager > Get More <...from URL and click Install button.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.14.zip

susisstrolch, 2020-07-19:
Great!
I‘ll give it a try and will report later this day.

susisstrolch, 2020-07-20:
Here are my findings:
- the archive still has the old version number, so it wasn't updated and I had to do a remove/reinstall
(which is only cosmetic for this test)
- by setting the option "Use Relative Z Offsets" the rendering fails completely, because the translated map contains "NaN" which won't render
- the map is shown as a "cross"
Attached my latest debug-log...
[image]
[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4945927/plugin_bedlevelvisualizer_debug.log)

jneilliii, 2020-07-20:
Yeah, this was what I was seeing, but like I mentioned, I'm not really sure why because the mesh data in the settings section shows those outer points in the center of each side. Maybe some math error on the javascript side during graphing of center origin. At least we're a step in the right direction.

As for installing over the same version, that's how I develop to prevent testers from not getting the latest when it comes to official release. You can install over the existing install without issues, you just may have to perform a force refresh in your browser after OctoPrint restarts to reload the cache for the UI/javascript side stuff.

susisstrolch, 2020-07-20:
Did a remove all/install, 'cause there's no valuable data to loose.

jneilliii, 2020-07-20:
Makes sense. There is a bug that happens occasionally when you do that which will break the config.yaml though and leave the plugin's section in config.yaml set to `null`. This basically breaks the plugin. Luckily it didn't happen with you here and the bug is slated to be fixed in the next release of OctoPrint which is currently in Release Candidate 3 now. I'll try to do some additional debugging this weekend to try and figure out why those most outer points are getting lost in the graph.

[image]

darconeous, 2020-07-27:
> Yeah, this was what I was seeing, but like I mentioned, I'm not really sure why because the mesh data in the settings section shows those outer points in the center of each side. Maybe some math [...]

The reason is that those edge values are surrounded by NaNs. The interpolated values are effectively 1-dimensional, so it seems like Plotly doesn't like rendering 1D slivers on a 2D graph.

Ideally we would display the out-of-radius values and just superimpose a disc, or perhaps draw a circle over the data. Or maybe even just have all of the data outside of the radius be semi-transparent. I realize it's not quite that simple to implement, but it would be really cool.

darconeous, 2020-07-27:
It would also be cool if it was possible to show the tower locations, to make it easier to keep track of the orientation of the graph.

jneilliii, 2020-07-27:
Tower locations will be nearly impossible I fear. Good point about the single point for this outer values. Is it possible to probe or report on odd number grid? That in theory would put two points on that outer radius.

susisstrolch, 2020-07-29:
You can probe on even or odd numbers. However, for some bed leveling types (like bilinear) you must recompile...

susisstrolch, 2020-07-29:
> Tower locations will be nearly impossible

Hmm, by configuring the xy locations of the pylons and plotting a simply circle / triangular at those  point?

susisstrolch, 2020-07-29:
slightly offtopic: Is it possible to upload a mesh (from PC) instead of reading from printer?

jneilliii, 2020-07-29:
> Hmm, by configuring the xy locations of the pylons and plotting a simply circle / triangular at those point?

I'd be happy to accept a pull request with the changes necessary to do this for every possible printer out there, I3 style that has 2 posts partially offset, core xy with 4 posts, 3 posts for Delta, etc.

jneilliii, 2020-07-29:
> slightly offtopic: Is it possible to upload a mesh (from PC) instead of reading from printer?

Not exactly, For me to mimic the report I have to use the virtual printer of octoprint, and create gcodes that send the data through as if it's your printer. You'll see all the examples I have in the base of this repository.

susisstrolch, 2020-10-14:
Just tested with the 0.1.15 release.
It still has the "NaN" problem with round bed and relative offset - not showing anything at all but a rotating indicator...

jneilliii, 2020-10-14:
Can you try it with the rc branch using the URL below to install. Should be version 1.0.0rc1 after install. There were some tweaks in that to resolve some other issues that might actually help in this one.
```
https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/rc.zip
```

susisstrolch, 2020-10-15:
Nope, no fun...
Did a "G29 P0; M500" to create an empty (all-Zero) printbed.
After that the bedvisualizer waits until timeout.

jneilliii, 2020-10-15:
What about if you downgrade to 0.1.14 version? I have some people reporting that the older version is working for them but the new version is timing out like this for rectangular reports.

jneilliii, 2021-02-27:
Thanks to a recent PR, this may be fixed in the newest release candidate if you want to try it out. You would just have to change your release channel for the plugin in software update settings and update when prompted.

closed 2022-01-01 by jneilliii