jneilliii/OctoPrint-BedLevelVisualizer #272: Probe a single point

cmontgo, 2020-07-16:
Is there a way to pull code from the terminal to interact with the custom buttons? I've set up a button to use G30 and probe a point, after selecting it from the visualizer, but the point locations that come back in terminal are for the "wrong point" (probe offset) and I have to manually pull that information from the terminal. Thoughts?

jneilliii, 2020-07-16:
Currently no, there is no way to do that unfortunately. What you're describing is similar to the custom controls implementation of reading data from terminal using regular expression extraction, but even then I don't think that is able to update parameters for buttons, but I may be wrong. I doubt I'll get to this any time soon, but will flag it for a potential future update.

cmontgo, 2020-07-16:
I was thinking maybe it would work the same way as the "Get Mesh" button. Granted I have no idea how you're doing that.

jneilliii, 2020-07-16:
Yeah, it's very similar. The only difference I think is that the Get Mesh button is instructing the plugin to start recording what's coming through on the terminal tab and comparing those lines against a regular expression until an OK is received.

cmontgo, 2020-07-16:
So this would be very similar. There's a return of Bed X: 0.00 Y: 110.00 Z: 2.00 (or similar, sorry printing and can't test), so we just need the piece after the Z, since I've already used the visualizer to select that point and could use that information. Then I would use G29 W1 X## Y## Z## to set the new point. If we wanted to use the X and Y coordinates it spits back, we would need to compensate for the probe offset as well.

cmontgo, 2020-07-17:
This should work. Will report back. https://github.com/Salandora/octoprint-customControl/issues/7

jneilliii, 2020-07-17:
That plugin has a pretty big bug. Once you get the buttons working the way you want disable it in plugin manager or your buttons will get wiped out. Once disabled your buttons will still work because that's a core octoprint feature, and was what I was referring to before.

cmontgo, 2020-07-17:
Good to know, thanks. I was kind of hoping if I could get it to work, we could port the read option over to the visualizer, is that possible?

jneilliii, 2020-07-17:
Anything is possible with enough time.

cmontgo, 2020-07-17:
Looks like even disabling the custom controls, before a restart wipes them for me. I guess I'll just watch the terminal for now. So in the end, what I need actually is a "Use Output" command, like the cons tom controls plugin, that would solve this issue.

jneilliii, 2020-07-17:
You could try my patched version by using the URL below in plugin manager > Get More > at the bottom.

https://github.com/jneilliii/octoprint-customControl/archive/devel.zip

jneilliii, 2020-07-17:
or add them manually to config.yaml...

https://docs.octoprint.org/en/master/features/custom_controls.html#sec-features-custom-controls

hansvanlin, 2022-06-30:
This is how I made the custom buttons and get the data from the terminal;

[image]
[image]

hansvanlin, 2022-06-30:
I have tried that but could not figure out how to do it. The manual has not enough examples for me to see how it works.

jneilliii, 2022-06-30:
Here's an example of z-offset: https://community.octoprint.org/t/custom-control-z-offset/4412/4?u=jneilliii

hansvanlin, 2022-06-30:
Thank you I will try. Not sure what    default: ' '   and     width: '5'   does.  Last time I crashed octoprint server with it.
But found that youre plugin also has the add button options. Is it also possible to use the regex function there?

[image]

jneilliii, 2022-06-30:
> Is it also possible to use the regex function there?

No, that's what #315 was requesting.

hansvanlin, 2022-06-30:
Indeed...it's getting late. Thanks.