jneilliii/OctoPrint-BedLevelVisualizer #273: Probe arbitrary grid via G30 or M48

dracode, 2020-07-17:
Hello,

I'm running Marlin with bilinear leveling turned on.  When I run a "G29" to get mesh data for the plugin, it can only probe my predefined 3x3 grid.  There doesn't appear to be any way to dynamically alter that -- I'd have to recompile the firmware to change the probe grid.

However, I'd like to get a grid for the visualizer at a higher resolution to see how lumpy my bed really is.

Looking in the Marlin docs, I see the  [G30](https://marlinfw.org/docs/gcode/G030.html) and [M48](https://marlinfw.org/docs/gcode/M048.html) commands will run a probe (or several) and report the Z height at the given location.  With this method, a user could generate g-code to probe the bed at any arbitrary resolution.

Would it be possible to have the plugin harvest the data from one or both of these commands to create a visualization?  (Or is there a simpler way to accomplish this goal that I've overlooked?)

jneilliii, 2020-07-17:
Doesn't G29 have a P parameter that tells the firmware to probe those amount of points?

jneilliii, 2020-07-18:
Additional information...

https://reprap.org/wiki/G-code#G29_Auto_Bed_Leveling_.28Marlin_-_MK4duo.29

dracode, 2020-07-18:
Yes and no.  Some ABL modes do -- such as the Unified Bed Leveling mode mentioned in that link.  However, the UBL mode takes up a huge amount of firmware, so it won't fit in the stock Creality Ender boards.  In the [Bilinear ABL mode](https://marlinfw.org/docs/gcode/G029-abl-bilinear.html) that I'm using, there is no P parameter.

jneilliii, 2020-07-18:
Yeah, I see now on the Marlin site that only seems to work for plain linear leveling, not bilinear. The P in UBL is an entirely different thing called phases and I'm familiar with that as it's what I use.

https://marlinfw.org/docs/gcode/G029-abl-linear.html

I'm not sure I'm willing to spend the time on this feature though because it seems very special use case and wouldn't be utilized by a larger audience. Typically people that want a more detailed grid will do like you said and just set it in firmware and leave it there at the higher density.

dracode, 2020-07-19:
Fair enough.  Thanks!

closed 2020-07-19 by jneilliii