jneilliii/OctoPrint-BedLevelVisualizer #275: Z axis auto limit

pyrokidd89, 2020-07-27:
in the recent update the Z axes limits changed to having set values. Can you add an auto function where the Z axes max is the max from the bed level mesh and the min is the min from the bed mesh?

jneilliii, 2020-07-27:
You are correct, the new z-axis limits the overall height. It was another request. IIRC, what you are asking for is the way it was previously?

pyrokidd89, 2020-07-27:
Yes you are correct I preferred the previous functionality. Maybe just add check box for auto axis or maybe allow the use of "auto,auto" in the new x axis limit box.

jneilliii, 2020-07-27:
Yeah, can't make everybody happy I suppose. I can definitely see about making the suggested change using auto,auto, Relative to this you would also have to change your colorscale to one of the standard named versions. I think the default is `RdBu`.

jneilliii, 2020-07-31:
Fixed in the above commit, if you enter auto in the z limits it will default to the previous autoscaling on the colorscale. To be released with the next 0.1.15 version.

commit fb774afd (2020-07-31): make colorscale limits allow option auto for the old method of coloring, resolves #275
commit 3fb614e6 (2020-10-03): 0.1.15 (#304)

fix axis label colors, resolves #278
make colorscale limits allow option auto for the old method of coloring, resolves #275
add responsive option to plotly config, resolves #274
first stab at SD card support, resolves #223
fix little issue with reactive option where graph was getting overlaid on top of correction table
change home button press to a different strategy, resolves #265
update plotly to version 1.54.7
add option to download snapshots on graph rendering completion, #239
remove custom home button in preference of resetCameraLastSave3d, resolves #265
numpy versioning fo [... cut]
commit 7565c066 (2020-10-03): 0.1.15 (#304)

fix axis label colors, resolves #278
make colorscale limits allow option auto for the old method of coloring, resolves #275
add responsive option to plotly config, resolves #274
first stab at SD card support, resolves #223
fix little issue with reactive option where graph was getting overlaid on top of correction table
change home button press to a different strategy, resolves #265
update plotly to version 1.54.7
add option to download snapshots on graph rendering completion, #239
remove custom home button in preference of resetCameraLastSave3d, resolves #265
numpy versioning fo [... cut]
closed 2020-08-18 by jneilliii