jneilliii/OctoPrint-BedLevelVisualizer #276: Right side bar not auto-scaling...

centauri61032, 2020-07-27:
OK, that probably didn't make any sense. But not sure how to explain it. See the attached screenshot.
[image]
Before the latest update, the bar on the right side of the screen would vary in its overall range to show measurements that correspond to the measured mesh. In other words, using my example, instead of the bar going from +2 to -2, it should actually go from something like +0.1 to -0.1 instead. At least that is what it used to do.
If this is a setting that I can change, I don't see it.
Thanks!

jneilliii, 2020-07-27:
I know what you mean, and that's on purpose based on previous feedback and confusion. There is already a request to get the old method back and flagged for next release, see https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/275

jneilliii, 2020-07-27:
BTW @centauri61032 the z axis can be set to -.1,.1 to possibly do what you want as well.

jneilliii, 2020-07-27:
This setting...

[image]

closed 2020-07-27 by jneilliii