jneilliii/OctoPrint-BedLevelVisualizer #277: Bed Level Visualizer Not Updating Mesh

kgrigio, 2020-07-28:
My Bed Level Visualizer just updated to 0.1.14 and when I "Update Mesh Now" the visualization updates to the mesh stored in memory and then doesn't update the visualization at the end of the bed leveling.  I have to save the mesh and rerun "Update Mesh Now" in order to see the previous bed leveling.  If I hit the home button, the mesh updates after a bed level, but when I go to a different tab, say the dashboard tab, and come back to bed visualization tab the mesh reverts to the previous mesh, even if I saved the results.  The previous version always updated the mesh at the end of a bed leveling.  Not sure what is going on.

jneilliii, 2020-07-28:
Please enabled debug logging in the plugin's settings, restart octoprint, and try the update mesh process again. Share screenshots of your settings and the plugin_bedlevelvisualizer_debug.log file from OctoPrint's settings in the logging section.

kgrigio, 2020-07-29:
Here is everything that you requested.  I hope I grabbed the debug logging file correctly.  One thing I didn't do today was to restart the OctoPi, forgot to do it, sorry.  I restarted it yesterday after doing a bunch of bed levels, trying to figure out the issue and that didn't help.  I am attaching a screenshot of the Settings, Current Mesh Data, and Angle Correction tabs.  I am also attaching screenshots of three different bed leveling events.
[image]
[image]
[image]

The first picture is labeled "Before Hitting Update Mesh Now.png" and is what the graph looked like before starting a bed level.
[image]

The second picture is labeled "Just after Hitting Update Mesh Now.png" and is what appeared after the screen said retrieving mesh or whatever it exactly says.
[image]

The third picture is labeled "Just after Bed Leveling Completed.png" and is what the graph looked like right after the bed leveling was completed.  
[image]

I realized here that I didn't move my bed at all to alter the level, so then I restarted a bed level after adjusting the leveling a bit.  The graph never changed at all during any of the above, which I wouldn't expect it to.  I didn't grab a screenshot until after the bed leveling was completed, because the graph didn't change.

The fourth picture is labeled "Just after Bed Leveling Completed 2nd time.png" which you would expect it to be different, and it does look slightly different, but not what I would expect from the adjustments I made to my bed level.
[image]

I next hit update mesh now to start a bed leveling process.  As soon as I did and the mesh was retrieved, the mesh updated to what had just happened.  That picture is labeled "Just after Hitting Update Mesh Now 3rd Time.png" and you can clearly see it is different than the mesh that was showing in the picture above just after the leveling completed.
[image]

Just for grins, while I was typing this up, I restarted OcotoPi and went through two more bed levelings with the same results as above.  The debug file doesn't contain these leveling events, just the above, if I did the debug correctly.

Thanks for looking into this and I hope this helps.

jneilliii, 2020-07-29:
Do me a favor and try putting @BEDLEVELVISUALIZER after the G28 command and just before the G29 command. After saving the settings restart octoprint (not necessary to restart the whole pi) and try again and let me know if that helps.

kgrigio, 2020-07-29:
That worked, however when I start to update the mesh, the graph only says retrieving the current mesh or whatever that message says.  Once the leveling finishes, the mesh updates to what was just done.  Thanks.

jneilliii, 2020-07-30:
So it's only updating after every restart of OctoPrint once and every subsequent attempt pulls from the stored mesh? Is your entire probing process able to complete in 10 minutes?