jneilliii/OctoPrint-BedLevelVisualizer #278: [BUG]: X-Y colors inverted on angle adjustment view

jon-daemon, 2020-07-31:
**Describe the bug**
I think that the x-Position and Y-Position text colors on angle adjustment view are inverted. See the screenshot below. My bed size is 220 x 205 (X x Y), so X-Position should be green and Y-Position should be black. The only option I have checked is "Descending y axis" to have a better view on angle adjustment table. Am I missing something?

**Screenshots**
[image]

**Firmware and Version**
Octoprint 1.4.0
Bed Visualizer 0.1.14
Marlin 1.1.9

jneilliii, 2020-07-31:
You are completely right, that angle correction color assignments on the labels are completely reversed. Thanks for pointing that out.

jneilliii, 2020-07-31:
Issue fixed in above commit, and will be part of release 0.1.15.

commit 3fb614e6 (2020-10-03): 0.1.15 (#304)

fix axis label colors, resolves #278
make colorscale limits allow option auto for the old method of coloring, resolves #275
add responsive option to plotly config, resolves #274
first stab at SD card support, resolves #223
fix little issue with reactive option where graph was getting overlaid on top of correction table
change home button press to a different strategy, resolves #265
update plotly to version 1.54.7
add option to download snapshots on graph rendering completion, #239
remove custom home button in preference of resetCameraLastSave3d, resolves #265
numpy versioning fo [... cut]
commit 7565c066 (2020-10-03): 0.1.15 (#304)

fix axis label colors, resolves #278
make colorscale limits allow option auto for the old method of coloring, resolves #275
add responsive option to plotly config, resolves #274
first stab at SD card support, resolves #223
fix little issue with reactive option where graph was getting overlaid on top of correction table
change home button press to a different strategy, resolves #265
update plotly to version 1.54.7
add option to download snapshots on graph rendering completion, #239
remove custom home button in preference of resetCameraLastSave3d, resolves #265
numpy versioning fo [... cut]
closed 2020-08-18 by jneilliii