jneilliii/OctoPrint-BedLevelVisualizer #282: [BUG]: Tilt image generated not corresponding to the real tilt

Freiheitpt, 2020-08-11:
**Describe the bug**
I don't know if this was already reported, but I didn't find anything.
I restarted octoprint, resetted the plugin settings and the behaviour persists.
When I adjust the back left knob, and make that corner of the bed very low (on purpose), in the image shows that the front left corner is low. It seems that the logic is somewhat inverted. Please check the screenshots.

**Expected behavior**
Image generated to be correct

**Screenshots**
[image]
[image]

**Firmware and Version**
Octoprint 1.4.2
Marlin 2.0

jneilliii, 2020-08-11:
That's what all those flip and rotate options are for.

Freiheitpt, 2020-08-11:
I tried those flip options. Thing is.. The 200 coordinates switches with the 0 coordinate (real flip).
but then my home position, is.. 0,200 taking in account the image. i will post a screenshot in a little bit

jneilliii, 2020-08-11:
So you're good, I can close the issue?

Freiheitpt, 2020-08-11:
I'll close it.
Sorry again

closed 2020-08-11 by Freiheitpt