jneilliii/OctoPrint-BedLevelVisualizer #286: [FR]: Mesh Statistics Display

DodgeDeBoulet, 2020-08-25:
This is an enhancement request for additional statistical information on the current mesh.

I'd like to have the option to display a table showing mesh min/max/mean/average variance and standard deviation along with the graphical display.

This would bring BV's reporting in line with Duet/RepRap's Bed Height Map 😁

It would also be useful to be able to graph (or at least extract) these values over time, in-line with my previous suggestion regarding a historical mesh database.

AcJoker2, 2020-12-27:
I can only conclude that the addon is still missing that

jneilliii, 2020-12-27:
That is why the issue is still open.

AcJoker2, 2020-12-27:
I know that, I have only strengthened the desire for this expansion.
The more people advocate a new role, the more interesting.

jneilliii, 2022-01-01:
This has been implemented in the latest release candidate version of the plugin. You can change the plugin's release channel in OctoPrint's Software Update settings to `Release Candidate` and upgrade when prompted to test and validate.

[image]

PR #497 (2022-01-01): 1.1.1
* add ability to configure graph height in visualization settings, resolves #462
* add min, max, and variance values of graphed mesh, resolves #286
* add OctoDash support, create Custom Action with `[!WEB]http://127.0.0.1:5000/plugin/bedlevelvisualizer/bedlevelvisualizer` as the command.
* adjust the tables for better theme support, resolves #479
* add option for showing mesh data on tab, resolves #496
* remove tooltip hover on tables, resolves #490
* add Prusa adjustment values to graph (still needs verification), resolves #242
* add appearance > title setting to automatically downloade [... cut]
closed 2022-01-01 by jneilliii