jneilliii/OctoPrint-BedLevelVisualizer #289: G29 T do nothing...

tiagomaricate, 2020-08-31:
Hi people!
Sorry to use here to verify what is happening...
If I use the G28 and G29 T I receive a lot of zeros.
Also, a message...

`
Bed Visualizer Error
Errors while attempting render of mesh data.
Error:
TypeError: Failed to execute 'uniformMatrix4fv' on 'WebGLRenderingContext': No function was found that matched the signature provided.
Received Data:
[object Object]`

I use marlin firmware.

Also, I did some tests using these instructions (https://marlinfw.org/docs/features/unified_bed_leveling.html) and the error above started.

Could you people help me to make this work? I'm completely noob on this.

jneilliii, 2020-08-31:
Well it all depends on the type of bed leveling you have enabled in your marlin firmware. Do you know if it's UBL or Bi-linear or something else? It would be good to reference these [examples](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/master/wiki/gcode-examples.md) based on the type of leveling you have. Enable debug logging in the plugin's settings when you adjust the GCODE to use, and then restart OctoPrint to initialize everything properly. Give it a try and if it doesn't work, grab your plugin_bedlevelvisualiver_debug.log file from OctoPrint's logging sections and drag it into your comment here to upload and share.

tiagomaricate, 2020-08-31:
> Well it all depends on the type of bed leveling you have enabled in your marlin firmware. Do you know if it's UBL or Bi-linear or something else? It would be good to reference these [...]

UBL is active on the firmware...
I'm receiving this error:
`Bed Visualizer Error
Errors while attempting render of mesh data.
Error:
TypeError: Failed to execute 'uniformMatrix4fv' on 'WebGLRenderingContext': No function was found that matched the signature provided.
Received Data:
[object Object]
[octoprint (3).log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/5152706/octoprint.3.log)
`

Logs are attached.
[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/5152708/plugin_bedlevelvisualizer_debug.log)

tiagomaricate, 2020-08-31:
[image]

jneilliii, 2020-09-01:
OK, if you're using UBL there is a process that you have to go through. I personally do the process outside of the plugin and send the commands below to the printer. 

```
;------------------------------------------
;--- Setup and initial probing commands ---
;------------------------------------------
M190 S60        ; Not required, but having the bed at temperature helps accuracy, use the bed temp you normally print at

G28             ; Home XYZ.
G29 P1          ; Do automated probing of the bed.
G29 P3 T        ; Fill in missed probe points
G29 S1          ; Save UBL mesh points to EEPROM.
G29 F 10.0      ; Set Fade Height for correction at 10.0 mm.
G29 A           ; Activate the UBL System.
M500            ; Save current setup. WARNING: UBL will be active at power up, before any `G28`.
```

Once the mesh is generated and saved I then use these commands in the plugin's setting.

```
@BEDLEVELVISUALIZER
M420 V
```

tiagomaricate, 2020-09-01:
G29 A
M500

> @BEDLEVELVISUALIZER
> M420 V

I followed the instruction and get exactly the same error.
On terminal I can see all the numbers from UBL. All the operations works great.
The problem is when I click on the Bed Visualizer tab.

`Send: M500
Recv: echo:Settings Stored (498 bytes; crc 18882)
Recv: Mesh saved in slot 1
Recv: ok
[...]
Send: M420 V
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv: (0,9)                                                                   (9,9)
Recv: (1,196)                                                                (196,196)
Recv:   19.889    19.801    19.714    19.782    19.825    19.981    19.894    19.770    19.715    19.641
Recv: 
Recv:   19.738    19.721    19.705    19.882    19.810    19.801    19.632    19.808    19.651    19.647
Recv: 
Recv:   20.186    19.925    19.664    19.746    19.671    19.635    19.538    19.455    19.465    19.361
Recv: 
Recv:   19.714    19.714    19.761    19.822    19.615    19.685    19.700    19.506    19.395    19.535
Recv: 
Recv:   19.834    19.722    19.611    19.759    19.495    19.435    19.497    19.343    19.263    19.425
Recv: 
Recv: [ 19.727]   19.675    19.622    19.459    19.410    19.482    19.286    19.311    19.201    19.166
Recv: 
Recv:   19.505    19.486    19.468    19.261    19.263    19.194    19.480    19.186    19.129    19.305
Recv: 
Recv:   19.454    19.380    19.306    19.336    19.264    19.161    19.166    19.308    19.009    19.176
Recv: 
Recv:   19.454    19.380    19.306    19.411    19.265    19.161    19.166    19.429    19.009    19.176 T:19.37 /0.00 B:20.47 /0.00 @:0 B@:0
Recv: 
Recv: 
Recv:      .          .          .          .          .          .          .          .          .          .
Recv: (1,1)                                                                    (196,1)
Recv: (0,0)                                                                     (9,0)
Recv: UBL_MESH_VALID = 0
Recv: ubl.state.storage_slot = 1
Recv: echo:Bed Leveling Ligado
Recv: ok`

[image]

jneilliii, 2020-09-01:
So your points are extremely high and probably outside of the range of graphing. I think if you enable the option ` Use Relative Z Offsets` it will fix your issue.

tiagomaricate, 2020-09-01:
> So your points are extremely high and probably outside of the range of graphing. I think if you enable the option ` Use Relative Z Offsets` it will fix your issue.

Nop... same error.
Enabled the z-offset on settings and redo all processes. Exact same error.

jneilliii, 2020-09-01:
Please enable debug logging in the plugin's settings, restart octoprint and try the update mesh process again. If it is still failing go into octoprint's logging section and download plugin_bedlevelvisualizer_debug.log and drag it into a comment here to upload.

tiagomaricate, 2020-09-01:
> Please enable debug logging in the plugin's settings, restart octoprint and try the update mesh process again. If it is still failing go into octoprint's logging section and download plugin_bedlevelvisualizer_debug.log and drag it into a comment here to upload.

Almost there!
Now I see this... but nothing happens... 

[image]

tiagomaricate, 2020-09-01:
The log only shows this...

` mesh collection started
[2020-09-01 13:51:05,105] DEBUG: []
[2020-09-01 13:54:14,992] DEBUG: Canceling mesh collection per user request
[2020-09-01 13:54:14,994] DEBUG: Mesh data collected prior to cancel:
[2020-09-01 13:54:14,995] DEBUG: []
[2020-09-01 13:54:14,996] DEBUG: Mesh data after clearing:
[2020-09-01 13:54:14,997] DEBUG: []
[2020-09-01 13:59:31,304] DEBUG: mesh collection started
[2020-09-01 13:59:31,345] DEBUG: []
[2020-09-01 13:59:32,976] DEBUG: [u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.']
[2020-09-01 13:59:33,035] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 197.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 197.0, 'z_max': 250.0}
[2020-09-01 13:59:33,036] DEBUG: stopping mesh collection
[2020-09-01 13:59:33,039] DEBUG: using relative offsets
[2020-09-01 14:00:46,039] DEBUG: Canceling mesh collection per user request
[2020-09-01 14:00:46,040] DEBUG: Mesh data collected prior to cancel:
[2020-09-01 14:00:46,040] DEBUG: [[u'.' u'.' u'.' u'.' u'.' u'.' u'.' u'.' u'.' u'.']]
[2020-09-01 14:00:46,041] DEBUG: Mesh data after clearing:
[2020-09-01 14:00:46,041] DEBUG: []
[2020-09-01 14:01:45,256] DEBUG: mesh collection started
[2020-09-01 14:01:45,278] DEBUG: []
[2020-09-01 14:01:46,913] DEBUG: [u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.']
[2020-09-01 14:01:46,975] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 197.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 197.0, 'z_max': 250.0}
[2020-09-01 14:01:46,976] DEBUG: stopping mesh collection
[2020-09-01 14:01:46,979] DEBUG: using relative offsets
[2020-09-01 14:02:00,736] DEBUG: Canceling mesh collection per user request
[2020-09-01 14:02:00,738] DEBUG: Mesh data collected prior to cancel:
[2020-09-01 14:02:00,738] DEBUG: [[u'.' u'.' u'.' u'.' u'.' u'.' u'.' u'.' u'.' u'.']]
[2020-09-01 14:02:00,740] DEBUG: Mesh data after clearing:
[2020-09-01 14:02:00,740] DEBUG: []
[2020-09-01 14:02:13,180] DEBUG: mesh collection started
`

jneilliii, 2020-09-01:
I'll have to do some testing on my side for this. It appears as if the data is not getting extracted properly for some reason. This line should be numbers not just decimals.

```
[u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.']
```

tiagomaricate, 2020-09-01:
> I'll have to do some testing on my side for this. It appears as if the data is not getting extracted properly for some reason. This line should be numbers not just decimals.
> 
> ```
> [u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.']
> ```

Very odd... while the sensor is capturing the points all the values are shown on terminal.
As you noted, my Z-axis is 0 on the bottom position and max on top.

jneilliii, 2020-09-02:
Now that you've completed a UBL process and saved your mesh, what do you get if you send the command `M420 V` in the terminal tab? If that comes back with real data, you can just put that command in the plugin's setting and see if it makes a difference.

jneilliii, 2020-09-02:
never mind, I see we tried that already.

tiagomaricate, 2020-09-02:
Yep... looks like it's not saving the results. Makes sense? Is there a way to force some value, save it, and try to recover? Just to check if it's really saving.

jneilliii, 2020-09-03:
You could try going into plugin manager and wiping the settings from the plugin, basically resetting it to default values. Then in the plugin's settings go ahead and enable debug logging and save, restart octoprint and try from scratch.

tiagomaricate, 2020-09-04:
> You could try going into plugin manager and wiping the settings from the plugin, basically resetting it to default values. Then in the plugin's settings go ahead and enable debug logging and save, restart octoprint and try from scratch.

I just see this message:

 **_Please wait, retrieving current mesh._**

tiagomaricate, 2020-09-04:
The log

[2020-09-04 10:50:21,536] DEBUG: mesh collection started
[2020-09-04 10:50:21,610] DEBUG: []
[2020-09-04 10:50:23,227] DEBUG: [u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.']
[2020-09-04 10:50:23,287] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 197.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 197.0, 'z_max': 250.0}
[2020-09-04 10:50:23,288] DEBUG: stopping mesh collection
[2020-09-04 10:50:23,292] DEBUG: using relative offsets
[2020-09-04 10:52:24,607] DEBUG: Canceling mesh collection per user request
[2020-09-04 10:52:24,607] DEBUG: Mesh data collected prior to cancel:
[2020-09-04 10:52:24,608] DEBUG: [[u'.' u'.' u'.' u'.' u'.' u'.' u'.' u'.' u'.' u'.']]
[2020-09-04 10:52:24,609] DEBUG: Mesh data after clearing:
[2020-09-04 10:52:24,609] DEBUG: []
[2020-09-04 10:53:00,554] DEBUG: mesh collection started
[2020-09-04 10:53:00,575] DEBUG: []
[2020-09-04 10:53:02,240] DEBUG: [u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.']
[2020-09-04 10:53:02,259] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 197.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 197.0, 'z_max': 250.0}
[2020-09-04 10:53:02,260] DEBUG: stopping mesh collection
[2020-09-04 10:53:02,261] DEBUG: using relative offsets
[2020-09-04 10:54:40,276] DEBUG: Canceling mesh collection per user request
[2020-09-04 10:54:40,277] DEBUG: Mesh data collected prior to cancel:
[2020-09-04 10:54:40,278] DEBUG: [[u'.' u'.' u'.' u'.' u'.' u'.' u'.' u'.' u'.' u'.']]
[2020-09-04 10:54:40,279] DEBUG: Mesh data after clearing:
[2020-09-04 10:54:40,279] DEBUG: []
[2020-09-04 10:55:00,226] DEBUG: mesh collection started
[2020-09-04 10:55:00,254] DEBUG: []
[2020-09-04 10:55:01,892] DEBUG: [u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.']
[2020-09-04 10:55:01,943] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 197.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 197.0, 'z_max': 250.0}
[2020-09-04 10:55:01,944] DEBUG: stopping mesh collection
[2020-09-04 10:55:01,946] DEBUG: using relative offsets
[2020-09-04 10:55:10,772] DEBUG: []
[2020-09-04 10:55:12,402] DEBUG: [u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.']
[2020-09-04 10:55:12,453] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 197.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 197.0, 'z_max': 250.0}
[2020-09-04 10:55:12,453] DEBUG: stopping mesh collection
[2020-09-04 10:55:12,455] DEBUG: using relative offsets

LMS0815, 2020-11-03:
I had the same issue. Clearing the configuration and reinstall OctoPrint-BedLevelVisualizer solved it for me.
It all started, when I was playing around with the "Clockwise Rotation" settings.
I strongly believe that OctoPrint-BedLevelVisualizer has issues, when the value initialization fails due to corrupted(?) configuration/settings.

jneilliii, 2020-11-03:
> I strongly believe that OctoPrint-BedLevelVisualizer has issues, when the value initialization fails due to corrupted(?) configuration/settings.

This is very possible, but I have no way of knowing if the configuration settings get corrupted or not or how they are getting corrupted. Most times doing something similar to what you describe resolves the issue and gets it working again, but with the log listed above the actual values are not getting extracted correctly, hence all the empty arrays of dots and not numbers. I'm not so sure what's causing that either.

LMS0815, 2020-11-04:
Let's wait and see if my proposed solution helps @tiagomaricate as well. I assume that an uninitialized variable is corrupting the extraction algorithm.  My two  OctoPi boxes are receiving 100% same output from the printer (as I just moved the USB connection).  One had no issues the other one was stuck and reached timeout without providing results (unfortunately I did not save the logs).

tiagomaricate, 2020-11-04:
> Let's wait and see if my proposed solution helps @tiagomaricate as well. I assume that an uninitialized variable is corrupting the extraction algorithm. My two OctoPi boxes are receiving 100% same [...]

I'm reassembling my printer with a new mainboard and drivers.
I think part of the problem could be the old configuration for the hardware.

LMS0815, 2020-11-04:
> This is very possible, but I have no way of knowing if the configuration settings get corrupted or not or how they are getting corrupted. Most times doing something similar to what you describe resolves the issue and gets it working again,

Usually I check the variable type and value after initialization, if it meets my requirements and/or force type.

`return ((parseFloat(item) - parseFloat(self.mesh_zero())) * parseFloat(self.mesh_unit()) * 360 / (self.imperial()?25.4/parseFloat(self.screw_hub()):parseFloat(self.screw_hub())));`

jneilliii, 2020-11-04:
that's on the js side though, the issue that I'm seeing with those `[u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.']` are on the python side. Pretty sure my regular expression extraction routine is bombing out here for some reason, but am unable to reproduce.

LMS0815, 2020-11-05:
> [u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.', u'.']

The printout indicates that the output values are Unicode string. I am not sure if you expect Unicode or Numbers or normal string.
I just met to explain, that - as I see there are lots of compares in your parser - there is a risk of comparing values of different types which usually ends with unexpected results. (Also there usually is no need in Python 2 to convert unicode and nonunicode strings) but I know regex on unicode has different behavior on certain classes like  \w or \b.

closed 2021-02-26 by github-actions[bot]