jneilliii/OctoPrint-BedLevelVisualizer #290: When running Visualizer I get this

uuys007, 2020-08-31:
**Describe the bug**

**Expected behavior**

**Debug Log**

**Screenshots**

**Firmware and Version**

uuys007, 2020-08-31:
When running I get this 

what could be the problem ?

normal marlin 1.1.9 on the board

Your help will be appreciated

octoprint running on raspberrypi 2

[image]
[image]

uuys007, 2020-08-31:
after while get this
[image]

jneilliii, 2020-08-31:
Well it all depends on the type of bed leveling you have enabled in your marlin firmware. Do you know if it's UBL or Bi-linear or something else? It would be good to reference these [examples](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/master/wiki/gcode-examples.md) based on the type of leveling you have. Enable debug logging in the plugin's settings when you adjust the GCODE to use, and then restart OctoPrint to initialize everything properly. Give it a try and if it doesn't work, grab your plugin_bedlevelvisualiver_debug.log file from OctoPrint's logging sections and drag it into your comment here to upload and share.

uuys007, 2020-08-31:
i think it is Bed Leveling (Bilinear)

uuys007, 2020-08-31:
how do i get the plugin_bedlevelvisualiver_debug.log

jneilliii, 2020-08-31:
So with bilinear you just use 
```
@BEDLEVELVISUALIZER
G29
```

If you enabled the debug logging option and restarted OctoPrint you can get the log file by going to OctoPrint's settings and clicking logging on the left hand side. You should see the file in the list.

uuys007, 2020-08-31:
i did set the settings to

but it did not create latest log file

uuys007, 2020-08-31:
Here is older log file

2020-08-19 23:13:03,779   /home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install file:///tmp/tmpnv1YB9/OctoPrint-BedLevelVisualizer-master.zip --no-cache-dir
2020-08-19 23:13:07,833 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2020-08-19 23:13:07,834 > Processing /tmp/tmpnv1YB9/OctoPrint-BedLevelVisualizer-master.zip
2020-08-19 23:13:10,257 > Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from Bed-Visualizer==0.1.14) (1.4.2)
2020-08-19 23:13:34,283 > Collecting numpy
2020-08-19 23:13:39,607 > Downloading https://files.pythonhosted.org/packages/b7/6f/24647f014eef9b67a24adfcbcd4f4928349b4a0f8393b3d7fe648d4d2de3/numpy-1.16.6.zip (5.1MB)
2020-08-19 23:14:00,954 > Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.0.0)
2020-08-19 23:14:00,955 > Requirement already satisfied: psutil<6,>=5.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (5.7.2)
2020-08-19 23:14:00,956 > Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (0.5.4)
2020-08-19 23:14:00,956 > Requirement already satisfied: monotonic<2,>=1.5; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.5)
2020-08-19 23:14:00,957 > Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.0.7)
2020-08-19 23:14:00,958 > Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (5.3.1)
2020-08-19 23:14:00,959 > Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (0.10.9)
2020-08-19 23:14:00,960 > Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (0.18.2)
2020-08-19 23:14:00,960 > Requirement already satisfied: feedparser<6,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (5.2.1)
2020-08-19 23:14:00,961 > Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (0.7.19)
2020-08-19 23:14:00,962 > Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (2.24.0)
2020-08-19 23:14:00,962 > Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (7.1.2)
2020-08-19 23:14:00,963 > Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.4)
2020-08-19 23:14:00,963 > Requirement already satisfied: watchdog<1,>=0.10.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (0.10.3)
2020-08-19 23:14:00,964 > Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.1.0)
2020-08-19 23:14:00,964 > Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.5.0.1)
2020-08-19 23:14:00,965 > Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (2.11.2)
2020-08-19 23:14:00,966 > Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.2)
2020-08-19 23:14:00,966 > Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (0.5.0)
2020-08-19 23:14:00,967 > Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.1.2)
2020-08-19 23:14:00,967 > Requirement already satisfied: rsa==4.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (4.0)
2020-08-19 23:14:00,968 > Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (2.8.5)
2020-08-19 23:14:00,969 > Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (0.4.21)
2020-08-19 23:14:00,969 > Requirement already satisfied: OctoPrint-FileCheck>=2020.07.06 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (2020.8.7)
2020-08-19 23:14:00,970 > Requirement already satisfied: futures<4,>=3.3; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (3.3.0)
2020-08-19 23:14:00,971 > Requirement already satisfied: websocket-client<1,>=0.57 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (0.57.0)
2020-08-19 23:14:00,971 > Requirement already satisfied: scandir<2,>=1.10; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.10.0)
[... cut ...]
2020-08-19 23:32:49,253 > Stored in directory: /tmp/pip-ephem-wheel-cache-2Z9RSP/wheels/d4/1e/2a/768922a8d7ae4a90c60782f16b419412ba579b0938c0fdfe1b
2020-08-19 23:32:50,408 > Successfully built Bed-Visualizer
2020-08-19 23:32:50,803 > Installing collected packages: Bed-Visualizer
2020-08-19 23:32:50,804 > Found existing installation: Bed-Visualizer 0.1.14
2020-08-19 23:32:50,805 > Uninstalling Bed-Visualizer-0.1.14:
2020-08-19 23:32:50,806 > Successfully uninstalled Bed-Visualizer-0.1.14
2020-08-19 23:32:51,312 > Successfully installed Bed-Visualizer-0.1.14

jneilliii, 2020-08-31:
Yeah, that doesn't help me debug your situation. verify it's enabled, restart octoprint and then pres the update mesh button. Let it timeout like before, and then look to see if that file is there.

uuys007, 2020-08-31:
and screen shots

[image]

[image]

uuys007, 2020-08-31:
im running again to see if log is created

uuys007, 2020-08-31:
im not getting a log created any suggestions just time out

jneilliii, 2020-08-31:
Yes, you are, it's in your screenshot...

[image]

uuys007, 2020-08-31:
i get this in the terminal

Send: G28
Recv: ok
Send: M155 S30
Recv: ok
Send: G29
Recv: Bed x: 5.00 y: 20.00 z: 6.48
Recv: Bed x: 117.00 y: 20.00 z: 6.48
Recv: Bed x: 229.00 y: 20.00 z: 6.22
Recv: Bed x: 230.00 y: 145.00 z: 6.41
Recv: Bed x: 118.00 y: 145.00 z: 6.65
Recv: Bed x: 6.00 y: 145.00 z: 6.55
Recv: Bed x: 5.00 y: 270.00 z: 5.92
Recv: Bed x: 117.00 y: 270.00 z: 6.43
Recv: Bed x: 229.00 y: 270.00 z: 6.48
Recv: Eqn coefficients: a: 0.00 b: -0.00 d: 6.44
Recv: planeNormal x: -0.00 y: 0.00 z: 1.00
Recv: ok

jneilliii, 2020-08-31:
Yep, that should be able to be extracted properly. So update your gcode settings to this.

```
@BEDLEVELVISUALIZER
G29
```

Restart OctoPrint and press the Update Mesh button.

uuys007, 2020-08-31:
still awaiting timeout been running like this now for 20min and then get this

[image]

uuys007, 2020-08-31:
[image]

Log not here

jneilliii, 2020-08-31:
Did you delete the previous log file that was there? Anytime you delete a log file you basically have to restart OctoPrint to initialize properly.

uuys007, 2020-08-31:
[image]

Setting not changed was like this with loggin on

uuys007, 2020-08-31:
not sure restarting now

uuys007, 2020-08-31:
[image]

uuys007, 2020-08-31:
after last reboot log file 0kb

uuys007, 2020-08-31:
running bed level again will take about 20min to time out

uuys007, 2020-08-31:
still te same

[image]

jneilliii, 2020-09-01:
Do this for me, test my current development version and see if you have the same problem or not. Copy/paste the URL below into Plugin Manager > Get More > ...from URL and click Install.

```
https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.15.zip
```

uuys007, 2020-09-01:
will do today

uuys007, 2020-09-01:
i did install the dev one as above still the same

uuys007, 2020-09-01:
this was log that was created last night

[2020-08-31 21:25:54,458] DEBUG: mesh collection started
[2020-08-31 21:26:12,205] DEBUG: mesh collection started
[2020-08-31 21:26:20,641] DEBUG: [u'5.00', u'20.00', u'6.46']
[2020-08-31 21:26:20,654] DEBUG: using old marlin flag
[2020-08-31 21:26:28,233] DEBUG: [u'117.00', u'20.00', u'6.29']
[2020-08-31 21:26:28,245] DEBUG: using old marlin flag
[2020-08-31 21:26:35,759] DEBUG: [u'229.00', u'20.00', u'6.09']
[2020-08-31 21:26:35,776] DEBUG: using old marlin flag
[2020-08-31 21:26:43,372] DEBUG: [u'230.00', u'145.00', u'6.18']
[2020-08-31 21:26:43,389] DEBUG: using old marlin flag
[2020-08-31 21:26:50,835] DEBUG: [u'118.00', u'145.00', u'6.41']
[2020-08-31 21:26:50,837] DEBUG: using old marlin flag
[2020-08-31 21:26:58,373] DEBUG: [u'6.00', u'145.00', u'6.25']
[2020-08-31 21:26:58,379] DEBUG: using old marlin flag
[2020-08-31 21:27:05,986] DEBUG: [u'5.00', u'270.00', u'5.99']
[2020-08-31 21:27:05,998] DEBUG: using old marlin flag
[2020-08-31 21:27:13,422] DEBUG: [u'117.00', u'270.00', u'6.24']
[2020-08-31 21:27:13,424] DEBUG: using old marlin flag
[2020-08-31 21:27:20,859] DEBUG: [u'229.00', u'270.00', u'6.33']
[2020-08-31 21:27:20,876] DEBUG: using old marlin flag
[2020-08-31 21:27:20,901] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:27:20,946] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:27:25,810] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:27:30,813] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:27:35,818] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:27:40,819] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:27:45,823] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:27:50,825] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:27:55,816] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:28:00,829] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:28:05,832] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:28:10,819] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:28:15,837] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:28:20,836] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:28:25,837] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:28:30,837] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:28:35,845] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:28:40,847] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:28:45,851] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:28:50,846] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:28:55,839] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:29:00,851] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:29:05,845] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:29:10,854] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:29:15,857] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:29:20,860] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:29:25,874] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:29:30,857] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:29:35,871] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-08-31 21:29:40,871] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[... cut ...]
[2020-09-01 09:07:33,728] DEBUG: Canceling mesh collection per user request
[2020-09-01 09:07:33,730] DEBUG: Mesh data collected prior to cancel:
[2020-09-01 09:07:33,730] DEBUG: [[u'5.00', u'20.00', u'6.26'], [u'117.00', u'20.00', u'6.40'], [u'229.00', u'20.00', u'6.15'], [u'230.00', u'145.00', u'6.36'], [u'118.00', u'145.00', u'6.51'], [u'6.00', u'145.00', u'6.20'], [u'5.00', u'270.00', u'5.91'], [u'117.00', u'270.00', u'6.38'], [u'229.00', u'270.00', u'6.32']]
[2020-09-01 09:07:33,732] DEBUG: Mesh data after clearing:
[2020-09-01 09:07:33,732] DEBUG: []

jneilliii, 2020-09-01:
Thanks. This is strange because what is happening is that the mesh collection is not seeing the response that all points have been reported. This is typically an `ok` from the printer after the collection started. Instead the timeout is occurring and then the previously collected data gets cleared. I've been thinking of ways to improve this "I'm done receiving the data" in the plugin but as of yet have not found a consistent approach that will work across multiple firmware variants. 

One last request if you don't mind. I'd like to try an older version of the plugin to see if it works any better for you in this situation. Please install using the URL below like before and let me know if it's able to graph your bed or not.

```
https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.12.zip
```

uuys007, 2020-09-02:
i have loaded older version  

Got this halve way through mesurements

[image]

uuys007, 2020-09-02:
[image]

uuys007, 2020-09-02:
after this tried auto update to version 1.14 hung again

any idee what else to test

uuys007, 2020-09-02:
Log file

[2020-09-02 12:14:10,646] DEBUG: mesh collection started
[2020-09-02 12:14:28,089] DEBUG: ok
[2020-09-02 12:14:28,110] DEBUG: ok
[2020-09-02 12:14:28,124] DEBUG: mesh collection started
[2020-09-02 12:14:36,525] DEBUG: Bed x: 5.00 y: 20.00 z: 6.44
[2020-09-02 12:14:36,544] DEBUG: [u'5.00', u'20.00', u'6.44']
[2020-09-02 12:14:36,565] DEBUG: using old marlin flag
[2020-09-02 12:14:43,983] DEBUG: Bed x: 117.00 y: 20.00 z: 6.23
[2020-09-02 12:14:43,986] DEBUG: [u'117.00', u'20.00', u'6.23']
[2020-09-02 12:14:43,988] DEBUG: using old marlin flag
[2020-09-02 12:14:51,425] DEBUG: Bed x: 229.00 y: 20.00 z: 6.15
[2020-09-02 12:14:51,429] DEBUG: [u'229.00', u'20.00', u'6.15']
[2020-09-02 12:14:51,447] DEBUG: using old marlin flag
[2020-09-02 12:14:59,012] DEBUG: Bed x: 230.00 y: 145.00 z: 6.29
[2020-09-02 12:14:59,022] DEBUG: [u'230.00', u'145.00', u'6.29']
[2020-09-02 12:14:59,033] DEBUG: using old marlin flag
[2020-09-02 12:15:06,405] DEBUG: Bed x: 118.00 y: 145.00 z: 6.48
[2020-09-02 12:15:06,409] DEBUG: [u'118.00', u'145.00', u'6.48']
[2020-09-02 12:15:06,426] DEBUG: using old marlin flag
[2020-09-02 12:15:07,280] DEBUG: Canceling mesh collection per user request
[2020-09-02 12:15:07,281] DEBUG: Mesh data collected prior to cancel:
[2020-09-02 12:15:07,282] DEBUG: [[u'5.00', u'20.00', u'6.44'], [u'117.00', u'20.00', u'6.23'], [u'229.00', u'20.00', u'6.15'], [u'230.00', u'145.00', u'6.29'], [u'118.00', u'145.00', u'6.48']]
[2020-09-02 12:15:07,283] DEBUG: Mesh data after clearing:
[2020-09-02 12:15:07,284] DEBUG: []
[2020-09-02 12:17:44,753] DEBUG: mesh collection started
[2020-09-02 12:17:53,168] DEBUG: Bed x: 5.00 y: 20.00 z: 6.55
[2020-09-02 12:17:53,171] DEBUG: [u'5.00', u'20.00', u'6.55']
[2020-09-02 12:17:53,172] DEBUG: using old marlin flag
[2020-09-02 12:18:00,744] DEBUG: Bed x: 117.00 y: 20.00 z: 6.22
[2020-09-02 12:18:00,747] DEBUG: [u'117.00', u'20.00', u'6.22']
[2020-09-02 12:18:00,748] DEBUG: using old marlin flag
[2020-09-02 12:18:08,247] DEBUG: Bed x: 229.00 y: 20.00 z: 6.09
[2020-09-02 12:18:08,266] DEBUG: [u'229.00', u'20.00', u'6.09']
[2020-09-02 12:18:08,267] DEBUG: using old marlin flag
[2020-09-02 12:18:15,797] DEBUG: Bed x: 230.00 y: 145.00 z: 6.40
[2020-09-02 12:18:15,816] DEBUG: [u'230.00', u'145.00', u'6.40']
[2020-09-02 12:18:15,818] DEBUG: using old marlin flag
[2020-09-02 12:18:23,266] DEBUG: Bed x: 118.00 y: 145.00 z: 6.48
[2020-09-02 12:18:23,278] DEBUG: [u'118.00', u'145.00', u'6.48']
[2020-09-02 12:18:23,283] DEBUG: using old marlin flag
[2020-09-02 12:18:27,100] DEBUG: Canceling mesh collection per user request
[2020-09-02 12:18:27,102] DEBUG: Mesh data collected prior to cancel:
[2020-09-02 12:18:27,102] DEBUG: [[u'5.00', u'20.00', u'6.55'], [u'117.00', u'20.00', u'6.22'], [u'229.00', u'20.00', u'6.09'], [u'230.00', u'145.00', u'6.40'], [u'118.00', u'145.00', u'6.48']]
[2020-09-02 12:18:27,103] DEBUG: Mesh data after clearing:
[2020-09-02 12:18:27,104] DEBUG: []
[2020-09-02 12:19:53,984] DEBUG: mesh collection started
[2020-09-02 12:20:02,447] DEBUG: Bed x: 5.00 y: 20.00 z: 6.22
[2020-09-02 12:20:02,455] DEBUG: [u'5.00', u'20.00', u'6.22']
[2020-09-02 12:20:02,456] DEBUG: using old marlin flag
[2020-09-02 12:20:09,874] DEBUG: Bed x: 117.00 y: 20.00 z: 6.37
[2020-09-02 12:20:09,878] DEBUG: [u'117.00', u'20.00', u'6.37']
[2020-09-02 12:20:09,894] DEBUG: using old marlin flag
[2020-09-02 12:20:17,401] DEBUG: Bed x: 229.00 y: 20.00 z: 5.99
[2020-09-02 12:20:17,404] DEBUG: [u'229.00', u'20.00', u'5.99']
[2020-09-02 12:20:17,406] DEBUG: using old marlin flag
[2020-09-02 12:20:24,946] DEBUG: Bed x: 230.00 y: 145.00 z: 6.35
[2020-09-02 12:20:24,962] DEBUG: [u'230.00', u'145.00', u'6.35']
[2020-09-02 12:20:24,964] DEBUG: using old marlin flag
[2020-09-02 12:20:32,409] DEBUG: Bed x: 118.00 y: 145.00 z: 6.48
[2020-09-02 12:20:32,425] DEBUG: [u'118.00', u'145.00', u'6.48']
[2020-09-02 12:20:32,430] DEBUG: using old marlin flag
[2020-09-02 12:20:36,319] DEBUG: Canceling mesh collection per user request
[2020-09-02 12:20:36,321] DEBUG: Mesh data collected prior to cancel:
[2020-09-02 12:20:36,322] DEBUG: [[u'5.00', u'20.00', u'6.22'], [u'117.00', u'20.00', u'6.37'], [u'229.00', u'20.00', u'5.99'], [u'230.00', u'145.00', u'6.35'], [u'118.00', u'145.00', u'6.48']]
[2020-09-02 12:20:36,323] DEBUG: Mesh data after clearing:
[2020-09-02 12:20:36,324] DEBUG: []
[2020-09-02 12:31:36,227] DEBUG: mesh collection started
[2020-09-02 12:31:53,927] DEBUG: mesh collection started
[... 193 lines cut ...]
[2020-09-02 12:47:31,417] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-09-02 12:47:36,432] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-09-02 12:47:41,432] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-09-02 12:47:46,485] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-09-02 12:47:51,438] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-09-02 12:47:56,455] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-09-02 12:48:01,456] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[... cut ...]
[2020-09-02 12:49:16,478] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-09-02 12:49:21,530] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-09-02 12:49:26,491] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-09-02 12:49:31,493] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}

uuys007, 2020-09-02:
is the problem the marlin firmware or the octoprint ?

jneilliii, 2020-09-02:
This could be a bug with the plugin. I do have some tweaks in my current dev version related to old marlin formats, and was wondering if you would be willing to test that version? I don't think you've tested it yet. Copy/Paste this URL into Plugin Manager > Get More > ...from URL and click Install. The version will still appear as 0.1.14 once installed.

```
https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.15.zip
```

uuys007, 2020-09-02:
Will do 

 

Will post when tested

 

Thx for updates so quickly

 

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.15.zip

uuys007, 2020-09-02:
i have loaded this version

[image]

uuys007, 2020-09-02:
[2020-09-02 12:14:10,646] DEBUG: mesh collection started
[2020-09-02 12:14:28,089] DEBUG: ok
[2020-09-02 12:14:28,110] DEBUG: ok
[2020-09-02 12:14:28,124] DEBUG: mesh collection started
[2020-09-02 12:14:36,525] DEBUG: Bed x: 5.00 y: 20.00 z: 6.44
[2020-09-02 12:14:36,544] DEBUG: [u'5.00', u'20.00', u'6.44']
[2020-09-02 12:14:36,565] DEBUG: using old marlin flag
[2020-09-02 12:14:43,983] DEBUG: Bed x: 117.00 y: 20.00 z: 6.23
[2020-09-02 12:14:43,986] DEBUG: [u'117.00', u'20.00', u'6.23']
[2020-09-02 12:14:43,988] DEBUG: using old marlin flag
[2020-09-02 12:14:51,425] DEBUG: Bed x: 229.00 y: 20.00 z: 6.15
[2020-09-02 12:14:51,429] DEBUG: [u'229.00', u'20.00', u'6.15']
[2020-09-02 12:14:51,447] DEBUG: using old marlin flag
[2020-09-02 12:14:59,012] DEBUG: Bed x: 230.00 y: 145.00 z: 6.29
[2020-09-02 12:14:59,022] DEBUG: [u'230.00', u'145.00', u'6.29']
[2020-09-02 12:14:59,033] DEBUG: using old marlin flag
[2020-09-02 12:15:06,405] DEBUG: Bed x: 118.00 y: 145.00 z: 6.48
[2020-09-02 12:15:06,409] DEBUG: [u'118.00', u'145.00', u'6.48']
[2020-09-02 12:15:06,426] DEBUG: using old marlin flag
[2020-09-02 12:15:07,280] DEBUG: Canceling mesh collection per user request
[2020-09-02 12:15:07,281] DEBUG: Mesh data collected prior to cancel:
[2020-09-02 12:15:07,282] DEBUG: [[u'5.00', u'20.00', u'6.44'], [u'117.00', u'20.00', u'6.23'], [u'229.00', u'20.00', u'6.15'], [u'230.00', u'145.00', u'6.29'], [u'118.00', u'145.00', u'6.48']]
[2020-09-02 12:15:07,283] DEBUG: Mesh data after clearing:
[2020-09-02 12:15:07,284] DEBUG: []
[2020-09-02 12:17:44,753] DEBUG: mesh collection started
[2020-09-02 12:17:53,168] DEBUG: Bed x: 5.00 y: 20.00 z: 6.55
[2020-09-02 12:17:53,171] DEBUG: [u'5.00', u'20.00', u'6.55']
[2020-09-02 12:17:53,172] DEBUG: using old marlin flag
[2020-09-02 12:18:00,744] DEBUG: Bed x: 117.00 y: 20.00 z: 6.22
[2020-09-02 12:18:00,747] DEBUG: [u'117.00', u'20.00', u'6.22']
[2020-09-02 12:18:00,748] DEBUG: using old marlin flag
[2020-09-02 12:18:08,247] DEBUG: Bed x: 229.00 y: 20.00 z: 6.09
[2020-09-02 12:18:08,266] DEBUG: [u'229.00', u'20.00', u'6.09']
[2020-09-02 12:18:08,267] DEBUG: using old marlin flag
[2020-09-02 12:18:15,797] DEBUG: Bed x: 230.00 y: 145.00 z: 6.40
[2020-09-02 12:18:15,816] DEBUG: [u'230.00', u'145.00', u'6.40']
[2020-09-02 12:18:15,818] DEBUG: using old marlin flag
[2020-09-02 12:18:23,266] DEBUG: Bed x: 118.00 y: 145.00 z: 6.48
[2020-09-02 12:18:23,278] DEBUG: [u'118.00', u'145.00', u'6.48']
[2020-09-02 12:18:23,283] DEBUG: using old marlin flag
[2020-09-02 12:18:27,100] DEBUG: Canceling mesh collection per user request
[2020-09-02 12:18:27,102] DEBUG: Mesh data collected prior to cancel:
[2020-09-02 12:18:27,102] DEBUG: [[u'5.00', u'20.00', u'6.55'], [u'117.00', u'20.00', u'6.22'], [u'229.00', u'20.00', u'6.09'], [u'230.00', u'145.00', u'6.40'], [u'118.00', u'145.00', u'6.48']]
[2020-09-02 12:18:27,103] DEBUG: Mesh data after clearing:
[2020-09-02 12:18:27,104] DEBUG: []
[2020-09-02 12:19:53,984] DEBUG: mesh collection started
[2020-09-02 12:20:02,447] DEBUG: Bed x: 5.00 y: 20.00 z: 6.22
[2020-09-02 12:20:02,455] DEBUG: [u'5.00', u'20.00', u'6.22']
[2020-09-02 12:20:02,456] DEBUG: using old marlin flag
[2020-09-02 12:20:09,874] DEBUG: Bed x: 117.00 y: 20.00 z: 6.37
[2020-09-02 12:20:09,878] DEBUG: [u'117.00', u'20.00', u'6.37']
[2020-09-02 12:20:09,894] DEBUG: using old marlin flag
[2020-09-02 12:20:17,401] DEBUG: Bed x: 229.00 y: 20.00 z: 5.99
[2020-09-02 12:20:17,404] DEBUG: [u'229.00', u'20.00', u'5.99']
[2020-09-02 12:20:17,406] DEBUG: using old marlin flag
[2020-09-02 12:20:24,946] DEBUG: Bed x: 230.00 y: 145.00 z: 6.35
[2020-09-02 12:20:24,962] DEBUG: [u'230.00', u'145.00', u'6.35']
[2020-09-02 12:20:24,964] DEBUG: using old marlin flag
[2020-09-02 12:20:32,409] DEBUG: Bed x: 118.00 y: 145.00 z: 6.48
[2020-09-02 12:20:32,425] DEBUG: [u'118.00', u'145.00', u'6.48']
[2020-09-02 12:20:32,430] DEBUG: using old marlin flag
[2020-09-02 12:20:36,319] DEBUG: Canceling mesh collection per user request
[2020-09-02 12:20:36,321] DEBUG: Mesh data collected prior to cancel:
[2020-09-02 12:20:36,322] DEBUG: [[u'5.00', u'20.00', u'6.22'], [u'117.00', u'20.00', u'6.37'], [u'229.00', u'20.00', u'5.99'], [u'230.00', u'145.00', u'6.35'], [u'118.00', u'145.00', u'6.48']]
[2020-09-02 12:20:36,323] DEBUG: Mesh data after clearing:
[2020-09-02 12:20:36,324] DEBUG: []
[2020-09-02 12:31:36,227] DEBUG: mesh collection started
[2020-09-02 12:31:53,927] DEBUG: mesh collection started
[2020-09-02 12:32:02,344] DEBUG: [u'5.00', u'20.00', u'6.59']
[2020-09-02 12:32:02,346] DEBUG: using old marlin flag
[... 997 lines cut ...]
[2020-09-02 20:28:59,310] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-09-02 20:29:04,313] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-09-02 20:29:09,349] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-09-02 20:29:14,306] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-09-02 20:29:19,323] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[2020-09-02 20:29:24,324] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 300.0, 'x_min': 0, 'type': 'rectangular', 'y_max': 300.0, 'z_max': 300.0}
[... cut ...]
[2020-09-02 20:30:38,529] DEBUG: Canceling mesh collection per user request
[2020-09-02 20:30:38,530] DEBUG: Mesh data collected prior to cancel:
[2020-09-02 20:30:38,531] DEBUG: [[u'5.00', u'20.00', u'6.03'], [u'117.00', u'20.00', u'6.47'], [u'229.00', u'20.00', u'6.19'], [u'230.00', u'145.00', u'6.40'], [u'118.00', u'145.00', u'6.51'], [u'6.00', u'145.00', u'6.50'], [u'5.00', u'270.00', u'6.13'], [u'117.00', u'270.00', u'6.38'], [u'229.00', u'270.00', u'6.21']]
[2020-09-02 20:30:38,533] DEBUG: Mesh data after clearing:
[2020-09-02 20:30:38,534] DEBUG: []

jneilliii, 2020-09-02:
Ok, so this is strange and probably some condition in the plugin that it's not seeing the completion of the mesh collection. That's when it gets into a loop about the bed dimensions and logging that data until the timeout is reached and then it cancels the collection. I see it does collect it properly, just for whatever reason never gets flagged that it finished, which is supposed to happen upon the next `ok` that is sent back from the printer. Can you do me one last favor and enable serial logging in octorpint > serial connection > general tab > at the bottom and restart octoprint. Run an update mesh process and then upload your serial.log file here (you can just drag the file into a comment), and then disable that option and restart octoprint again. You don't want that running all the time.

uuys007, 2020-09-02:
here is the serial.log

Hope this helps 

2020-09-02 23:40:29,245 - Changing monitoring state from "Offline" to "Opening serial connection"
2020-09-02 23:40:29,268 - Connecting to port /dev/ttyUSB0, baudrate 250000
2020-09-02 23:40:29,299 - Changing monitoring state from "Opening serial connection" to "Connecting"
2020-09-02 23:40:29,311 - Connected to: Serial<id=0x6ff55310, open=True>(port='/dev/ttyUSB0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2020-09-02 23:40:29,337 - Send: N0 M110 N0*125
2020-09-02 23:40:30,337 - Recv: start
2020-09-02 23:40:30,463 - Send: N0 M110 N0*125
2020-09-02 23:40:30,496 - Recv: echo:Marlin1.0.0
2020-09-02 23:40:30,512 - Recv: echo: Last Updated: Aug 25 2020 00:54:05 | Author: (none, default config)
2020-09-02 23:40:30,595 - Recv: Compiled: Aug 25 2020
2020-09-02 23:40:30,726 - Recv: echo: Free Memory: 3162  PlannerBufferBytes: 1232
2020-09-02 23:40:30,827 - Recv: echo:Hardcoded Default Settings Loaded
2020-09-02 23:40:30,832 - Recv: echo:Steps per unit:
2020-09-02 23:40:31,121 - Recv: echo:  M92 X202.00 Y202.00 Z3200.00 E840.00
2020-09-02 23:40:31,271 - Recv: echo:Maximum feedrates (mm/s):
2020-09-02 23:40:31,342 - Recv: echo:  M203 X300.00 Y300.00 Z15.00 E22.00
2020-09-02 23:40:31,432 - Recv: echo:Maximum Acceleration (mm/s2):
2020-09-02 23:40:31,486 - Recv: echo:  M201 X2000 Y2000 Z100 E2000
2020-09-02 23:40:31,627 - Recv: echo:Acceleration: S=acceleration, T=retract acceleration
2020-09-02 23:40:31,703 - Recv: echo:  M204 S500.00 T500.00
2020-09-02 23:40:31,770 - Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
2020-09-02 23:40:31,858 - Recv: echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
2020-09-02 23:40:31,920 - Recv: echo:Home offset (mm):
2020-09-02 23:40:32,062 - Recv: echo:  M206 X0.00 Y0.00 Z0.00
2020-09-02 23:40:32,118 - Recv: echo:PID settings:
2020-09-02 23:40:32,121 - Recv: echo:   M301 P22.20 I1.08 D114.00
2020-09-02 23:40:32,598 - Recv: echo:SD card ok
2020-09-02 23:40:32,936 - Recv: ok
2020-09-02 23:40:33,850 - Send: N0 M110 N0*125
2020-09-02 23:40:33,854 - Changing monitoring state from "Connecting" to "Operational"
2020-09-02 23:40:45,872 - Recv: ok
2020-09-02 23:40:45,881 - Send: N0 M110 N0*125
2020-09-02 23:40:45,887 - Recv: ok
2020-09-02 23:40:45,906 - Send: N1 M105*38
2020-09-02 23:40:45,913 - Recv: ok T:15.6 /0.0 B:14.3 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-02 23:40:45,944 - Send: N2 M115*36
2020-09-02 23:40:45,963 - Recv: FIRMWARE_NAME:Marlin V1; Sprinter/grbl mashup for gen6 FIRMWARE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Mendel EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
2020-09-02 23:40:45,978 - Recv: ok
2020-09-02 23:40:45,982 - Send: M20
2020-09-02 23:40:45,989 - Recv: Begin file list
2020-09-02 23:40:46,185 - Recv: End file list
2020-09-02 23:40:46,191 - Recv: ok
2020-09-02 23:40:51,185 - Send: M105
2020-09-02 23:40:51,210 - Recv: ok T:15.6 /0.0 B:14.2 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-02 23:40:55,959 - Send: M105
2020-09-02 23:40:55,975 - Recv: ok T:15.6 /0.0 B:14.2 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-02 23:41:00,972 - Send: M105
2020-09-02 23:41:01,040 - Recv: ok T:15.6 /0.0 B:14.4 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-02 23:41:05,976 - Send: M105
2020-09-02 23:41:06,018 - Recv: ok T:15.6 /0.0 B:14.3 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-02 23:41:10,977 - Send: M105
2020-09-02 23:41:10,988 - Recv: ok T:15.6 /0.0 B:14.5 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-02 23:41:16,004 - Send: M105
2020-09-02 23:41:16,066 - Recv: ok T:15.6 /0.0 B:14.5 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-02 23:41:21,096 - Send: M105
2020-09-02 23:41:21,112 - Recv: ok T:15.6 /0.0 B:14.5 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-02 23:41:26,098 - Send: M105
2020-09-02 23:41:26,119 - Recv: ok T:15.6 /0.0 B:14.5 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-02 23:41:31,106 - Send: M105
2020-09-02 23:41:31,113 - Recv: ok T:15.6 /0.0 B:14.6 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-02 23:41:36,108 - Send: M105
2020-09-02 23:41:36,120 - Recv: ok T:15.6 /0.0 B:14.4 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-02 23:41:41,110 - Send: M105
2020-09-02 23:41:41,135 - Recv: ok T:15.6 /0.0 B:14.7 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-02 23:41:46,111 - Send: M105
2020-09-02 23:41:46,184 - Recv: ok T:15.6 /0.0 B:14.3 /0.0 T0:15.6 /0.0 @:0 B@:0
[... 1271 lines cut ...]
2020-09-03 00:35:23,135 - Recv: ok T:15.6 /0.0 B:14.3 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-03 00:35:28,114 - Send: M105
2020-09-03 00:35:28,139 - Recv: ok T:15.6 /0.0 B:14.2 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-03 00:35:33,117 - Send: M105
2020-09-03 00:35:33,145 - Recv: ok T:15.6 /0.0 B:14.3 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-03 00:35:38,122 - Send: M105
2020-09-03 00:35:38,146 - Recv: ok T:15.6 /0.0 B:14.3 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-03 00:35:43,124 - Send: M105
2020-09-03 00:35:43,146 - Recv: ok T:15.6 /0.0 B:14.4 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-03 00:35:48,125 - Send: M105
2020-09-03 00:35:48,148 - Recv: ok T:15.7 /0.0 B:14.5 /0.0 T0:15.7 /0.0 @:0 B@:0
2020-09-03 00:35:53,135 - Send: M105
2020-09-03 00:35:53,160 - Recv: ok T:15.6 /0.0 B:14.3 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-03 00:35:58,137 - Send: M105
2020-09-03 00:35:58,163 - Recv: ok T:15.6 /0.0 B:14.4 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-03 00:36:03,138 - Send: M105
2020-09-03 00:36:03,161 - Recv: ok T:15.6 /0.0 B:14.2 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-03 00:36:08,140 - Send: M105
2020-09-03 00:36:08,155 - Recv: ok T:15.6 /0.0 B:14.4 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-03 00:36:13,141 - Send: M105
2020-09-03 00:36:13,214 - Recv: ok T:15.7 /0.0 B:14.3 /0.0 T0:15.7 /0.0 @:0 B@:0
2020-09-03 00:36:18,143 - Send: M105
2020-09-03 00:36:18,160 - Recv: ok T:15.6 /0.0 B:14.4 /0.0 T0:15.6 /0.0 @:0 B@:0
2020-09-03 00:36:23,145 - Send: M105
2020-09-03 00:36:23,166 - Recv: ok T:15.7 /0.0 B:14.4 /0.0 T0:15.7 /0.0 @:0 B@:0

uuys007, 2020-09-02:
must i leave the lserial.log on for now ? while testing ?

jneilliii, 2020-09-03:
no, I got the data in that I needed to test with, thanks. You can turn that back off and restart octorpint.

jneilliii, 2020-09-03:
I have been able to reproduce your issue with provided serial log and have identified where it is breaking in the plugin. After looking at your lines returned from the printer I see now why it's messing up. The plugin is expecting the points being probed to be on an even grid, but based on this information the x values are different/unexpected as they should filter down to the same 3 positions the same way the y values do. It will take a me a little while to figure out this logic, but plan on working on it this weekend.

```
2020-09-03 00:06:22,012 - Send: G29
2020-09-03 00:06:30,517 - Recv: Bed x: 5.00 y: 20.00 z: 5.98
2020-09-03 00:06:37,903 - Recv: Bed x: 117.00 y: 20.00 z: 6.28
2020-09-03 00:06:45,420 - Recv: Bed x: 229.00 y: 20.00 z: 5.66
2020-09-03 00:06:52,930 - Recv: Bed x: 230.00 y: 145.00 z: 6.17
2020-09-03 00:07:00,316 - Recv: Bed x: 118.00 y: 145.00 z: 6.43
2020-09-03 00:07:07,774 - Recv: Bed x: 6.00 y: 145.00 z: 6.34
2020-09-03 00:07:15,506 - Recv: Bed x: 5.00 y: 270.00 z: 5.74
2020-09-03 00:07:22,821 - Recv: Bed x: 117.00 y: 270.00 z: 6.25
2020-09-03 00:07:30,262 - Recv: Bed x: 229.00 y: 270.00 z: 6.28
2020-09-03 00:07:30,293 - Recv: Eqn coefficients: a: 0.00 b: 0.00 d: 6.05
2020-09-03 00:07:30,305 - Recv: planeNormal x: -0.00 y: -0.00 z: 1.00
2020-09-03 00:07:30,320 - Recv: ok
``` 

so looking at the above, you'll see the following x values: 117. 118, 229, 230, 5, 6 but the y values are collapse and there are only 145, 20, 270. That is a pretty atypical response from what I've experienced historically and the x values should collapse down to just 118, 230, 6 or 117. 229. 5. 

I'm sure that probably doesn't make much sense to you, but the good new is I know what's broke now, I just have to figure out how to adjust the code for it. The only other option is for you to upgrade your firmware to something newer that might not have this bug.

uuys007, 2020-09-03:
Will do on what firmware am I ?

jneilliii, 2020-09-03:
It appears to be marlin 1, but not sure of which specific sub version. I was curious, if you get a chance can you go the terminal tab and run the command `G29 P4 T` and let me know what's returned in that report? I'm still confused why the x values are not aligned with each other and have that 1mm shift during the probing process.

uuys007, 2020-09-04:
2020-09-03 00:41:31,860 - Changing monitoring state from "Offline" to "Opening serial connection"
2020-09-03 00:41:31,881 - Connecting to port /dev/ttyUSB0, baudrate 250000
2020-09-03 00:41:31,918 - Changing monitoring state from "Opening serial connection" to "Connecting"
2020-09-03 00:41:31,928 - Connected to: Serial<id=0x6e6c0970, open=True>(port='/dev/ttyUSB0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2020-09-03 00:41:31,944 - Send: N0 M110 N0*125
2020-09-03 00:41:33,355 - Recv: start
2020-09-03 00:41:33,914 - Recv: echo:Marlin1.0.0
2020-09-03 00:41:33,972 - Recv: echo: Last Updated: Aug 25 2020 00:54:05 | Author: (none, default config)
2020-09-03 00:41:34,081 - Recv: Compiled: Aug 25 2020
2020-09-03 00:41:34,085 - Send: N0 M110 N0*125
2020-09-03 00:41:34,168 - Recv: echo: Free Memory: 3162  PlannerBufferBytes: 1232
2020-09-03 00:41:34,356 - Recv: echo:Hardcoded Default Settings Loaded
2020-09-03 00:41:34,674 - Recv: echo:Steps per unit:
2020-09-03 00:41:34,737 - Recv: echo:  M92 X202.00 Y202.00 Z3200.00 E840.00
2020-09-03 00:41:34,766 - Recv: echo:Maximum feedrates (mm/s):
2020-09-03 00:41:34,782 - Recv: echo:  M203 X300.00 Y300.00 Z15.00 E22.00
2020-09-03 00:41:34,941 - Recv: echo:Maximum Acceleration (mm/s2):
2020-09-03 00:41:35,136 - Recv: echo:  M201 X2000 Y2000 Z100 E2000
2020-09-03 00:41:35,433 - Recv: echo:Acceleration: S=acceleration, T=retract acceleration
2020-09-03 00:41:35,501 - Recv: echo:  M204 S500.00 T500.00
2020-09-03 00:41:35,699 - Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
2020-09-03 00:41:36,003 - Recv: echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
2020-09-03 00:41:36,156 - Recv: echo:Home offset (mm):
2020-09-03 00:41:36,329 - Recv: echo:  M206 X0.00 Y0.00 Z0.00
2020-09-03 00:41:36,366 - Recv: echo:PID settings:
2020-09-03 00:41:36,382 - Recv: echo:   M301 P22.20 I1.08 D114.00
2020-09-03 00:41:36,387 - Recv: echo:SD card ok
2020-09-03 00:41:36,419 - Recv: ok
2020-09-03 00:41:36,431 - Send: N0 M110 N0*125
2020-09-03 00:41:36,437 - Changing monitoring state from "Connecting" to "Operational"
2020-09-03 00:41:36,459 - Recv: ok
2020-09-03 00:41:36,468 - Send: N0 M110 N0*125
2020-09-03 00:41:36,475 - Recv: ok
2020-09-03 00:41:36,481 - Send: N1 M115*39
2020-09-03 00:41:36,494 - Recv: FIRMWARE_NAME:Marlin V1; Sprinter/grbl mashup for gen6 FIRMWARE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Mendel EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
2020-09-03 00:41:36,510 - Recv: ok
2020-09-03 00:41:36,515 - Send: M20
2020-09-03 00:41:36,521 - Recv: Begin file list
2020-09-03 00:41:36,689 - Recv: End file list
2020-09-03 00:41:36,695 - Recv: ok
2020-09-04 07:59:35,202 - Send: M105
2020-09-04 07:59:35,211 - Recv: ok T:16.2 /0.0 B:16.5 /0.0 T0:16.2 /0.0 @:0 B@:0
2020-09-04 07:59:40,193 - Send: M105
2020-09-04 07:59:40,200 - Recv: ok T:16.2 /0.0 B:16.4 /0.0 T0:16.2 /0.0 @:0 B@:0
2020-09-04 07:59:45,199 - Send: M105
2020-09-04 07:59:45,382 - Recv: ok T:16.2 /0.0 B:16.5 /0.0 T0:16.2 /0.0 @:0 B@:0
2020-09-04 07:59:50,208 - Send: M105
2020-09-04 07:59:50,243 - Recv: ok T:16.2 /0.0 B:16.5 /0.0 T0:16.2 /0.0 @:0 B@:0
2020-09-04 07:59:55,232 - Send: M105
2020-09-04 07:59:55,242 - Recv: ok T:16.3 /0.0 B:16.5 /0.0 T0:16.3 /0.0 @:0 B@:0
2020-09-04 08:00:00,231 - Send: M105
2020-09-04 08:00:00,240 - Recv: ok T:16.2 /0.0 B:16.5 /0.0 T0:16.2 /0.0 @:0 B@:0
2020-09-04 08:00:05,292 - Send: M105
2020-09-04 08:00:05,413 - Recv: ok T:16.2 /0.0 B:16.7 /0.0 T0:16.2 /0.0 @:0 B@:0
2020-09-04 08:00:10,296 - Send: M105
2020-09-04 08:00:10,308 - Recv: ok T:16.2 /0.0 B:16.6 /0.0 T0:16.2 /0.0 @:0 B@:0
2020-09-04 08:00:15,328 - Send: M105
2020-09-04 08:00:15,383 - Recv: ok T:16.4 /0.0 B:16.6 /0.0 T0:16.4 /0.0 @:0 B@:0
2020-09-04 08:00:20,330 - Send: M105
2020-09-04 08:00:20,342 - Recv: ok T:16.2 /0.0 B:16.6 /0.0 T0:16.2 /0.0 @:0 B@:0
2020-09-04 08:00:25,331 - Send: M105
2020-09-04 08:00:25,340 - Recv: ok T:16.4 /0.0 B:16.6 /0.0 T0:16.4 /0.0 @:0 B@:0
2020-09-04 08:00:30,381 - Send: M105
2020-09-04 08:00:30,430 - Recv: ok T:16.4 /0.0 B:16.5 /0.0 T0:16.4 /0.0 @:0 B@:0
2020-09-04 08:00:35,383 - Send: M105
2020-09-04 08:00:35,408 - Recv: ok T:16.3 /0.0 B:16.5 /0.0 T0:16.3 /0.0 @:0 B@:0
2020-09-04 08:00:40,391 - Send: M105
2020-09-04 08:00:40,399 - Recv: ok T:16.3 /0.0 B:16.5 /0.0 T0:16.3 /0.0 @:0 B@:0
2020-09-04 08:00:45,387 - Send: M105
2020-09-04 08:00:45,401 - Recv: ok T:16.4 /0.0 B:16.5 /0.0 T0:16.4 /0.0 @:0 B@:0
[... 17 lines cut ...]
2020-09-04 08:01:55,856 - Recv: Bed x: 229.00 y: 20.00 z: 6.18
2020-09-04 08:02:03,409 - Recv: Bed x: 230.00 y: 145.00 z: 6.58
2020-09-04 08:02:10,880 - Recv: Bed x: 118.00 y: 145.00 z: 6.66
2020-09-04 08:02:18,362 - Recv: Bed x: 6.00 y: 145.00 z: 6.51
2020-09-04 08:02:25,946 - Recv: Bed x: 5.00 y: 270.00 z: 6.43
2020-09-04 08:02:33,400 - Recv: Bed x: 117.00 y: 270.00 z: 6.41
2020-09-04 08:02:40,783 - Recv: Bed x: 229.00 y: 270.00 z: 6.74
2020-09-04 08:02:40,803 - Recv: Eqn coefficients: a: -0.00 b: -0.00 d: 6.62
2020-09-04 08:02:40,809 - Recv: planeNormal x: 0.00 y: 0.00 z: 1.00
2020-09-04 08:02:40,816 - Recv: ok
2020-09-04 08:02:40,821 - Recv: echo:endstops hit:  Z:6.74
2020-09-04 08:02:45,408 - Send: M105
2020-09-04 08:02:45,424 - Recv: ok T:16.9 /0.0 B:16.6 /0.0 T0:16.9 /0.0 @:0 B@:0
[... cut ...]
2020-09-04 08:02:55,422 - Recv: ok T:16.9 /0.0 B:16.7 /0.0 T0:16.9 /0.0 @:0 B@:0
2020-09-04 08:03:00,414 - Send: M105
2020-09-04 08:03:00,439 - Recv: ok T:17.3 /0.0 B:17.0 /0.0 T0:17.3 /0.0 @:0 B@:0
2020-09-04 08:03:05,415 - Send: M105
2020-09-04 08:03:05,440 - Recv: ok T:17.2 /0.0 B:16.6 /0.0 T0:17.2 /0.0 @:0 B@:0
2020-09-04 08:03:10,420 - Send: M105
2020-09-04 08:03:10,445 - Recv: ok T:17.1 /0.0 B:16.5 /0.0 T0:17.1 /0.0 @:0 B@:0
2020-09-04 08:03:15,424 - Send: M105
2020-09-04 08:03:15,498 - Recv: ok T:17.2 /0.0 B:16.8 /0.0 T0:17.2 /0.0 @:0 B@:0

uuys007, 2020-09-04:
did this help ?

jneilliii, 2020-09-04:
Nope, looks like it's doing the same exact 3x3 grid and returned the same weird x values not inline with each other. I'm not 100% sure I'll be able to get around this right now but I'm thinking of possible solutions. You may have to see if there is the possibility of upgrading the firmware.

uuys007, 2020-09-05:
Thx for help so far , I will try the firmware need to get myself first clude up with that part

uuys007, 2020-09-10:
i found out im on marlin 1.0.0 RC2 verion

im not so farmiliar with firmware yet  i will be trying 1.1.9 soon will this solve the problem as i have tried fw 2.0  and not getting it so far to work on my printer

uuys007, 2020-09-10:
Send: G29 P4 T
Recv: echo:Home X/Y before Z
Recv: ok

jneilliii, 2020-09-10:
So the plugin's worked for me with Marlin 1.1.9 and above, I'm currently running 2.0.6.1 and everything is working fine for me, but I use UBL, which gives the report in a slightly different format. For your command above, Try doing G28 first and then G29 P4 T and see what comes out. The idea is we're looking for a report that shares the same x position during the probing.

uuys007, 2020-09-10:
Send: G28
Recv: ok
[...]
Send: G29 P4 T
Recv: Bed x: 5.00 y: 20.00 z: 6.56
Recv: Bed x: 117.00 y: 20.00 z: 6.32
Recv: Bed x: 229.00 y: 20.00 z: 6.19
Recv: Bed x: 230.00 y: 145.00 z: 6.44
Recv: Bed x: 118.00 y: 145.00 z: 6.41
Recv: Bed x: 6.00 y: 145.00 z: 6.47
Recv: Bed x: 5.00 y: 270.00 z: 6.11
Recv: Bed x: 117.00 y: 270.00 z: 6.34
Recv: Bed x: 229.00 y: 270.00 z: 6.44
Recv: Eqn coefficients: a: -0.00 b: -0.00 d: 6.41
Recv: planeNormal x: 0.00 y: 0.00 z: 1.00
Recv: ok
Recv: echo:endstops hit:  Z:6.44

jneilliii, 2020-09-10:
Yeah, so same issue, the x points are not aligned for some reason.

uuys007, 2020-09-10:
not sure what you mean by that?

jneilliii, 2020-09-11:
Might be easier to show you graphically what I mean. If you look at the middle row of points being probed the x is offset by 1mm from the top and bottom rows. The `X` in this graph is the actual point from your report and where the red and yellow dashed lines intersect is where they are supposed to be, or at least where they need to be for my plugin to work properly, This is not normal when probing a grid, because it's not actually probing the correct grid points (where the lines intersect). You may have to zoom in on this image to see it completely.

[image]

uuys007, 2020-09-11:
i have got marlin 1.1.9 now loaded but visualiser still hanging

do ii need to to send you

uuys007, 2020-09-11:
[image]

jneilliii, 2020-09-11:
I need to at least see the output of the leveling command again.

```
G28
G29
```

Fusseldieb, 2020-09-11:
Marlin 2.x user here. Faced the same or similar problem.

Stuck on

`Please wait, retrieving current mesh...`

Tried all commands listed in this thread, doesn't work.

Even if I try `G29 P4 T` my printer still probes three points, as usual, but the mesh reporting never concludes.

I receive following after running this command:

`Recv: X:102.53 Y:185.05 Z:9.26 E:0.00 Count A:23115 B:6633 Z:4177
Recv: ok`

[9 more replies]

closed 2020-09-11 by jneilliii