jneilliii/OctoPrint-BedLevelVisualizer #291: SyntaxError: Unexpected token . in JSON at position 3

PinOkeyOh, 2020-09-02:
**Describe the bug**
My visualizer was working great, except that i couldn't change the z values of my graph to get a more accurate reading of my bed. I was looking through other plugins and after downloading a few my visualizer was still working but i could still not get the z values to change on my graph. Im positive i was updating the intergers in the correct space(graph z limits) but i could be wrong.    

After searching around the plugins some more i downloaded the Themify plugin and after that i started receiving the error in my octoprint window. Now my printer will go through the auto bed leveling process and will update the mesh. But the actual visual graph will not show up and i receive the error in my title. I have enabled debug logging so i will update that in here. 
 
I am using the commands
G28 
M155 S30    
@BEDLEVELVISUALIZER 
G29 T   
M155 S3

**Debug Log**

[plugin_bedlevelvisualizer_debug (2).log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/5160676/plugin_bedlevelvisualizer_debug.2.log)

**Screenshots**
[image]
[image]
[image]

**Firmware and Version**
I am using the most up to date Tiny Machines firmware on my cr10s-pro v2.

Any help would be appreciated.

jneilliii, 2020-09-02:
Your debug log indicates it's collecting the data properly. I'm not sure why this would be happening all of a sudden. I use Themeify and am not aware of any conflicts with that plugin. If you open the developer console of your browser (F12) are there any errors when you initially load the page, or when switching to the Bed Visualizer tab or when updating mesh?

PinOkeyOh, 2020-09-02:
I get the same error as soon as i click the “bed visualizer” tab. I will try the f12 thing when i get off work tonight. Thank you

PinOkeyOh, 2020-09-03:
Thank you for the quick reply, I ended up uninstalling and rebooting then reinstalling and rebooting and it works! along with the Graph Z limits changing now! Thanks!

jneilliii, 2020-09-03:
Ok, closing this then. Feel free to comment here again if it comes back.

closed 2020-09-03 by jneilliii