jneilliii/OctoPrint-BedLevelVisualizer #293: [BUG]:

Andy-Prusa, 2020-09-09:
**Describe the bug**
When I run bed visualiser the output appears to hold no relation to the print test outcomes. E.g. where visualiser shows the bed farthest away is where the test print is squashed and rough. See attached image, front right is low but is squished, middle is higher, but prints too far away

**Expected behavior**
Bed visualiser output would reflect what I find in printing

**Debug Log**

Done

**Firmware and Version**

?latest

[
[image]
[192.168.0.3%2Fdownloads%2Flogs%2Fplugin_bedlevelvisualizer_debug.log(1).pdf](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/5192205/192.168.0.3.2Fdownloads.2Flogs.2Fplugin_bedlevelvisualizer_debug.log.1.pdf)
[image]

](url)

jneilliii, 2020-09-09:
This is what the flip and rotate options are for in the plugin's settings. If for some reason the output is unexpected relative to the orientation of your bed's axis you have to kind of play with those until it's right. In your case I think you would flip both x and y, or potentially just rotate by 180 degrees.

Lumute, 2020-09-27:
I'm using an Ender 5 Pro which has 0,0 at the Front-Left. I wanted to rotate the graph 60 degrees so its more intuitive but these settings in the general tab do not seem to be doing anything. I didn't start a new issue because maybe I'm not understanding the concept? or maybe I don't know how to make the new settings apply? or is it necessary to update the mesh again to get the graph to show using the new settings?

I tries Ctrl+F5 and the graph shows exactly the same, I even tried restarting Octoprint + Ctrl+F5 and same. Also tried other settings like flipping x and y axis and nothing I don't see any difference in the graph... 

Everything I try it always displays the graph with 0,0 in the front like images above...

jneilliii, 2020-09-27:
Those settings effect how the data is interpreted during the update mesh process. Only a couple of the settings will directly effect the graph without updating the mesh using the button.

jneilliii, 2020-09-27:
Oh, and the graph itself will always begin its rendering with 0,0 facing you directly.

Lumute, 2020-09-28:
Ah, yes I completely misunderstood the concept... so I guess the idea then is to use those options so that the coordinates the plugin assigns to each measurement received match the actual coordinates where the measurements were taken, correct?

Any idea how to practically figure that out? I I could put something on top of the bed on one of the corners and update the mesh to see which one has a very high value and flip until correct, or is there a better way?

Andy-Prusa, 2020-09-28:
So I rotated mine 90 degrees after checking by doing what you said, ie put a piece of paper under one corner. The issue I still have is where it says it is too low it’s actually too high. Not sure what to do with this?

Lumute, 2020-09-28:
Not sure if this is the same problem you are having but I had a similar issue because on my printer (Ender 5 Pro) Z axis is reversed (0 at the top and goes to 300 as the bed goes down).

So in my case, the graph is reversed (positive values up, negative own) and I have to do the opposite to what the graph shows, not very intuitive... I think the solution for this would be some separate settings that would control the way the graph is rendered so once you have your coordinates matching correctly you could flip the graph axis.

jneilliii, 2020-09-28:
As I replied in the other FR, copying here for the benefit of others...there's this little move nozzle button that allows you to enable that option and then click on a point on the surface graph and the nozzle will move to that position. Which can assist on figure out what flip and rotate options are necessary to get your graph aligned with the real world layout of your bed.

[image]

I'm still curious if this is related to core x/y motion versus other cartesian style movement. My assumption still is that if you slice a gcode file and look in it with a text editor it's not moving the z axis in the negative direction, it's still sends it as positive, but please let me know if that's not the case.

Lumute, 2020-09-28:
Indeed, moving the nozzle with the graph is quite useful to figure out what is what, still, the trick with a prop in one corner helps confirm in case an axis is interpreted wrong and needs to be flipped.

About the Z axis, in the Ender 5 as you print the Z increases in the positive direction (because the bed goes down) but it is possible to go a little bit to the negative direction before you crash into the nozzle, that is mainly useful to level and calculate the Z Offset.

closed 2020-10-24 by stale[bot]