jneilliii/OctoPrint-BedLevelVisualizer #294: Home XYZ in other languages don't get recognized

jneilliii, 2020-09-11:
Marlin 2.x user here. Faced the same or similar problem.

Stuck on

`Please wait, retrieving current mesh...`

Tried all commands listed in this thread, doesn't work.

Even if I try `G29 P4 T` my printer **still probes three points**, as usual, but the mesh reporting never concludes.

I receive following after running this command:

```
Recv: X:102.53 Y:185.05 Z:9.26 E:0.00 Count A:23115 B:6633 Z:4177
Recv: ok
```

Three-point probing as seen in image:
[image]
Begins in bottom left, goes to right, then top.

**EDIT:** Also, be aware to handle "Home XYZ first" messages in other languages! Mine reports in my language (PT-BR) :
```
Send: G29 P4 T
Recv: echo:Home XYZ Primeiro
Recv: ok
```
You can probably catch this message by searching for "XYZ". Currently the plugin doesn't recognize it and still waits.

Regex suggestion:

```echo:.*?XYZ```

_Originally posted by @Fusseldieb in https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/290#issuecomment-691214421_

RedCoder09, 2020-09-22:
You... made this plugin though.

jneilliii, 2020-09-22:
lol, yeah it was a comment that came over from the lined issue where it was originally posted, but a different unrelated issue.