jneilliii/OctoPrint-BedLevelVisualizer #296: Bed Visualizer non-functional after OctoPrint Python 3 upgrade

MontyTheMooch, 2020-09-12:
Ran across this this afternoon after running the Python 3 upgrade for OctoPrint this afternoon (https://octoprint.org/blog/2020/09/10/upgrade-to-py3/?utm_source=octoprint&utm_medium=announcements&utm_content=1.4.2).

Bed Visualizer fails to load even though the Python 3 verification passed the plugin for upgrade. Upon reinstalling the plugin after the Python 3 upgrade the Bed Visualizer plugin reports as "unknown" on install and does not show up in the installed plugins list even though reporting a successful installation.

jneilliii, 2020-09-13:
That python upgrade script also warns you of the issue and links you to the known issues on the readme here, try this and see if it resolves your issue.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer#known-issues

5teveO, 2020-09-16:
I followed the issue - followed the instructions and its all working here.. Nice job!

pediegoivan, 2020-09-20:
It works for me!

jneilliii, 2020-09-20:
Thanks guys. I'm going to go ahead and close this issue out since it appears not to be an issue with the work around in known issues.

closed 2020-09-20 by jneilliii

MontyTheMooch, 2020-09-20:
Just getting back to this after being on the road for work. Ran through the steps and have it working now. Thanks!