jneilliii/OctoPrint-BedLevelVisualizer #297: [BUG]: Bed Visualizer does not appear in list of installed plugins

tpsherrill, 2020-09-14:
**Describe the bug**
I've installed Bed Visualizer a few times and rebooted OctoPrint. It shows successful but is not listed in the plugins list and is not available in the settings menu.

**Expected behavior**
Bed Level Visualizer should be listed in the menu system.

**Debug Log**
[BedLevelVisualizer-install.txt](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/5219886/BedLevelVisualizer-install.txt)

**Screenshots**
No settings link present. 
[image]

**Firmware and Version**
Ender 3 Pro with Marlin 2

tpsherrill, 2020-09-14:
I had issues with this in a previous try. My challenges with build 14 were due to the documented dependency issue with Python 3:
Running
`sudo apt install libgfortran5 libatlas3-base libatlas-base-dev`
fixed the issue.

jneilliii, 2020-09-14:
Thanks, that's why I had posted that in the known issues, glad you found it.

closed 2020-09-14 by jneilliii