jneilliii/OctoPrint-BedLevelVisualizer #299: [BUG]: Processing BML results fails due to IndexError when processing gcode

mafredri, 2020-09-23:
**Describe the bug**

The plugin fails / hangs when processing bed mesh leveling results on OctoPrint upgraded to Python 3 with the following error:

```
2020-09-23 15:02:39,217 - octoprint.util.comm - ERROR - Error while processing hook bedlevelvisualizer:
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 2952, in _readline
    ret = hook(self, ret)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_bedlevelvisualizer/__init__.py", line 308, in process_gcode
    self.mesh = np.subtract(self.mesh, self.mesh[len(self.mesh[0]) / 2, len(self.mesh) / 2],
IndexError: only integers, slices (`:`), ellipsis (`...`), numpy.newaxis (`None`) and integer or boolean arrays are valid indices
```

I have followed the instructions in the readme and installed additional packages and I've deleted (+data) and reinstalled the plugin. Neither affected the issue the issue.

Cancel button remains visible after results.

**Expected behavior**

Results are processed.

**Debug Log**
[octoprint.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/5268666/octoprint.log)
[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/5268667/plugin_bedlevelvisualizer_debug.log)

**Screenshots**

[image]

**Firmware and Version**

Original Prusa MK3S 3.9.1

jneilliii, 2020-09-23:
Quick question, why are you using center origin on a non-delta printer?

mafredri, 2020-09-23:
To be honest, no other reason than me finding center origin to make the most sense when comparing the level between different sides of the bed. Your question did prompt me to try without center origin and then there were no issues. But I've been using center origin without issue until I upgraded to python 3 so this didn't dawn on me. Happy to at least have a work-around for the time-being, thanks.

jneilliii, 2020-09-23:
There were changes to the last version IIRC that effected center origin, because it was specific changes being made for better delta support, but that may have broken yours. I'll look into that at some point and see what might be happening here.

jneilliii, 2020-11-12:
@mafredri can you test 1.0.0rc4 version for me to see if resolves your issue with center origin? Copy/paste the URL into plugin manager > get more > ...from URL and click install.

```
https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/1.0.0rc4.zip
```

mafredri, 2020-11-12:
Thanks @jneilliii, seems to be working as it should now 👍🏻

closed 2020-11-21 by jneilliii