jneilliii/OctoPrint-BedLevelVisualizer #301: [BUG]: Color-coding of x-Position and y-Position labels are reversed

tpsherrill, 2020-09-24:
**Describe the bug**
The labels on the Current Mesh Data and Angle Corrections tabs show the x and y axis labels switched. My understanding is that the XY coordinates are standard (x along the bottom, y going up the left edge). The color coding shows the bottom axis as 'y - Position' and the vertical axis the 'x - Position'
[image]

The CSS class on the vertical and horizontal axis labels should be reversed. The vertical axis is getting the x-axis-label class.
css: x-axis-label, y-axis-label
[image]

**Expected behavior**
The bold/black and bold/green coloring on the labels should be switched.

**Debug Log**

**Screenshots**

**Firmware and Version**

tpsherrill, 2020-09-24:
As a new user to mesh bed leveling, this was very difficult to get my head around since the labels were backwards. With different printers using different origin strategies/home locations and the somewhat cryptic marlin output from the mesh levels in terminal, I struggled to follow what was happening. I think I have figured this out experimentally...

There may be another related issue with the buttons in the settings for 'Flip X Axis' and 'Flip Y Axis'

If this isn't an issue, then I need an orientation on how to read the table. Thanks for your great plugin!

jneilliii, 2020-09-24:
This is actually a duplicate of #278 and has already been resolved in my current development branch.

tpsherrill, 2020-09-24:
Good to know. I reviewed the open issues before posting. Did you catch the issue with the item I mentioned in the comment?
_There may be another related issue with the buttons in the settings for 'Flip X Axis' and 'Flip Y Axis'_
I believe they are also reversed.

jneilliii, 2020-09-24:
There's two different flip options, one in the correction area ans one in the normal area, related to how the data is collected. The ones in the first tab should have no effect on the corrections.

jneilliii, 2020-10-03:
fixed the label colors in version 0.1.15

closed 2020-10-03 by jneilliii