jneilliii/OctoPrint-BedLevelVisualizer #302: Installer stuck.  "Building wheel for numpy (setup.py): still running..."

ineedhungry, 2020-09-25:
Installing plugin from https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install file:///tmp/tmp9PuJwJ/OctoPrint-BedLevelVisualizer-master.zip --no-cache-dir
/home/pi/oprint/local/lib/python2.7/site-packages/OpenSSL/crypto.py:12: CryptographyDeprecationWarning: Python 2 is no longer supported by the Python core team. Support for it is now deprecated in cryptography, and will be removed in a future release.
from cryptography import x509
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmp9PuJwJ/OctoPrint-BedLevelVisualizer-master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from Bed-Visualizer==0.1.14) (1.4.2)
Collecting numpy
Downloading https://files.pythonhosted.org/packages/b7/6f/24647f014eef9b67a24adfcbcd4f4928349b4a0f8393b3d7fe648d4d2de3/numpy-1.16.6.zip (5.1MB)
Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.0.0)
Requirement already satisfied: psutil<6,>=5.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (5.7.2)
Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (0.5.4)
Requirement already satisfied: monotonic<2,>=1.5; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.5)
Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.0.7)
Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (5.3.1)
Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (0.10.9)
Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (0.18.2)
Requirement already satisfied: feedparser<6,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (5.2.1)
Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (0.7.19)
Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (2.24.0)
Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (7.1.2)
Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.4)
Requirement already satisfied: watchdog<1,>=0.10.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (0.10.3)
Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.1.0)
Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.5.0.1)
Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (2.11.2)
Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.2)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (0.5.0)
Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.1.2)
Requirement already satisfied: rsa==4.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (4.0)
Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (2.8.5)
Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (0.4.21)
Requirement already satisfied: OctoPrint-FileCheck>=2020.07.06 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (2020.8.7)
Requirement already satisfied: futures<4,>=3.3; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (3.3.0)
Requirement already satisfied: websocket-client<1,>=0.57 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (0.57.0)
Requirement already satisfied: scandir<2,>=1.10; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.10.0)
Requirement already satisfied: tornado==5.1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (5.1.1)
Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (3.4)
Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.1.1)
Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (3.1.1)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (2019.8.19)
Requirement already satisfied: chainmap<2,>=1.0.3; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.0.3)
Requirement already satisfied: typing<4,>=3.7.4.1; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (3.7.4.1)
Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (0.1.5.post0)
Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (0.17.4)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.06.22 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (2020.9.23)
Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.2.0)
Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.0.1)
Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (1.12.1)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (2.0)
Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.14) (0.1)
Requirement already satisfied: Babel>=2.3 in /home/pi/oprint/lib/python2.7/site-packages (from Flask-Babel<2,>=1.0->OctoPrint->Bed-Visualizer==0.1.14) (2.7.0)
Requirement already satisfied: pytz in /home/pi/oprint/lib/python2.7/site-packages (from Flask-Babel<2,>=1.0->OctoPrint->Bed-Visualizer==0.1.14) (2019.3)
Requirement already satisfied: idna<3,>=2.5 in /home/pi/oprint/lib/python2.7/site-packages (from requests<3,>=2.23.0->OctoPrint->Bed-Visualizer==0.1.14) (2.8)
Requirement already satisfied: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in /home/pi/oprint/lib/python2.7/site-packages (from requests<3,>=2.23.0->OctoPrint->Bed-Visualizer==0.1.14) (1.25.6)
Requirement already satisfied: certifi>=2017.4.17 in /home/pi/oprint/lib/python2.7/site-packages (from requests<3,>=2.23.0->OctoPrint->Bed-Visualizer==0.1.14) (2019.9.11)
Requirement already satisfied: chardet<4,>=3.0.2 in /home/pi/oprint/lib/python2.7/site-packages (from requests<3,>=2.23.0->OctoPrint->Bed-Visualizer==0.1.14) (3.0.4)
Requirement already satisfied: pathtools>=0.1.1 in /home/pi/oprint/lib/python2.7/site-packages (from watchdog<1,>=0.10.2->OctoPrint->Bed-Visualizer==0.1.14) (0.1.2)
[... cut ...]
Stored in directory: /tmp/pip-ephem-wheel-cache-0XTher/wheels/4b/fc/16/9ce6a570aefa5d4c7fc2b1c87d6bd40aee71d8a41f47964bc8
Building wheel for numpy (setup.py): started
Building wheel for numpy (setup.py): still running...
Building wheel for numpy (setup.py): still running...
Building wheel for numpy (setup.py): still running...
Building wheel for numpy (setup.py): still running...
Building wheel for numpy (setup.py): still running...
Building wheel for numpy (setup.py): still running...
Building wheel for numpy (setup.py): still running...
Building wheel for numpy (setup.py): still running...

jneilliii, 2020-09-25:
I have only heard of the compiling of the numpy install failing on low memory devices like the Pi A. Otherwsie, yes the compile takes a long time and is not actually stuck unless it locks up your machine entirely. This is also documented in the [Known Issues](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer#known-issues). Just let it run and it will eventually complete.

closed 2020-09-25 by jneilliii

ineedhungry, 2020-09-25:
This is what happens letting it run though. 

Error!
Could not install plugin, unknown error, please consult octoprint.log for details

jneilliii, 2020-09-25:
Ok, you may have to SSH to your Pi and manually install the numpy requirement ahead of time then. To do that run this commands on your pi.

`~/oprint/bin/pip install numpy`

Then once that install/compile completes you should be able to install the plugin without issue.

Sllarsen, 2020-12-09:
I ran the above command and my cmdline errors out. 
[image]

I get a big block of red text and then I get this 
[image]

Do you have any idea why numpy isn't installing? I believe this is why my plug in won't download :(

jneilliii, 2020-12-09:
I think you might be missing the necessary build tools that are typically installed in an octopi installation. This could also be related to system resources or not enough memory. What type of device are you trying to install on? You might be able to upgrade your python environment to version 3 using [these](https://github.com/cp2004/Octoprint-Upgrade-To-Py3) instructions, and then running the command `sudo apt install libatlas3-base` , and then the numpy install should pull from piwheels and be pre-compiled and ready to go.

Sllarsen, 2020-12-09:
I'm currently running octopi on a raspberry pi 3b+

jneilliii, 2020-12-09:
I can only recommend to create a backup from your existing octoprint instance and flash octopi 0.18rc1 and then install. I have found it to be very stable from. It would put you on the python 3 implementation instantly and the install should just work. Otherwise you could try the upgrade process I linked previously and see if that's any difference. The only other thing I can think to try is installing the apt version using `sudo apt install numpy` if that's even available.