jneilliii/OctoPrint-BedLevelVisualizer #303: [FR]: Add Z Axis Flip

Lumute, 2020-09-27:
First let me thank you for the great plugin!

I was trying to adjust my leveling using this plugin kept adjusting the wrong way and after a couple of tries realized that the Ender 5's have the Z axis inverted (0 at the top and moves down to 300 as you print).

Settings have options to invert X and Y axis but not Z, could this be please added?

Thanks!

jneilliii, 2020-09-27:
I'm not quite sure I understand how that would effect the graph. Technically even if the bed is moving down isn't that still a positive move in gcode? If you're referring to the adjustments there is an option to reverse the direction of the screws.

Lumute, 2020-09-28:
Yes, I misunderstood and though the settings were there to change the way the graph is rendered, not to the way the measurements are interpreted...

I guess as it is this request does not make sense at all. It would be ideal if we could change the way the graph is rendered so we could at least flip the axis, that way when you see one of the corners is low you can lower go adjust that corner a little bit up. Right know is quite a challenge to figure out which corner matches which and which way you need to adjust the bed...

jneilliii, 2020-09-28:
It's hard for me not to understand how the graph works since I made it and understand how the data is getting rendered. Do you mean you'd like to configure the starting angle of the "camera's position" so to say? 

What about if the correction matrix option is enabled, does that make more sense?

[image]

You know you can interact with the graph as well right? There's also the little move nozzle button that allows you to enable that option and then click on a point on the surface graph and the nozzle will move to that position. Which also can assist on figure out what flip and rotate options are necessary to get your graph aligned with the real world layout of your bed.

[image]

Lumute, 2020-09-28:
"Do you mean you'd like to configure the starting angle of the "camera's position" so to say?"

Yes that would be cool, but as you say, you can move / rotate the graph with those controls above its not tat critical, whats more important is to invert an axis, specifically the Z in my case. So 0 in the middle, negative values going up, positive values going down...

"What about if the correction matrix option is enabled, does that make more sense?"

Yes, but not as it, it would help a lot if apart from changing the x and y direction we could also swap them (y on the column, x on the row), that would perfectly line up with the way we see our bed in real life...

jneilliii, 2020-09-28:
I'm still trying to figure out why I'm so confused. Am I incorrect in my assumption of cartesian movement on a core x/y machine? It doesn't matter that the bed is going down because the positioning is still a relative distance from the nozzle's tip is it not? If it is, I totally want to fix this for you guys, but I've never had anyone with a core x/y machine bring this up before that I can remember. If you enter the commands below in the terminal command which way does the nozzle move relative to the bed?

```
G28
G1 Z5
```

jneilliii, 2020-09-28:
It might also help me understand the positioning on your ender if I could get a photo from the "front" of the machine with the nozzle at the following two positions...

x = 0, y = 0 and x = 200, y = 200

Lumute, 2020-09-28:
Lol,  you wrote this while I was taking the picture... this is how we see our bed:

[image]

jneilliii, 2020-09-28:
Ok, perfect. Thank you. That's exactly how I see my axis relative to the front of my CR-10. I think maybe the confusion here is the labeling of the correction matrix being incorrectly colored, which is prepped to be fixed in the next release. When you're looking at the correction grid the colors are incorrect on the labels, relative tickets #301 and #278. Sorry for the confusion.

jneilliii, 2020-09-28:
If you want to try out the fixed version you can copy/paste the URL below into plugin manager > get more > ...from URL and click install.

```
https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.15.zip
```

Lumute, 2020-09-28:
Ah, yes this is perfect!, no need to swap x & y and it does help visualizing it.

But the Z axis is still inverted, I guess that would the the main request here, an option to make the graph render with inverted Z.

jneilliii, 2020-09-28:
Would this help make it clearer? The grid is not currently configured as a representation of your print volume, just a representation of the probed positions relative to what 0 should be. 

[image]

jneilliii, 2020-09-28:
The reason it's zoomed into those offsets rather than visualizing the print volume as part of the grid is because of this.

[image]

If you ignore the actual color scale, you should be able to see that because the z is so high the variation from the probed points is reduced down so far you can't really see what is happening. Both this screenshot and the previous one are using the same exact probe points.

Lumute, 2020-09-28:
My graph looks very similar to the one above. I understand what you mean about zooming and yes showing only from -2 to 2 in Z helps a lot visualizing it, no need to see the whole print volume.

Still, on that first graph, the point at 0,0 is visually below the Z = 0 plane and looks like it needs to be raised but in our printer it actually needs to be lowered...

mrseeker, 2020-09-28:
Okay, after a long discussion on Discord with jneilli, I came to the conclusion that the carthesian axis are based on the bed, not the nozzle. So, if you put an imaginary cube of 20mm on the bed, it would sit on top of the build plate and not under it. Hope this makes more sense?

Lumute, 2020-09-28:
Not sure I understand the point you are trying to make, yes the plane shown in the graph is a visual representation of the bed and yes if you put that object on the bed then it sits on top. 

BUT, if leave the object there and run an Mesh Update (G29), the mesh built will think the bed is 20mm warped in that place.

Lets say you placed the object in (0,0): on very flat and leveled Ender 3 or CR-10, a 4 point mesh it would look like this:

(0,0) = 20   (220,0) = 0
(0,220) = 0   (220,220) = 0

The graph will clearly show the (0,0) point raised above the z=0 plane and you clearly understand the bed is too high on the (0,0) corned and go lower it.

On an Ender 5 the mesh will look like this:

(0,0) = **-20**   (220,0) = 0
(0,220) = 0   (220,220) = 0

Currently the graph will clearly show the (0,0) point way below the z=0 plane and you would think you need to go raise it when in reality you need to also lower it.

That is why for printers with inverted Z axis like the Ender 5 it would be ideal if the graphic can be rendered with the Z axis inverted, so the visual representation of the bed matches the reality, otherwise its quite confusing...

Lumute, 2020-09-28:
I guess if changing the way the graph is rendered is not possible or too difficult, a simple workaround for this would be to multiply every mesh value by -1 during interpretation to literally invert them.

This has the downside than when you look at the mesh or hover over a point in the graph the value displayed will be incorrect but the actual graph will look the way it needs to look, the plugin is called "BedLevel**Vizualizer**" so I this is more important...

jneilliii, 2020-09-28:
maybe some backing data to your statement would help. can you post a screenshot of this and the correlating mesh data that is reported from your printer? Even better, enabled debug logging, restart octoprint, and do an update mesh process with that block at a probing position and show me where it returns -20 in the file?

Lumute, 2020-09-28:
AGR!!! I need to eat my hat now... It is displayed exactly as it needs to be displayed. I think what happened is the bug in 1.14 about the axis colors confused me and to correct I had the bed rotated in the settings, I fixed that later after installing 1.15 and didn't realized I was looking at the wrong point in the mesh before... 

So sorry I wasted your time guys, feel like an idiot now... lol... Evertythin is looking the way it needs to look.

I would say for new dumb uses like me it would be good to add more info to the settings like specifying which ones affect interpretation and which ones affect the graph. Also the fixes in the new version make it much clearer.

jneilliii, 2020-09-28:
Awesome @Lumute thanks for the update. I guess I really need to rush the release of that new version with those labels corrected, because it definitely seems to be causing some unneeded confusion, not just for you.

closed 2020-09-28 by jneilliii