jneilliii/OctoPrint-BedLevelVisualizer #305: can't install BedLevelVisualizer into raspberry pi 3b

juavinas, 2020-10-03:
Hello Can't install to raspberry pi 3b

pi@octoprint:~ $ python3 --version
Python 3.7.3
pi@octoprint:~ $ python --version
Python 2.7.16

pi@octoprint:~ $ pip install "https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip"
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip
  Using cached https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip (2.7 MB)
    ERROR: Command errored out with exit status 255:
     command: /usr/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-pCl9g0/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-pCl9g0/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-EmhEip
         cwd: /tmp/pip-req-build-pCl9g0/
    Complete output (1 lines):
    Could not import OctoPrint's setuptools, are you sure you are running that under the same python installation that OctoPrint is installed under?
    ----------------------------------------
ERROR: Command errored out with exit status 255: python setup.py egg_info Check the logs for full command output.

Rjbsnxt, 2020-10-03:
I too have the deprecation warning and Python version errors on a Raspberry Pi 3B. 
I have updated OctoPrint to Python 3 recently.

jneilliii, 2020-10-03:
@Rjbsnxt for python 3 there is a known issue posted on the readme [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer#known-issues). Just run that command and you should be fine.

For your error @juavinas how are you installing the plugin? Because it looks like you might be doing it manually and your not installing within the virtualenv.

Rjbsnxt, 2020-10-03:
@jneilliii 
Thanks for that! 
It was the failed install from the Plugin Manager that lead me to try to install from the CLI ( "pip install....") which in turn gave me the exact same error messages as @juavinas, even after installing libatlas3.
<code> sudo apt install libatlas3-base </code> followed by installing from within the Plugin Manager in OctoPrint worked.

juavinas, 2020-10-04:
Thank you, this will helped me too.

Ps.: we can close this discussion.

closed 2020-10-04 by juavinas