jneilliii/OctoPrint-BedLevelVisualizer #311: ipad 12.9 ios 14.0.1 Webgl is not supported

ingo916, 2020-10-13:
ipad 12.9 ios 14.0.1 error Webgl is not supported by your browser - visit http://get.webgl.org for more info
Works fine with my iphone ios 14.0.1

visiting the link says my browser is supported and a rotating cube is displayed.

**Debug Log**
will include this log soon.

**Screenshots**
[image]
[image]

**Firmware and Version**
OctoPrint 1.4.2 Python 3.7.3 OctoPi 0.17.0

jneilliii, 2020-10-13:
do you have the temps graph plugin installed or any other plugin that includes graphing features? I just looked and the version of plotly that is included with the plugin should already be patched for iPad 12.9 so not sure if it's a ios version issue or safari version issue possibly. 

https://github.com/plotly/plotly.js/issues/4502

ingo916, 2020-10-13:
> plotly/plotly.js#4502

These are the plugins I have installed

Ingo Azarvand
209-207-2243
Licensed FFL Dealer

jneilliii, 2020-10-13:
Lol, images don't come over in email replies.

jneilliii, 2020-10-13:
Ah, I might need to check tplink. If you disable that plugin temporarily will this plugin work for you?

ingo916, 2020-10-13:
disabling the TPLink plugin worked! Please let me know if you have a solution that doesn't require disabling the TPLink plugin.

jneilliii, 2020-10-13:
I just need to add the latest plotly library to that plugin. I can do that tonight when I get home and provide a link you can use to install a pre-release version to use.

jneilliii, 2020-10-14:
I've added the same plotly version to TPLink plugin. Install it by copying/pasting the URL below into plugin manager > Get More > ...from URL and dclick Install.

```
https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.25.zip
```

ingo916, 2020-10-14:
Thanks the updated TPLinkSmartplug update you provided resolved my issue. Its all working great now!!!

closed 2020-10-14 by ingo916

ingo916, 2020-10-14:
FYI after installing the 0.9.25.zip i can no longer manage my plug using the plugin. I deleted the plug hoping that i just needed to recreate it and I cannot. tried adding the plug and nothing happens. I uninstalled the plugin and removed all data, reinstalled using the link you provided and still cannot add a plug. I included a video attached.

[2020-10-14_12-58-34.zip](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/5380570/2020-10-14_12-58-34.zip)

jneilliii, 2020-10-15:
Sorry about that.  I just pushed an update that should fix the adding issue, you can just reinstall using the same URL as before. Should show up as 0.9.25rc2. If it still doesn't work please open an issue on that repository. Thanks.