jneilliii/OctoPrint-BedLevelVisualizer #312: [BUG]: Clearing the rotation field causes an error

poptix, 2020-10-18:
**Describe the bug**
I set the rotation, decided I didn't like it, then cleared the settings box and clicked save. Everything seems fine, but after clicking update mesh it just stalls forever

**Expected behavior**
An empty value should reset to the default value

**Debug Log**
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2952, in _readline
    ret = hook(self, ret)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bedlevelvisualizer/__init__.py", line 313, in process_gcode
    if int(self._settings.get_int(["rotation"])) > 0:
TypeError: int() argument must be a string or a number, not 'NoneType'

poptix, 2020-10-18:
I wouldn't be surprised if this happens with the other settings.

jneilliii, 2020-10-18:
This actually sounds like your config.yaml settings for bedlevelvisualizer section got corrupted. If you SSH to your pi and run the command below and then use the arrows on your keyboard to scroll down to the bedlevelivisualizer section and look to see if the whole section is set to null. If it is completely remove the bedlevelvisualizer section.

```
nano ~/.octoprint/config.yaml
```

if you make a change press ctrl+x to close and then answer y to save and keep the same filename by hitting return. then restart OctoPrint.

poptix, 2020-10-18:
It's not that, I can reproduce it in the web UI. Select whatever you have in the rotation box, delete it, click save. It saves a null/empty value and causes the reported error.

jneilliii, 2020-10-18:
Oh, right...thanks. That's something that I thought I had already fixed. I'll make sure to get that in the next release.

jneilliii, 2020-10-18:
Fixed by above commit, will be included in next release.

closed 2020-11-21 by jneilliii