jneilliii/OctoPrint-BedLevelVisualizer #313: Cannot install plugin version 0.1.15 under fresh build of OctoPrint 1.4.2

lmcbmai, 2020-10-19:
**Describe the bug**
Rather than upgrading my working Octoprint 1.3.12, I started fresh from Octopi 0.17.0 on new SD card, then upgraded it from  Octoprint 1.3.12 to 1.4.2 and migrated to Python 3 as described here: https://octoprint.org/blog/2020/09/10/upgrade-to-py3/

Everything worked without issues. Then using 1.4.2 Plugin Manager I started installing the plugins that I had on my old Octoprint. Bed Visualizer was the first in the list. It installed successfully, but when Plugin management log popped up it showed "Install _Unknown_" and after restart the plugin was absent from the UI in both places - frontend tab was missing and under Settings it was not in the list of installed plugins. So I cannot even uninstall it. I tried twice and the log from the 2nd attempt is attached.  Other Python 3 compatible plugins installed without issues.

**Expected behavior**
Plugin management log popup should show installed plugin name and the plugin should be present in UI after installation

**Debug Log**
```
2020-10-18 23:33:13,537   /home/pi/oprint/bin/python -m pip --disable-pip-version-check install file:///tmp/tmpot0vde6h/OctoPrint-BedLevelVisualizer-master.zip --no-cache-dir
2020-10-18 23:33:15,772 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2020-10-18 23:33:15,773 > Processing /tmp/tmpot0vde6h/OctoPrint-BedLevelVisualizer-master.zip
2020-10-18 23:33:18,828 > Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from Bed-Visualizer==0.1.15) (1.4.2)
2020-10-18 23:33:18,829 > Requirement already satisfied: numpy<=1.19.2,>=1.16.0 in /home/pi/oprint/lib/python3.7/site-packages (from Bed-Visualizer==0.1.15) (1.19.2)
2020-10-18 23:33:18,829 > Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (0.5.0)
2020-10-18 23:33:18,829 > Requirement already satisfied: watchdog<1,>=0.10.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (0.10.3)
2020-10-18 23:33:18,829 > Requirement already satisfied: rsa==4.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (4.0)
2020-10-18 23:33:18,830 > Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (0.4.21)
2020-10-18 23:33:18,830 > Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (1.0.1)
2020-10-18 23:33:18,830 > Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (0.10.9)
2020-10-18 23:33:18,830 > Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (0.18.2)
2020-10-18 23:33:18,831 > Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (1.5.0.1)
2020-10-18 23:33:18,831 > Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (1.12.1)
2020-10-18 23:33:18,831 > Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (1.0.0)
2020-10-18 23:33:18,831 > Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (5.3.1)
2020-10-18 23:33:18,831 > Requirement already satisfied: psutil<6,>=5.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (5.7.2)
2020-10-18 23:33:18,832 > Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (2.24.0)
2020-10-18 23:33:18,832 > Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (0.1.1)
2020-10-18 23:33:18,832 > Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (2020.10.15)
2020-10-18 23:33:18,832 > Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (3.4)
2020-10-18 23:33:18,833 > Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (7.1.2)
2020-10-18 23:33:18,833 > Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (2.8.5)
2020-10-18 23:33:18,833 > Requirement already satisfied: tornado==5.1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (5.1.1)
2020-10-18 23:33:18,833 > Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (0.6.0)
2020-10-18 23:33:18,833 > Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (1.2)
2020-10-18 23:33:18,834 > Requirement already satisfied: OctoPrint-FileCheck>=2020.07.06 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (2020.8.7)
2020-10-18 23:33:18,834 > Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.06.22 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (2020.9.23)
2020-10-18 23:33:18,834 > Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (0.8.0)
2020-10-18 23:33:18,834 > Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (1.4)
2020-10-18 23:33:18,834 > Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (2.11.2)
2020-10-18 23:33:18,835 > Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (2.0)
2020-10-18 23:33:18,835 > Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (0.19.0)
2020-10-18 23:33:18,835 > Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (3.1.1)
2020-10-18 23:33:18,835 > Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (1.1.1)
2020-10-18 23:33:18,835 > Requirement already satisfied: websocket-client<1,>=0.57 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (0.57.0)
2020-10-18 23:33:18,836 > Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (0.1.5.post0)
2020-10-18 23:33:18,836 > Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (1.2.0)
2020-10-18 23:33:18,836 > Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (1.0.7)
2020-10-18 23:33:18,836 > Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (1.1.2)
2020-10-18 23:33:18,837 > Requirement already satisfied: feedparser<6,>=5.2.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (5.2.1)
2020-10-18 23:33:18,837 > Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.15) (1.1.0)
2020-10-18 23:33:18,837 > Requirement already satisfied: pathtools>=0.1.1 in /home/pi/oprint/lib/python3.7/site-packages (from watchdog<1,>=0.10.2->OctoPrint->Bed-Visualizer==0.1.15) (0.1.2)
2020-10-18 23:33:18,837 > Requirement already satisfied: pyasn1>=0.1.3 in /home/pi/oprint/lib/python3.7/site-packages (from rsa==4.0->OctoPrint->Bed-Visualizer==0.1.15) (0.4.8)
[... cut ...]
2020-10-18 23:33:22,445 > Attempting uninstall: Bed-Visualizer
2020-10-18 23:33:22,445 > Found existing installation: Bed-Visualizer 0.1.15
2020-10-18 23:33:22,445 > Uninstalling Bed-Visualizer-0.1.15:
2020-10-18 23:33:22,445 > Successfully uninstalled Bed-Visualizer-0.1.15
2020-10-18 23:33:22,445 > Successfully installed Bed-Visualizer-0.1.15
```

**Screenshots**
[image]

**Firmware and Version**
OctoPrint 1.4.2 Python 3.7.3 OctoPi 0.17.0

I can provide additional information, if required.

Thanks

jneilliii, 2020-10-19:
Did you run the command listed in the known issues of the readme [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer#known-issues)?

lmcbmai, 2020-10-19:
The following command from provided link fixed the issue:
`sudo apt install libatlas3-base`

Many thanks!!

closed 2020-10-19 by jneilliii