jneilliii/OctoPrint-BedLevelVisualizer #315: [FR]: Add nozzle offset probe and set

LMS0815, 2020-10-28:
To have all ABL information in one place please add a probe of the XYZ offset to the GUI screen.

`M851; XYZ Probe Offset`

Additionally it would be nice being able to adjust the Z-Offset (`M851 Z-1.8`).

Please be careful using `M851 X-12.5 Y-38.0 Z-1.8` with more than one axis as there seems to be a bug (in some Marli FW) that this is not working. Use `M851 X-12.5`, `M851 Y-38.0` and  `M851 Z-1.8` instead.

Alternatively add the ability to you custom buttons providing/showing values retrieved via regex.

jneilliii, 2020-10-28:
Not sure how much effort would be involved in getting the regex functionality similar to custom controls to work within the plugin for displaying returned data, but there are custom commands in the settings that you can use to add the buttons to send the commands, with inputs.

LMS0815, 2020-10-29:
Actually I am already using _custom controls_ but I would like to have all ABL controls  one one page.

jneilliii, 2020-10-29:
Bed level visualizer also has custom controls in its settings...

LMS0815, 2021-12-08:
Seems to be abandon.

jneilliii, 2021-12-08:
Not abandoned, just haven't had the time for plugin development.