jneilliii/OctoPrint-BedLevelVisualizer #319: [BUG]: BLV Broken in OctoPrint 1.5.0

jneilliii, 2020-11-10:
**Describe the bug**
when using OctoPrint 1.5.0rc1 the tables don't load.

**Expected behavior**
mesh data and angle corrections tables to work.

**Debug Log**

**Screenshots**
[image]

**Firmware and Version**
N/A

jneilliii, 2020-11-10:
@LMS0815 I was wondering if you might know why the latest knockout upgrade included in OctoPrint 1.5.0 broke your correction matrix and mesh data tables? Seems there is an error related to `$parentContext.$index is not a function`.

Pavulon87, 2020-11-11:
Octolapse is affected too:
`packed_core.js?4e2bc246:1021 Could not bind view model  to target #octolapse_tab : ReferenceError: Unable to process binding "template: function(){return 'snapshot-plan-status-template' }"`
Pretty gcode:
`Could not bind view model PrettyGCodeViewModel to target #tab_plugin_prettygcode : Error: You cannot apply bindings multiple times to the same element.`

BLV:
`Could not bind view model bedlevelvisualizerViewModel to target #settings_plugin_bedlevelvisualizer : TypeError: Unable to process binding "template: function(){return {name:'bedlevelvisualize_meshtable',data:{mesh:$root.mesh_data(),len:3}} }"`

`packed_core.js?4e2bc246:1021 Could not bind view model bedlevelvisualizerViewModel to target #tab_plugin_bedlevelvisualizer : TypeError: Unable to process binding "if: function(){return $root.showdegree() && $root.mesh_data().length }"`

jneilliii, 2020-11-11:
pretty gcode actually has a PR against it from foosel that resolves its issue. The error with BLV is a little more involved and will probably take a while for me to fix, unless @LMS0815 is able to step in and advise, as all the effected parts were contributed to this plugin by them.

LMS0815, 2020-11-11:
I am currently bound to an other project with limited time. I'll try installing a RC test environment (earliest next week). 
This is probably related to the ViewModel container parent/child references...

jneilliii, 2020-11-12:
I think I was able to figure it out with a little tweak to the childContext option of the inner most foreach binding. If anyone makes it here looking for a fix, you can copy/paste the URL below in plugin manager > get more > ...from URL.

```
https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/1.0.0rc4.zip
```

Pavulon87, 2020-11-12:
> I think I was able to figure it out with a little tweak to the childContext option of the inner most foreach binding. If anyone makes it here looking for a fix, you can copy/paste the URL below in [...]
For me it is fixed.

jneilliii, 2020-11-12:
Awesome, thanks for confirming @Pavulon87.

closed 2020-11-21 by jneilliii