jneilliii/OctoPrint-BedLevelVisualizer #320: [BUG]: Illegal Instruction

racerxdl, 2020-11-15:
**Describe the bug**
After installing the BedLevelVisualizer, the Octoprint doesnt start anymore giving a Illegal instruction message.

**Expected behavior**
Octoprint works

**Debug Log**
Can't get log since octoprint does not start after installing the plugin.

**Screenshots**

[image]
[image]

**Firmware and Version**
Octoprint installed on ubuntu using https://octoprint.org/download/ manual install tutorial.

I did a fresh install of the Octoprint, and it gives me the same error. Its an AArch64 platform (ODROID C2)

jneilliii, 2020-11-15:
My assumption here it's related to numpy. You can uninstall by running the following command, replacing the path to your venv folder's pip command. The path listed below is what you would see on an octopi installation.

```
~/oprint/bin/pip uninstall Bed-Visualizer
```

miklschmidt, 2020-11-21:
Same problem here on octopi0.17.0, octoprint version 1.4.2, python 2.7. Failed to install, then server shutdown. Restarting it works fine, but i can't install the plugin.

jneilliii, 2020-11-22:
This is probably due to the numpy dependency not compiling. Try to run the following commands on your pi via SSH while not printing.

```
~/oprint/bin/pip install numpy
sudo service octoprint restart
```

then try again and report back.

jneilliii, 2020-11-22:
@miklschmidt are you also on AArch64 platform (ODROID C2)?

miklschmidt, 2020-11-22:
@jneilliii no sorry, i'm on a regular Raspberry Pi 4 Model B 4gb.

ecob81, 2020-11-25:
Same problem on Pi 3 B+. Fails to install Numpy

tried

~/oprint/bin/pip install numpy

and stopped at

Running setup.py bdist_wheel for numpy ...

jneilliii, 2020-11-25:
did it stop, or you just didn't wait long enough? The compile time on Pi 3 B+ would be around 20-40 minutes if I remember correctly. Another option is to upgrade your octoprint environment to python 3 using the script linked below, running the command `sudo apt install libatlas3` after the upgrade is complete and then install the plugin.

https://github.com/cp2004/Octoprint-Upgrade-To-Py3

ecob81, 2020-11-25:
Thankyou, I just needed to wait longer, it took around 15mins to install numpy.

I used "~/oprint/bin/pip install numpy" and then installed plugin but guessing it would have all installed ok from octoprint ui if I waited longer.

closed 2020-12-19 by jneilliii

matueranet, 2021-06-07:
For anyone else landing here: I use an Odroid C2, and the wheel that pip downloaded turned out not to be compatible with the architecture of the device. I forced a local compilation of numpy using: `pip install --no-binary :all: numpy==1.20.3`