jneilliii/OctoPrint-BedLevelVisualizer #322: [BUG]: Adjustment Angle Color mismatch when Reverse turn direction is checked.

RCNet46, 2020-11-15:
**Describe the bug**
If the "Reverse turn direction" option is checked, it swaps the position of the Turn direction legend, but does not change the color key of the turn directions.  Now the color of the turn direction guidance in the table no longer matches the legend at the bottom.  Table suggest turn right in RED and legend shows "clockwise (right)" in BLUE. Example in attached screenshot.

**Expected behavior**
Colors between table and legend should match for the given direction.

**Debug Log**
N/A

**Screenshots**

[image]

**Firmware and Version**
Bed Visualizer (0.1.15)

jneilliii, 2020-11-15:
possible duplicate to #278, resolved in the current RC release. If you'd like to try that version to see you can copy/paste the URL below into plugin manager > get more > ...from URL and click install.

```
https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/1.0.0rc4.zip
```

RCNet46, 2020-11-16:
It is similar, but different than #278, and is still present in RC4.  Here is a little better picture illustrating the issue from the release candidate. "Turn right" is indicated in RED in one location and "clockwise (right)" is BLUE in the other.  The colors are only mismatched when the "Reverse turn direction" box is checked.  

[image]

Checking the "Reverse turn direction" box appears to change the order of the clockwise/anticlockwise labels instead of changing the text color associated with them.

unchecked:

[image]

versus checked:

[image]

jneilliii, 2020-11-16:
Thanks, that helps. I'll look into fixing that as part of the next release.

jneilliii, 2020-11-20:
I think I got this squared away now with the above commit, if you'd be so kind as to test the latest rc5 that would be great. Copy/paste the URL below into plugin manager > get more > ...from URL and click install and it should upgrade without issue.

```
https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/1.0.0rc5.zip
```

RCNet46, 2020-11-21:
Yep, rc5 fixed the issue.  Thank you for your efforts!!

closed 2020-11-21 by RCNet46