jneilliii/OctoPrint-BedLevelVisualizer #324: [FR]: Add an option to force homing before doing a mesh update.

sushibagel, 2020-11-21:
**Is your feature request related to a problem? Please describe.**
Multiple times I've tried to update my mesh manually without remembering to home all axis resulting in a crash, this has resulted in dents in my build surface and broken air ducts. 

**Describe the solution you'd like**
Having an option that forgetful people like me can set to auto home first will help solve this problem in many cases.

jneilliii, 2020-11-22:
Add the homing command to your gcode scripts, as documented [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/master/wiki/gcode-examples.md). You'll notice every example has a homing command.

closed 2020-11-22 by jneilliii

sushibagel, 2020-11-22:
Thanks didn't realize there was a script that could be modified in settings