jneilliii/OctoPrint-BedLevelVisualizer #325: [BUG]: BLV Not Loading in 1.5.0rc3

DodgeDeBoulet, 2020-11-25:
**Describe the bug**

Bed Level Visualizer is not loading and displaying a tab in OctoPrint 1.5.0rc3

**Expected behavior**

Bed Level Visualizer loading and displaying a tab
**Debug Log**

Cannot set BLV debug logging. It does not display in the installed plugins list and is not selectable from the debug settings.
This appears in octoprint.log:

```
2020-11-25 10:13:02,768 - octoprint.plugin.core - ERROR - Error loading plugin bedlevelvisualizer
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/numpy/core/__init__.py", line 22, in <module>
    from . import multiarray
  File "/home/pi/oprint/lib/python3.7/site-packages/numpy/core/multiarray.py", line 12, in <module>
    from . import overrides
  File "/home/pi/oprint/lib/python3.7/site-packages/numpy/core/overrides.py", line 7, in <module>
    from numpy.core._multiarray_umath import (
ImportError: libf77blas.so.3: cannot open shared object file: No such file or directory
```

**Screenshots**

Not available, since it's not loading.
**Firmware and Version**

Marlin 2.0.7.2

DodgeDeBoulet, 2020-11-25:
Searching for the ImportError turned up the following (apparent) fix:

```
sudo apt install libatlas-base-dev
```

Not sure why this was necessary.

DodgeDeBoulet, 2020-11-25:
Found the same problem in closed issues. Closing.

closed 2020-11-25 by DodgeDeBoulet

jneilliii, 2020-11-25:
It's not that it's 1.5.0 causing this but what appears to be you switching to Python 3, and the command you ran is documented on the known issues section of the readme. And intersetigly enough, is also flagged on the bug report template when you started it...

[image]

DodgeDeBoulet, 2020-11-25:
Yet I somehow managed to miss all of those notifications ... but I did
close the bug within 15 minutes of opening it. Was hoping you wouldn't
notice 😁