jneilliii/OctoPrint-BedLevelVisualizer #329: [FR]:  Mesh Save And Restore

gbkwiatt, 2020-12-06:
Since there is a Marlin command to output  current loaded Mesh into GCode backup ( Gcode to enter to bring mesh back). 
Would be nice if Bed Visualiser could have that option, to save it to file and bring backup when needed. 
All it is, is just parsing output from terminal.

I am currently using UBL, after I have Mesh loaded you have to execute "G29 S-1", 
and you get 
```gcode
Send: G29 S-1
Recv: echo:  G29 I999
Recv: echo:  M421 I0 J0 Z0.4185
Recv: echo:  M421 I0 J1 Z0.3610
Recv: echo:  M421 I0 J2 Z0.3045
Recv: echo:  M421 I0 J3 Z0.2855
Recv: echo:  M421 I0 J4 Z0.2270
Recv: echo:  M421 I0 J5 Z0.1950
Recv: echo:  M421 I0 J6 Z0.1460
Recv: echo:  M421 I0 J7 Z0.1010
Recv: echo:  M421 I0 J8 Z0.0770
Recv: echo:  M421 I0 J9 Z0.0245
Recv: echo:  M421 I1 J0 Z0.4125
Recv: echo:  M421 I1 J1 Z0.3975
Recv: echo:  M421 I1 J2 Z0.3420
Recv: echo:  M421 I1 J3 Z0.2610
Recv: echo:  M421 I1 J4 Z0.1765
Recv: echo:  M421 I1 J5 Z0.1400
Recv: echo:  M421 I1 J6 Z0.0920
Recv: echo:  M421 I1 J7 Z0.0850
Recv: echo:  M421 I1 J8 Z0.0765
Recv: echo:  M421 I1 J9 Z0.0265
Recv: echo:  M421 I2 J0 Z0.3620
Recv: echo:  M421 I2 J1 Z0.3105
Recv: echo:  M421 I2 J2 Z0.2550
Recv: echo:  M421 I2 J3 Z0.2100
Recv: echo:  M421 I2 J4 Z0.1320
Recv: echo:  M421 I2 J5 Z0.0960
Recv: echo:  M421 I2 J6 Z0.0455
Recv: echo:  M421 I2 J7 Z0.0370
Recv: echo:  M421 I2 J8 Z0.0350
Recv: echo:  M421 I2 J9 Z0.0025
Recv: echo:  M421 I3 J0 Z0.3475
Recv: echo:  M421 I3 J1 Z0.3180
Recv: echo:  M421 I3 J2 Z0.2165
Recv: echo:  M421 I3 J3 Z0.1895
Recv: echo:  M421 I3 J4 Z0.0900
Recv: echo:  M421 I3 J5 Z0.0695
Recv: echo:  M421 I3 J6 Z0.0335
Recv: echo:  M421 I3 J7 Z0.0620
Recv: echo:  M421 I3 J8 Z0.0160
Recv: echo:  M421 I3 J9 Z0.0030
Recv: echo:  M421 I4 J0 Z0.3035
Recv: echo:  M421 I4 J1 Z0.2665
Recv: echo:  M421 I4 J2 Z0.1915
Recv: echo:  M421 I4 J3 Z0.1550
Recv: echo:  M421 I4 J4 Z0.0450
Recv: echo:  M421 I4 J5 Z0.0090
Recv: echo:  M421 I4 J6 Z-0.0315
Recv: echo:  M421 I4 J7 Z-0.0360
Recv: echo:  M421 I4 J8 Z-0.0195
Recv: echo:  M421 I4 J9 Z-0.0365
Recv: echo:  M421 I5 J0 Z0.2565
Recv: echo:  M421 I5 J1 Z0.1935
Recv: echo:  M421 I5 J2 Z0.1250
Recv: echo:  M421 I5 J3 Z0.0830
Recv: echo:  M421 I5 J4 Z0.0125
Recv: echo:  M421 I5 J5 Z-0.0130
Recv: echo:  M421 I5 J6 Z-0.0265
Recv: echo:  M421 I5 J7 Z-0.0325
Recv: echo:  M421 I5 J8 Z-0.0225
Recv: echo:  M421 I5 J9 Z-0.0315
Recv: echo:  M421 I6 J0 Z0.2235
Recv: echo:  M421 I6 J1 Z0.1920
Recv: echo:  M421 I6 J2 Z0.1500
Recv: echo:  M421 I6 J3 Z0.0930
Recv: echo:  M421 I6 J4 Z0.0125
Recv: echo:  M421 I6 J5 Z0.0080
Recv: echo:  M421 I6 J6 Z-0.0470
Recv: echo:  M421 I6 J7 Z-0.0700
Recv: echo:  M421 I6 J8 Z-0.0510
Recv: echo:  M421 I6 J9 Z-0.0470
Recv: echo:  M421 I7 J0 Z0.1815
Recv: echo:  M421 I7 J1 Z0.1505
Recv: echo:  M421 I7 J2 Z0.0825
Recv: echo:  M421 I7 J3 Z0.0230
Recv: echo:  M421 I7 J4 Z-0.0205
Recv: echo:  M421 I7 J5 Z-0.0535
Recv: echo:  M421 I7 J6 Z-0.0550
Recv: echo:  M421 I7 J7 Z-0.0975
Recv: echo:  M421 I7 J8 Z-0.0860
Recv: echo:  M421 I7 J9 Z-0.0605
Recv: echo:  M421 I8 J0 Z0.1675
Recv: echo:  M421 I8 J1 Z0.1610
Recv: echo:  M421 I8 J2 Z0.0880
Recv: echo:  M421 I8 J3 Z0.0270
Recv: echo:  M421 I8 J4 Z-0.0130
Recv: echo:  M421 I8 J5 Z-0.0490
Recv: echo:  M421 I8 J6 Z-0.0710
Recv: echo:  M421 I8 J7 Z-0.0925
Recv: echo:  M421 I8 J8 Z-0.0480
Recv: echo:  M421 I8 J9 Z-0.0535
Recv: echo:  M421 I9 J0 Z0.1675
Recv: echo:  M421 I9 J1 Z0.1715
Recv: echo:  M421 I9 J2 Z0.0935
Recv: echo:  M421 I9 J3 Z0.0310
Recv: echo:  M421 I9 J4 Z-0.0055
Recv: echo:  M421 I9 J5 Z-0.0445
Recv: echo:  M421 I9 J6 Z-0.0710
Recv: echo:  M421 I9 J7 Z-0.0875
Recv: echo:  M421 I9 J8 Z-0.0100
Recv: echo:  M421 I9 J9 Z-0.0465
Recv: ok
```

So that could be save then into gcode file, and executed on restore

jneilliii, 2020-12-06:
I actually have a companion plugin [here](https://github.com/jneilliii/OctoPrint-BLVLogger) that will log each mesh run and then allow you to switch the graph between each run? 

Are you using the recommended settings for the UBL or not as documented [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/master/wiki/gcode-examples.md#unified-bed-leveling-ubl)? I programmed this with UBL and personally don't do the probing procedure in the plugin's settings and just use the `M420 V` to pull the mesh data from EEPROM.

gbkwiatt, 2020-12-06:
I did try both approaches. I was just thinking, it would be nice to have something to back it up. I know mesh stays in eeprom even after initialize eeprom, but sometimes I want some cleanup or changing the board.
Seems like nice addon

jneilliii, 2020-12-06:
So if you're controlling it through gcode you could use the `M118 E1 BEDLEVELVISUALIZER` just before the load command, just not familiar with `S-1` you are using and have only personally used `M420 L#` to switch between.

gbkwiatt, 2020-12-06:
I am not having any issues with bed visualiser.  
Marlin has option `G29 S-1` to print out to terminal, Gcode on brining back current loaded mesh. 
That could be used to save backup physically in octoprint somewhere and then you could bring that back by running that gcode on new board etc. 
I am not talking about switching meshes, but that could be also good idea, to have some slot switcher.

jneilliii, 2020-12-06:
can enlighten me on the gcode that could be used to load the mesh in from an export? I don't see this being a very difficult thing to add to the plugin, but it might be a while before I get to it.

gbkwiatt, 2020-12-06:
It's on my first post. So on `G29 S-1`
You get: 
```gcode
Send: G29 S-1
Recv: echo:  G29 I999
Recv: echo:  M421 I0 J0 Z0.4185
Recv: echo:  M421 I0 J1 Z0.3610
Recv: echo:  M421 I0 J2 Z0.3045
Recv: echo:  M421 I0 J3 Z0.2855
..
```
Depending on how many points you have in your mesh. I had 100 so there will be 100 points. 
it takes it from currently loaded Mesh. 
Then if you run it back so:
```gcode
G29 I999
M421 I0 J0 Z0.4185
M421 I0 J1 Z0.3610
M421 I0 J2 Z0.3045
M421 I0 J3 Z0.2855
...
```

It brings back your points into the loaded Mesh, then you could save loaded mesh to new slot or whatever you like.
The thing is, you can save that gcode physically in octoprint as a backup

jneilliii, 2020-12-06:
Oh, cool, so it basically reports to you the gcode that you would use to "import" those values. If that's the case I could fairly easily do that, by collecting the responses, saving them to a file in the plugin's data folder and then provide a UI to be able to manage them. Thanks for the info .

gbkwiatt, 2020-12-06:
yup exactly. I thought it could be easy to do. I just copied and pasted it and saved and had a thought, that would be cool to have in bedvis. 
Great looking forward to id ;) Awesome job btw.