jneilliii/OctoPrint-BedLevelVisualizer #331: [BUG]:

Deadbot1, 2020-12-06:
Mesh data gets further and further off with each probing.
Mesh data should be reset between probing. 

**Debug Log**

**Screenshots**

**Firmware and Version**
Marlin 2+bugfix

I had an issue where after doing a paper level and then running the probe to get a visualized mesh it would be wildly off. Like showing 2 corners off by 2+mm. I cleared all the mesh data in firmware then disabled mesh leveling in settings before running a new probe and it reported correctly. I have now added an M420 S0 right before the actual probing occurs in the settings page for Bed Visualizer in Octoprint. This makes sure that the mesh bed leveling is not occurring while probing. Seems to have corrected the issue.

jneilliii, 2020-12-06:
which type of leveling are you using? I'd like to update the examples [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/master/wiki/gcode-examples.md).

Deadbot1, 2020-12-07:
I have an Ender 5 Plus with the BL touch. I am running on a custom build of
Marlin from Kersey Fabrications with an SKR 1.4 Turbo board from BTT.

jneilliii, 2020-12-07:
Kersey Fabrications is a new one to me. Do you have a link to their source?

Deadbot1, 2020-12-07:
https://github.com/KerseyFabrications Is his git hub. I actually just
downloaded his precompiled from a video of his for setting up dual drivers
on the z axis.

jneilliii, 2020-12-07:
Yeah, so looking at this branch [here](https://github.com/KerseyFabrications/Marlin/tree/2.0.6-BTT-SKR-V1.4-Turbo-Ender5Plus-DualZ) based on what you've provided it appears it's using `AUTO_BED_LEVELING_BILINEAR`, but the branch is 4 months old compared to some of his other branches.

https://github.com/KerseyFabrications/Marlin/branches/all?page=1

Deadbot1, 2020-12-14:
Yeah,it's a bit dated, but it works. I'm doing more testing to try and nail
this thing down. So far what I see is that if I make sure to home z before
running a probe I get clean data. If I run multiple probes in a row then it
starts stacking results....

I ran a series of test runs and watched the output in terminal. I had the
Nozzle at 210C and the Bed at 60C. Nozzle should not affect data that I can
see. I homed out and then ran a couple of runs by just updating Bed
Visualizer through Octoprint.  Terminal output follows....

Send: G91Recv: okSend: G28 Z0[...]Recv: X:227.00 Y:195.00 Z:11.88
E:0.00 Count X:18160 Y:15600 Z:9504Recv: okSend: G90[...]Recv: okSend:
M155 S100Recv: okSend: M420 S0Recv: echo:Bed Leveling OFFRecv:
echo:Fade Height OFFRecv: okSend: G29 T[...]Recv: Bilinear Leveling
Grid:Recv:       0      1      2      3      4Recv:  0 -0.005 +0.029
+0.034 +0.056 +0.092Recv:  1 +0.043 +0.059 +0.044 +0.070 +0.136Recv:
2 +0.035 +0.067 +0.058 +0.083 +0.149Recv:  3 +0.005 +0.026 +0.045
+0.056 +0.173Recv:  4 +0.001 +0.015 -0.007 +0.055 +0.124Recv: Recv:
X:350.00 Y:320.00 Z:11.58 E:0.00 Count X:28000 Y:25600 Z:9504Recv:
okSend: M155 S3Recv: ok[...]Send: M155 S100Recv: okSend: M420 S0Recv:
echo:Bed Leveling OFFRecv: echo:Fade Height OFFRecv: X:350.00 Y:320.00
Z:11.73 E:0.00 Count X:28000 Y:25600 Z:9386Recv: okSend: G29
T[...]Recv: Bilinear Leveling Grid:Recv:       0      1      2      3
    4Recv:  0 -0.145 -0.116 -0.107 -0.082 -0.044Recv:  1 -0.082 -0.101
-0.102 -0.073 -0.008Recv:  2 -0.091 -0.094 -0.107 -0.068 +0.014Recv:
3 -0.118 -0.095 -0.114 -0.096 +0.033Recv:  4 -0.139 -0.145 -0.163
-0.084 -0.038Recv: Recv: X:350.00 Y:320.00 Z:11.73 E:0.00 Count
X:28000 Y:25600 Z:9504Recv: okSend: M155 S3Recv: ok[...]Send: M155
S100Recv: okSend: M420 S0Recv: echo:Bed Leveling OFFRecv: echo:Fade
Height OFFRecv: X:350.00 Y:320.00 Z:11.68 E:0.00 Count X:28000 Y:25600
Z:9345Recv: okSend: G29 T[...]Recv: Bilinear Leveling Grid:Recv:
0      1      2      3      4Recv:  0 -0.348 -0.319 -0.295 -0.278
-0.238Recv:  1 -0.286 -0.280 -0.292 -0.274 -0.207Recv:  2 -0.305
-0.298 -0.293 -0.253 -0.189Recv:  3 -0.334 -0.320 -0.298 -0.272
-0.162Recv:  4 -0.342 -0.333 -0.358 -0.286 -0.226Recv: Recv: X:350.00
Y:320.00 Z:11.92 E:0.00 Count X:28000 Y:25600 Z:9504Recv: okSend: M155
S3Recv: ok[...]Send: M155 S100Recv: okSend: M420 S0Recv: echo:Bed
Leveling OFFRecv: echo:Fade Height OFFRecv: X:350.00 Y:320.00 Z:11.71
E:0.00 Count X:28000 Y:25600 Z:9372Recv: okSend: G29 T[...]Recv:
Bilinear Leveling Grid:Recv:       0      1      2      3      4Recv:
0 -0.505 -0.483 -0.453 -0.430 -0.398Recv:  1 -0.458 -0.447 -0.451
-0.432 -0.366Recv:  2 -0.446 -0.445 -0.454 -0.420 -0.345Recv:  3
-0.483 -0.457 -0.472 -0.454 -0.329Recv:  4 -0.509 -0.504 -0.520 -0.446
-0.383Recv: Recv: X:350.00 Y:320.00 Z:12.09 E:0.00 Count X:28000
Y:25600 Z:9504Recv: okSend: M155 S3Recv: ok[...]Send: M155 S100Recv:
okSend: M420 S0Recv: echo:Bed Leveling OFFRecv: echo:Fade Height
OFFRecv: X:350.00 Y:320.00 Z:11.72 E:0.00 Count X:28000 Y:25600
Z:9380Recv: okSend: G29 T[...]Recv: Bilinear Leveling Grid:Recv:
0      1      2      3      4Recv:  0 -0.662 -0.625 -0.604 -0.585
-0.554Recv:  1 -0.591 -0.612 -0.607 -0.575 -0.510Recv:  2 -0.606
-0.604 -0.613 -0.571 -0.493Recv:  3 -0.639 -0.644 -0.633 -0.605
-0.479Recv:  4 -0.666 -0.662 -0.668 -0.592 -0.545Recv: Recv: X:350.00
Y:320.00 Z:12.25 E:0.00 Count X:28000 Y:25600 Z:9504Recv: okSend: M155
S3Recv: ok

Then I homed and reprobed...

Send: G91Recv: okSend: G28 Z0[...]Recv: X:227.00 Y:195.00 Z:11.88
E:0.00 Count X:18160 Y:15600 Z:9504Recv: okSend: G90Recv: ok[...]Send:
M155 S100Recv: okSend: M420 S0Recv: echo:Bed Leveling OFFRecv:
echo:Fade Height OFFRecv: okSend: G29 T[...]Recv: Bilinear Leveling
Grid:Recv:       0      1      2      3      4Recv:  0 -0.031 -0.004
+0.015 +0.022 +0.044Recv:  1 +0.026 +0.035 +0.008 +0.018 +0.068Recv:
2 +0.033 +0.043 +0.017 +0.026 +0.086Recv:  3 -0.012 +0.001 -0.017
+0.005 +0.115Recv:  4 -0.023 -0.034 -0.063 +0.000 +0.049Recv: Recv:
X:350.00 Y:320.00 Z:10.58 E:0.00 Count X:28000 Y:25600 Z:9504Recv:
okSend: M155 S3Recv: ok[...]

It seems that the mesh isn't getting clean data without a fresh homing of
Z. I am altering my Bed Visualizer code as follows...

M155 S100 ; set temperature reporting delay, use a value longer than the
time it takes for your leveling command to complete.
G28; home all axis
@BEDLEVELVISUALIZER ; instruct plugin to start recording responses from
printer.
G29 T ; report the bed leveling mesh points.
M155 S3 ; set the temperature reporting delay back to a shorter time span.

closed 2021-02-08 by github-actions[bot]