jneilliii/OctoPrint-BedLevelVisualizer #332: [FR]: Adjust Z-range colormap

dryphi, 2020-12-08:
I recently started using the bed visualizer and it's a neat tool.
However the default Z range (-2 to 2) is too large IMO to view small aberrations.
Using plotly I figured out how to adjust the Z-range (Axes - Range - Z) but the colormap on the right does not adjust along with the new Z range. I can't figure out how to adjust the colormap.

Can we enable some feature within the plugin itself to set a custom Z-range? In this way the colormap would be adjusted as well.

Thanks

jneilliii, 2020-12-08:
You mean this one that already exists?

[image]

dryphi, 2020-12-09:
> You mean this one that already exists?
> 
> [image]

where

jneilliii, 2020-12-09:
Yeah, looks like you found it.

closed 2020-12-09 by jneilliii

dryphi, 2020-12-09:
Would be nice if there was a way to refresh the graph without reloading a page. For instance if you wanted to view the graph while a print was running, or just wanted to view the current stored values without running a full ABL.

jneilliii, 2020-12-11:
I think you can actually switch tabs and it will re-draw when you switch back, no need to do a full refresh.

dryphi, 2020-12-16:
> I think you can actually switch tabs and it will re-draw when you switch back, no need to do a full refresh.

Doesn't seem to for me although I'll double check. The other graphs do refresh though.

Would it be useful to add a "redraw" button? Just an idea. For me, it would be, as I could see the currently stored values.

jneilliii, 2020-12-16:
Are you using stored mesh in eeprom?

dryphi, 2020-12-16:
> Are you using stored mesh in eeprom?

Yes I believe so. I'm using the community firmware on a CR6

jneilliii, 2020-12-16:
It may make sense to add a refresh button, but the graph will redraw when switching tabs within OctoPrint, the data may not change from any changes that you've performed from the printer directly until you run the next level process though (assuming you're using the setting in the plugin to Save Mesh). If your firmware does support storing the mesh in EEPROM, then you could potentially use the `M420 V` command in the plugin's settings to graph the mesh that gets created whenever and wherever the mesh is created. It just depends if your firmware returns the mesh data with that command or not. Then it just loads the mesh from EEPROM reporting and doesn't do a full probing process.

dryphi, 2020-12-16:
> It may make sense to add a refresh button, but the graph will redraw when switching tabs within OctoPrint, the data may not change from any changes that you've performed from the printer directly [...]

Yes I understand. It does not do a full probing process. That's the whole point. I just want to refresh the graph along with any changes I've made within OctoPrint (i.e. changing the Z-limits).
Currently there's no way to refresh the graph without reprobing which is what I want to avoid.

dryphi, 2020-12-16:
>  but the graph will redraw when switching tabs within OctoPrint, 

It doesn't for me. And it may not if you just change something like the Z-limits.

What if you already have a mesh stored in EEPROM that you made by using the controls on the firmware itself. Then you install OctoPrint and this plugin. How do you view the graph in that case? Currently there's no way. If there's some G-code that will grab it, great, but then how do you make the plugin use that to generate a graph?

jneilliii, 2020-12-16:
That's what I'm trying to explain. If your firmware supports it the command `M420 V` should give you the mesh report from EEPROM and the plugin can use it. So in the plugin's settings for the mesh put in `M420 V` and it will graph whatever was the last stored mesh in EEPROM. A custom button can be added in the plugin's settings as well to run the probing process outside of the update mesh button. You can also update your start gcode and add the line `M118 E1 BEDLEVELVISUALIZER` just prior to the leveling command and that should also allow automatic updating of the graph on the plugin side, assuming once again your firmware supports the M118 command. Not all firmware is equal or enabled all the same features as others so I cannot tell you what you have the capabilities of using within the plugin.

dryphi, 2020-12-16:
The mesh still does not refresh when I change the Z-limits unless I reload the page.

dryphi, 2020-12-16:
> A custom button can be added in the plugin's settings as well to run the probing process outside of the update mesh button. 

How do you do this?

dryphi, 2020-12-16:
> You can also update your start gcode and add the line `M118 E1 BEDLEVELVISUALIZER` just prior to the leveling command and that should also allow automatic updating of the graph on the plugin side, [...]

Are you sure M118 is the correct G-code? That's just serial print. https://marlinfw.org/docs/gcode/M118.html

jneilliii, 2020-12-16:
> I added a button labeled "Grab Mesh from EEPROM" and the only command is M420 V. Is that correct?

You should probably add the `@BEDLEVELVIUALIZER` command just prior to the M420 V, that way the plugin is instructed to collect mesh data. I was suggesting that you use that command for the update mesh button, not necessarily having a custom button specifically for it. I've tried to leave the plugin as open as possible to work with as many firmware variants as possible.

> Are you sure 'M118 E1 BEDLEVELVISUALIZER' is the correct G-code? It just echos "BEDVISUALIZER".

That is correct, but just after that command you should have a G29 T or whatever command you use to probe/report mesh data. The M118 command is just telling the plugin again to start collecting the mesh data that it uses to graph. You should notice on the plugin's tab that when that message is received by your printer that the "processing" happens until an ok is received after the probing process.

dryphi, 2020-12-16:
> I was suggesting that you use that command for the update mesh button, not necessarily having a custom button specifically for it.

But that would defeat the purpose. I'm trying to avoid doing the probing routine if there's already a mesh stored in EEPROM that I'm happy with. But I'd still like to see the graph for that mesh.

> That is correct, but just after that command you should have a G29 T or whatever command you use to probe/report mesh data. The M118 command is just telling the plugin again to start collecting the [...]

Is there an @ missing in that G-code then?

jneilliii, 2020-12-16:
> I'm trying to avoid doing the probing routine if there's already a mesh stored in EEPROM that I'm happy with. But I'd still like to see the graph for that mesh.

open the plugin's settings and put this in...

[image]

add a custom button that runs these commands and call it "probe my bed" or something else that makes sense to you.

```
@BEDLEVELVISUALIZER
G29 T
```

Press the update mesh button to load stored EEPROM mesh data.
Press the custom "probe my bed" button to actually run a probing process.

dryphi, 2020-12-17:
> open the plugin's settings and put this in...

That's exactly what I did already. It works fine. See post:
https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/332#issuecomment-745798040

Now I'm trying to determine what you meant with the M118 command. Check the code, I feel like something is missing there.

dryphi, 2020-12-17:
> open the plugin's settings and put this in...

You have the right idea but that's backwards from how I'm envisioning it.
I like the default "Update Mesh" button. I want to keep that as it is. I use that frequently. I adjusted the temps, but that's about the only changes I made to the default G-code in the "Update Mesh" button.

However I added a custom button that runs `M420 V` already. It's working fine. See post: https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/332#issuecomment-745798040

Now I'm just trying to determine what you meant about the `M118` command...

jneilliii, 2020-12-17:
I understand now. If you use a custom button for the `M420 V` command, if you want the mesh to update in the graph you also need to add `@BEDLEVELVISUALIZER` right before it. This basically tells the plugin, get ready you are about to get the mesh report from firmware.

The M118 command was just an aside relative to automatically updating the graph if your start gcode in your slicer runs probing every print. If you don't do that then you can just ignore that comment altogether. There are some users that run probing every print, and the M118 command allows for it to collect the mesh data to graph, even if printing from SD card.

dryphi, 2020-12-17:
> I understand now. If you use a custom button for the `M420 V` command, if you want the mesh to update in the graph you also need to add `@BEDLEVELVISUALIZER` right before it. This basically tells [...]

Perfect thanks!
Okay I see. Well if I have the "Grab Mesh from EEPROM" button then I won't need the M118 because I'll be doing that anyway after the fact.
Thanks for your patients 👍

dryphi, 2021-01-12:
The Move command doesn't seem to work. How do I move the button? I'd like it centered under the other one.

jneilliii, 2021-01-12:
Move command? Unfortunately, I don't think I have the ability to position the buttons in the plugin currently. Please open a new issue/feature request if you'd like that implemented. It will probably be a while before I get a chance to even look at it.