jneilliii/OctoPrint-BedLevelVisualizer #333: [FR]: Flip the angle corrections table upside down

mfurlend, 2020-12-09:
First I wanna say great job with this plugin. IMO it is the most useful plugin out there.
My issue is with the angle correction table. It appears  vertically flipped in respect to a top-town view of the mesh. This makes it kind of hard and annoying to determine which sides I actually need to correct by how much. A simple toggle to flip it upside down would be awesome.

Thanks!

jneilliii, 2020-12-10:
Could you please elaborate, possibly with screenshots, not sure I quite understand.

mfurlend, 2020-12-10:
[image]
In this picture you can see that the top right corner needs to be lowered and the bottom right corner needs to be raised, however in the table it is flipped.

jneilliii, 2020-12-11:
On the Current Mesh Data tab in settings check the box for descending y axis?

holocronology, 2020-12-11:
Most people are, I would guess, looking at their bed from the front with the 0,0 (front-left) corner closest to them. So, at the least, the table should have 0,0 in the lower left view on the screen, rather than at the upper left.

jneilliii, 2020-12-11:
I will make the descending y default to enabled.

closed 2021-01-31 by github-actions[bot]