jneilliii/OctoPrint-BedLevelVisualizer #336: [FR]: Angle Adjustment Info View on iPad

holocronology, 2020-12-11:
**Is your feature request related to a problem? Please describe.**
In some ways this feels like a bug, but it is more of an enhancement. My printer is in a remote location from any desktop computer. I found that using an iPad when I'm near the printer and needing to access the web UI of Octoprint is useful. However, the functionality of the angle adjustment table is broken in this situation. You cannot "mouse over" a cell to see the adjustment information. If you tap on the cell, it sets that cell to 0 and updates all the other cells.

**Describe the solution you'd like**
Ability to view the angle adjustment info for a cell without causing that cell to set to 0. One solution could be to split the cell into to parts, place the value in one half and the adjustment graphic in the other. Then, if you tap the value to set it to 0, everything will update. If you tap the graphic, the value doesn't change but the larger graphic view to the right updates to the currently selected cell with expanded information. This removes reliance on the mouse-over of the cell to update the info graphic.

**Describe alternatives you've considered**
Functionally, there isn't really an alternative, other than accessing the information on a computer, writing the values on a piece of paper, then going to the printer.

**Additional context**
None.

jneilliii, 2020-12-11:
Yeah, I didn't program that part of the plugin unfortunately, so there probably is some weird quirks here. Basically you want it to be more mobile friendly for touch devices, which may take me a while to figure out, but I've flagged it as an enhancement to take a look eventually.

holocronology, 2020-12-11:
Is that form some other plugin?

jneilliii, 2020-12-12:
No, the changes were merged in from another contributor. I'm not even sure how to begin to make it mobile friendly TBH. It will take some digging.

ayourk, 2020-12-17:
If there were some sort of tool controls like left arrow, right arrow (x axis), up arrow and down arrow (y/z axis) I think that would help.  Similar what is done within Octoprint itself as well as 1, 10, 100 stepping modes.