jneilliii/OctoPrint-BedLevelVisualizer #339: [FR]: Printer Profiles for Bed Visualization?

eblieb, 2020-12-19:
I have an Ender 3 V2 printer with a BLTouch and can not figure out how to orient the bed map. On my build plate X0 is all the way to the left and Y0 is all the way to the back. 

I can only get the maps to orient either the X axis correct or the Y axis correct, and flipping the axis doesnt seem to work either? Am I missing something?

jneilliii, 2020-12-19:
I could see where the confusion is here. The actual axes labels never change, just the mesh inside the graph. This means it's misleading relative to your reversed Y axis or am I misinterpreting? That could definitely be an issue when it comes to using the move nozzle to position button if that is the case. It would help me if you could enable debug logging in the plugin, restart octoprint, and try the update mesh process once and then share your plugin_bedlevelvisualizer_debug.log file in the logging section of OctoPrint's settings. You can just drag it into a comment here. As long as I've had this plugin available you're the first person that has mentioned this type of setup and I'm actually surprised now that I think about it.

eblieb, 2020-12-19:
[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/5718750/plugin_bedlevelvisualizer_debug.log)
I had already turned logging on for another thing so here you go.

jneilliii, 2020-12-19:
Thanks, so if I understand correctly from what you said the furthest back is Y = 0 and the closest edge of the bed toward you while facing is Y = 220. So if you enter the command `G1 X10 Y200` the nozzle moves to the left side and toward the front of the bed?

eblieb, 2020-12-19:
Nevermind, it looks like the front of the plate and farthest left is 0,0 so the orientation is correct. It is very confusing sometimes to think and visual in 3d where something is in a 2d plane

closed 2020-12-19 by jneilliii