jneilliii/OctoPrint-BedLevelVisualizer #341: [BUG]: G29 T?

eblieb, 2020-12-21:
In the GCode for Mesh Update Process there is 

G29 T			; report the bed leveling mesh points.

But in the MarlinFW docs https://marlinfw.org/docs/gcode/G029-abl-bilinear.html 
T is not an argument. So what is it suppose to do?

Also I have save mesh enabled but it doesn't seem to be saving it. I added a M500 at the end of my gcode to force it to update the eeprom.

jneilliii, 2020-12-21:
Adding M500 to the end of your script is good. The save option within the plugin is to save the mesh in the plugin's settings, thus avoids running the probing process every time you switch to the tab to render the mesh data. 

You can remove the T from that line. It's supposed to report the mesh data, but that might be specific to UBL or different versions of marlin.

eblieb, 2020-12-21:
So I am guessing the "update mesh button" wasn't actually suppose to trigger the bed leveling routine in Marlin? Whats funny is I have been using it purposely to run an ABL G29.

jneilliii, 2020-12-21:
That all depends on the gcode commands you configure in the plugin. It can do the probing process or it can just read mesh data stored in EEPROM, assuming the firmware supports it (ie M420 V returns mesh report). So for example, personally I use UBL and just put in M420 V in my gcode settings to graph/visualize the data and do the actual probing process outside of the update mesh button, which can be done with the custom buttons in the plugin's settings. It's really all up to how you want to configure it to work.

jneilliii, 2020-12-21:
For example, another option for you is to update your slicers start gcode (if you run G29 with every print) to include the `@BEDLEVELVISUALIZER` command just prior to the G29 and it will update your "saved mesh" in the plugin. This way you are always looking at the last run probed data. Hope that makes sense.

eblieb, 2020-12-21:
Ok that is much better. I didn't realize the plugin could be setup that way (I think by default you have it run G29 when updating mesh?). 

I added a button to run G28 R, G29 and M500 and now the update mesh just grabs the mesh from my eeprom so it is much faster.

jneilliii, 2020-12-21:
By default, after install, that setting is completely blank.

closed 2021-02-08 by github-actions[bot]