jneilliii/OctoPrint-BedLevelVisualizer #342: [FR]: Default Z axis scale

AcJoker2, 2020-12-27:
Hi,

as I can see, the Z axis is always scaled to -2 to + 2mm.
A bed will never have such deviations.
Therefore, in my opinion, it would make sense to adapt the default values to e.g. -0.3 to + 0.3.
That is still a lot, but it should fit most users and still clearly show the warps of the bed.

Otherwise a good addon and thanks for the work

Best regards
Markus

jneilliii, 2020-12-27:
Look at the settings, you can change the z heights of the grid, have fun.

closed 2020-12-27 by jneilliii

AcJoker2, 2020-12-27:
That's clear, I said that myself.
My suggestion was to add this nonsensical DEFAULT value to e.g. -0.3 to + 0.3mm.
Or you can make the Z axis dynamic so that it is always scaled to Min / Max.

jneilliii, 2020-12-27:
If the default starts that low then people get initial graphs with meshes that get clipped, which is even more non sensical for first time users.