jneilliii/OctoPrint-BedLevelVisualizer #343: [BUG]: X/Y Label

AcJoker2, 2020-12-27:
Hi,

is it normal that with a 10x10 mesh of a 235x235mm bed, the axes simply split the measuring points from 0 to 235mm, and not the real coordinates are used?

Like this
0 13 26 39 52 65 78 91 104 118 131 144 157 170 183 196 209 222 235

Ender 3 Pro with Klipper

Best regards
Markus

jneilliii, 2020-12-27:
The plugin interprets the probed positions based on number of points and the dimensions of the bed. There are only a few mesh reports that plugin interprets directly from mesh report data from firmware and I believe that to be an old marlin version format.

closed 2020-12-27 by jneilliii

AcJoker2, 2020-12-27:
I am using the latest Klipper firmware, the firmware outputs the positions.
So my question is whether this is a bug or is unfortunately not implemented.
Unfortunately, this shows a completely wrong picture, which is particularly unfavorable for beginners.

jneilliii, 2020-12-27:
It is not implemented I suppose then. Happy to accept a PR with the necessary changes to do so, just like I did for the other firmware.