jneilliii/OctoPrint-BedLevelVisualizer #344: [FR]: Z axis scale with automatic settings

dewi-ny-je, 2020-12-29:
I see in a previous thread and commit from July that the Z scale can be adapted.

I use klipper and, as recommended, my printer "homes away", the endstop are at the end of the Z axis.
This means that the bed mesh is not centered around zero, but around a value that doesn't need to be zero, and depending on the surface I use on that print, it may change.

As of now, colours are useless to me because they assume zero is the best.

Of course I'm setting a different Z range in the settings, but I think that one of the options should be to automatically center the colour scale to whatever the average value or the centre value (but meshes could be 4*4 and there is no centre value) are.

If you think about it, once bed mesh is used, there is no reason why the mesh should be ideally zero, what matters is that the min and max values are as close as possible.

Auto-centering the scale around the average would do that.

Proposal to combine the improvement you already implemented with my suggestion: instead of setting z min and z max in the settings, offer two text boxes: reference Z value and max Z deviation. Put a checkbox next to each one of them labelled "auto".
Reference Z (0 as default) and Z max deviation (1 mm as default, as it is now) both manual are equivalent to what you implemented, just a different way of setting them.
Z reference with auto would auto-centre the scale.
Max Z deviation auto would scale the amplitude for users with very warped beds or for those who want to zoom in if they have a very flat bed.

jneilliii, 2020-12-29:
I don't think I quite I understand the request, but I think if you were to set the z limits to `auto,auto` it will get you closer to what you are after?

jneilliii, 2020-12-29:
and the colormap set to `RdBu`.

dewi-ny-je, 2021-01-18:
Actually I found that there is already an option in the settings, it was just named in a way I didn't recognise it. With that option the values are shifted to the average value, as I requested (I don't have the printer available right now, but the checkbox is in the top right I think).

FR closed.

closed 2021-01-18 by dewi-ny-je