jneilliii/OctoPrint-BedLevelVisualizer #345: [BUG]: Will not install

NA7KR, 2020-12-30:
root@octopi:/home/pi/oprint/bin# pip install "https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip"
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip
  Using cached https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip (2.8 MB)
    ERROR: Command errored out with exit status 255:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-zyqptf0c/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-zyqptf0c/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-h36mgt12
         cwd: /tmp/pip-req-build-zyqptf0c/
    Complete output (1 lines):
    Could not import OctoPrint's setuptools, are you sure you are running that under the same python installation that OctoPrint is installed under?
    ----------------------------------------
ERROR: Command errored out with exit status 255: python setup.py egg_info Check the logs for full command output.
root@octopi:/home/pi/oprint/bin#

Will not install from plugin or cmd line

NA7KR, 2020-12-30:
Please close or delete OK now.

closed 2020-12-30 by NA7KR

Youras, 2023-04-03:
I am having the same error. 
How has this issue been solved?

In my case I don't  have internet access on my octopi. Thus, I downloaded the "https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip" file to my pc and sent it to the octopi. Then In tried install with

`pip install ./OctoPrint-BedLevelVisualizer-master.zip`

Which gives almost the same error.

jneilliii, 2023-04-03:
Are you running the pip command inside the OctoPrint venv? You can install via zip from plugin manager in OctoPrint web interface too. Sounds like you're running into issue that will be fixed with OctoPrint 1.9.0.