jneilliii/OctoPrint-BedLevelVisualizer #348: [FR]: Update texts to clarify that firmware with mesh bed leveling or a physical auto level sensor is needed

ChristophDerndorfer, 2020-12-30:
**Is your feature request related to a problem? Please describe.**

I installed the plugin on OctoPrint 1.5.2 for my Anycubic i3 Mega S printer. It then took me a while and some research to figure out that I would need firmware with mesh bed leveling support and ideally an auto level sensor like the BLTouch to actually gather the data that the plugin processes. I realize that this meets the definition of a newbie mistake but as I'm not the only [one](https://www.thingiverse.com/groups/octoprint/forums/general/topic:35347#comment-2392094) it would be helpful if these requirements were clarified somewhere.

**Describe the solution you'd like**

I think that updating the description on https://plugins.octoprint.org/, the README.md, and the wiki with a single, clear sentence would go a long way in helping other newcomers to avoid this misunderstanding.

jneilliii, 2020-12-30:
So I'm a little confused because the very first sentence in the readme is exactly that is it not? I see that the plugin repo entry should be updated to indicate that more clearly, but would appreciate any input as to how I could word it to be more clear than in the readme?

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer#octoprint-bedlevelvisualizer

ChristophDerndorfer, 2021-01-01:
How about adding something along the lines of:

"Please note that in order to use this plugin your 3D printer needs to support bed mesh leveling in software or come equipped with an auto level sensor supported by the printer's firmware (some entry-level printers do not provide such support per default)."

The README.md would then look like this:

This plugin visualises the output from various firmware that support bed mesh leveling, noteably the Prusa G81 mesh leveling report and the Marlin G29 T bed topography report. The plugin utilizes Plotly js library to render a 3D surface of the bed's reported mesh on a tab within OctoPrint.
Please note that in order to use this plugin your 3D printer needs to support bed mesh leveling in software or come equipped with an auto level sensor supported by the printer's firmware (some entry-level printers do not provide such support per default).

jneilliii, 2021-01-02:
Thanks that makes sense. I'll get that updated on both sides in the near future.

PR #398 (2021-02-28): 1.0.1
**Added**
* optional timeout override to @BEDLEVELVISUALIZER command, resolves #314
* settings button on tab for quick access, resolves #349

**Updated**
* README.md adding note that auto bed leveling must be possible on printer, resolves #348

**Fixed**
* allow for probe points that contain more than single digits, resolves #352, resolves #269
* resolve issues with circular beds
closed 2021-02-28 by jneilliii