jneilliii/OctoPrint-BedLevelVisualizer #350: [BUG]: MacOS - Buggy Accelerate Backend when using numpy 1.19

AlxSZP, 2021-01-03:
**Describe the bug**
After installing and successfully running Octoprint on MAcOS Catalina, I proceeded to install the plugin through the Octoprint UI, install dialog completed after a few minutes and required a restart of Octoprint.

Once restarting the server, the following message appeared:

**Python (697,0x10fb3cdc0) malloc: can't allocate region**
Listing a number of files in different directories (see screenshot)
**RuntimeError: Polyfit sanity test emitted a warning, most likely due to using a buggy Accelerate backend. If you compiled yourself, see site.cfg.example for information. Otherwise report this to the vendor that provided NumPy.**

**Expected behavior**
Octorpint should run with the plugin

Per research on the following link / Open ticket:
https://github.com/numpy/numpy/issues/15947
**"Closing. We strongly recommend not using Accelerate."**

**Debug Log**
Don't have a log

**Screenshots**
[image]

**Firmware and Version**
MacOs Catalina 10.15.7
Python 3.9 (installed in virtual environment)

AlxSZP, 2021-01-03:
Hi @jneilliii Thanks for the quick answer! 
I'm in the process of cleaning the MacBook in preparation for a clean install of the OS tomorrow. The idea will be to wipe it clean for Octoprint to run on it.
Once I do this, I'll give it another try and let you know. 
I'll be happy to help you test out with MacOS (I know there's a low population opting for MacOS vs the usual pi setup.

vtpearce91, 2021-02-12:
I had this same issue and the recommended

> I don't have Mac available to test against but this looks like an issue with the wheel for MacOS, or the numpy module trying to compile based on that issue you linked. You should be able to install a slightly older version of numpy and it would work.
> 
> ```
> /Users/wero/OctoPrint/venv/bin/pip install numpy==1.18.0
> ```

I had this same issue and the recommended solution worked for me.

jneilliii, 2021-02-12:
Reopening to remind me to update known issues.

jneilliii, 2022-01-01:
unnecessary now that numpy has been removed.

closed 2022-01-01 by jneilliii